grzegorz914/homebridge-enphase-envoy #2: Plugin is crashing homebridge

hoefs, 2020-07-01:
[7/1/2020, 8:57:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, power production: 0 kW
[7/1/2020, 8:57:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, energy production Today: 0 kWh
[7/1/2020, 8:57:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, energy production last 7 Days: 0 kWh
[7/1/2020, 8:57:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, energy production Lifetime: 0 kWh
[7/1/2020, 8:57:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, max power production: 0 kW
[7/1/2020, 8:58:20 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, state: Online.
[7/1/2020, 8:58:20 AM] TypeError: Cannot read property 'whToday' of undefined
    at envoyDevice.getDeviceState (/homebridge/node_modules/homebridge-enphase-envoy/index.js:207:61)
    at envoyDevice.getDeviceInfo (/homebridge/node_modules/homebridge-enphase-envoy/index.js:163:8)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
[7/1/2020, 8:58:20 AM] Got SIGTERM, shutting down Homebridge...
[7/1/2020, 8:58:20 AM] [ZP] closed http://192.168.1.10:34587/notify
[7/1/2020, 8:58:20 AM] [enphaseEnvoy] -------- Envoy-S --------
[7/1/2020, 8:58:20 AM] [enphaseEnvoy] Manufacturer: Enphase
[7/1/2020, 8:58:20 AM] [enphaseEnvoy] Model: Envoy-S
[7/1/2020, 8:58:20 AM] [enphaseEnvoy] Serialnr: 121921106049
[7/1/2020, 8:58:20 AM] [enphaseEnvoy] Firmware: R4.10.35
[7/1/2020, 8:58:20 AM] [enphaseEnvoy] Inverters: 14
[7/1/2020, 8:58:21 AM] [enphaseEnvoy] ----------------------------------

Seems like it is detecting it, but is crashing after some seconds.

hoefs, 2020-07-01:
this seems to be working:
http://192.168.2.25/production.json
{"production":[{"type":"inverters","activeCount":14,"readingTime":1593594523,"wNow":812,"whLifetime":2854507}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}

But I can't see a whToday...

grzegorz914, 2020-07-01:
Ok, 
I see U do not use any meters, will be fixed in next update.

Thanks

hoefs, 2020-07-01:
Can I add meters to my installation?
Seems like all the needed data is in the /api/production or am I wrong. Total for the day and the current wattage are there. A reason why the json is used and not the local api?

grzegorz914, 2020-07-01:
If Your Envoy-S have inputs for Productions and Consumption CT then Yes.

closed 2020-07-02 by grzegorz914

hoefs, 2020-07-02:
I have updated to latest. Homebridge is not crashing, but I cant add the accessory, it does not show up as before the update...

hoefs, 2020-07-02:
Fixed with manually removing the 2 files in the persistent folder

grzegorz914, 2020-07-02:
Change its name restart HB and add it again

commit 5ee63f8c (2020-07-02): fix #2

- fixed #2 crash if no production meters are installed