grzegorz914/homebridge-enphase-envoy #3: Connection appears successful, but no accessories are appearing

jcgillespie, 2020-12-29:
I've got this plugin configured and it looks like it is able to retrieve all the data from my Envoy, but I'm not seeing any accessories in HomeBridge. Any suggestions?

**Log snip**
```
[29/12/2020, 10:21:40] [enphaseEnvoy] Device 192.168.9.104 Envoy-S, get device status data: { production: [Array], consumption: [Array], storage: [Array] }
[29/12/2020, 10:21:40] [enphaseEnvoy] Device: 192.168.9.104 Envoy-S, power production: 0.257 kW
[29/12/2020, 10:21:40] [enphaseEnvoy] Device: 192.168.9.104 Envoy-S, power production max: 0.257 kW
[29/12/2020, 10:21:40] [enphaseEnvoy] Device: 192.168.9.104 Envoy-S, power production max detected: Yes
[29/12/2020, 10:21:40] [enphaseEnvoy] Device: 192.168.9.104 Envoy-S, energy production Lifetime: 7728.135 kWh
[29/12/2020, 10:21:40] [enphaseEnvoy] Device: 192.168.9.104 Envoy-S, powerConsumptionTotalMaxFile saved successful in: /var/lib/homebridge/enphaseEnvoy/ 1.270155 kW
[29/12/2020, 10:21:40] [enphaseEnvoy] Device: 192.168.9.104 Envoy-S, total power consumption : 1.270155 kW
[29/12/2020, 10:21:40] [enphaseEnvoy] Device: 192.168.9.104 Envoy-S, total power consumption max: 0.257 kW
[29/12/2020, 10:21:40] [enphaseEnvoy] Device: 192.168.9.104 Envoy-S, total power consumption max detected: Yes
[29/12/2020, 10:21:40] [enphaseEnvoy] Device: 192.168.9.104 Envoy-S, total energy consumption Today: 16.266201 kWh
[29/12/2020, 10:21:40] [enphaseEnvoy] Device: 192.168.9.104 Envoy-S, total energy consumption last 7 Days: 226.321201 kWh
[29/12/2020, 10:21:40] [enphaseEnvoy] Device: 192.168.9.104 Envoy-S, total energy consumption Lifetime: 8637.574201 kWh
```

**Config**
```
{
            "devices": [
                {
                    "name": "Envoy-S",
                    "host": "192.168.9.104",
                    "refreshInterval": 30,
                    "enchargeStorage": false,
                    "enchargeStorageOffset": 0,
                    "powerProductionMeter": 0,
                    "powerProductionMaxDetected": 0,
                    "energyProductionLifetimeOffset": 0,
                    "powerConsumptionMeter": 1,
                    "powerConsumptionTotalMaxDetected": 0,
                    "energyConsumptionTotalLifetimeOffset": 0,
                    "powerConsumptionNetMaxDetected": 0,
                    "energyConsumptionNetLifetimeOffset": 0
                }
            ],
            "platform": "enphaseEnvoy"
        }
```

grzegorz914, 2020-12-29:
U need to pair first in HomeKit app
https://github.com/grzegorz914/homebridge-enphase-envoy#homekit-pairing

jcgillespie, 2020-12-29:
They’re not showing in HomeBridge. Do you need to pair to HomeKit to get them to show in the HomeBridge Config UI?

https://github.com/grzegorz914/homebridge-enphase-envoy#homekit-pairing

grzegorz914, 2020-12-29:
Ah OK, Homebridge config UI X don’t support all accessory type, this one is not supported and will not be visible in Config UI X, please Use EVE App or Controller app on iPhone/iPad

jcgillespie, 2020-12-30:
That did it. Thank you. I didn’t realize the config UI didn’t support all the types

closed 2020-12-30 by jcgillespie