grzegorz914/homebridge-enphase-envoy #5: Homebrdige crashing with v3.5.63

hoefs, 2021-01-29:
Hi,

Just updated to latest version. Now homebridge is crashing:

```
[1/29/2021, 2:58:40 PM] ReferenceError: me is not defined
    at new envoyDevice (/homebridge/node_modules/homebridge-enphase-envoy/index.js:274:47)
    at HomebridgeAPI.<anonymous> (/homebridge/node_modules/homebridge-enphase-envoy/index.js:195:11)
    at HomebridgeAPI.emit (events.js:327:22)
    at HomebridgeAPI.signalFinished (/usr/local/lib/node_modules/homebridge/src/api.ts:251:10)
    at Server.start (/usr/local/lib/node_modules/homebridge/src/server.ts:158:14)
    at cli (/usr/local/lib/node_modules/homebridge/src/cli.ts:80:10)
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge/bin/homebridge:17:22)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:72:12)
    at internal/main/run_main_module.js:17:47
[1/29/2021, 2:58:40 PM] Got SIGTERM, shutting down Homebridge...
[1/29/2021, 2:58:40 PM] ERR_SERVER_CLOSED: Cannot send packets on a closed mdns server!
    at MDNSServer.assertBeforeSend (/usr/local/lib/node_modules/homebridge/node_modules/@homebridge/ciao/src/MDNSServer.ts:403:13)
    at MDNSServer.sendOnAllNetworksForService (/usr/local/lib/node_modules/homebridge/node_modules/@homebridge/ciao/src/MDNSServer.ts:302:10)
    at MDNSServer.sendQueryBroadcast (/usr/local/lib/node_modules/homebridge/node_modules/@homebridge/ciao/src/MDNSServer.ts:263:26)
    at Prober.sendProbeRequest (/usr/local/lib/node_modules/homebridge/node_modules/@homebridge/ciao/src/responder/Prober.ts:165:17)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
```

hoefs, 2021-01-29:
Now it is starting but with address collision:
```
1/29/2021, 3:33:07 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, last inverter: undefined power: NaN W
[1/29/2021, 3:33:07 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, max inverter: undefined power: NaN W
[1/29/2021, 3:33:07 PM] [enphaseEnvoy] prepareAccessory
[1/29/2021, 3:33:07 PM] [enphaseEnvoy] prepareInformationService
[1/29/2021, 3:33:07 PM] [enphaseEnvoy] prepareEnvoyService
[1/29/2021, 3:33:07 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, publishExternalAccessories.
[1/29/2021, 3:33:07 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, update Device state error: Error: Accessory Envoy-S experienced an address collision.
    at /usr/local/lib/node_modules/homebridge/src/server.ts:581:15
    at Array.forEach (<anonymous>)
    at Server.handlePublishExternalAccessories (/usr/local/lib/node_modules/homebridge/src/server.ts:561:17)
    at HomebridgeAPI.emit (events.js:315:20)
    at HomebridgeAPI.publishExternalAccessories (/usr/local/lib/node_modules/homebridge/src/api.ts:302:10)
    at envoyDevice.prepareAccessory (/homebridge/node_modules/homebridge-enphase-envoy/index.js:328:14)
    at envoyDevice.updateDeviceState (/homebridge/node_modules/homebridge-enphase-envoy/index.js:949:10)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
```

hoefs, 2021-01-29:
Last update _homebridge-enphase-envoy v3.5.75_ seems to fix that issue...

What is this inverter thing?
```
[1/29/2021, 3:34:43 PM] [enphaseEnvoy] Device 192.168.2.25 Envoy-S, get device status data: {
  wattHoursToday: 2511,
  wattHoursSevenDays: 26004,
  wattHoursLifetime: 4781330,
  wattsNow: 179
}, data1 { production: [Array], storage: [Array] }
[1/29/2021, 3:34:43 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, power production: 0.179 kW
[1/29/2021, 3:34:43 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, power production max: 3.421 kW
[1/29/2021, 3:34:43 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, power production max detected: No
[1/29/2021, 3:34:43 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, energy production Today: 2.511 kWh
[1/29/2021, 3:34:43 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, energy production last 7 Days: 26.004 kWh
[1/29/2021, 3:34:43 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, energy production Lifetime: 4781.33 kWh
[1/29/2021, 3:34:46 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, last inverter: undefined power: NaN W
[1/29/2021, 3:34:46 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, max inverter: undefined power: NaN W
```

hoefs, 2021-01-29:
homebridge-enphase-envoy v3.5.77 (fast updates...)

This drops a lot of html code in de homebridge logs...

```
[1/29/2021, 3:48:07 PM] [enphaseEnvoy] Device 192.168.2.25 Envoy-S, get device status data: {
  wattHoursToday: 2545,
  wattHoursSevenDays: 25877,
  wattHoursLifetime: 4781364,
  wattsNow: 112
}, data1 { production: [Array], storage: [Array] }
[1/29/2021, 3:48:07 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, power production: 0.112 kW
[1/29/2021, 3:48:07 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, power production max: 3.421 kW
[1/29/2021, 3:48:07 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, power production max detected: No
[1/29/2021, 3:48:07 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, energy production Today: 2.545 kWh
[1/29/2021, 3:48:07 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, energy production last 7 Days: 25.877 kWh
[1/29/2021, 3:48:07 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, energy production Lifetime: 4781.364 kWh
[1/29/2021, 3:48:09 PM] [enphaseEnvoy]       <!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01//EN">
      <html lang="en">
      <!-- START HEAD CONTENT -->
      <head>
        <title>error</title>
        <meta http-equiv="Content-Type" content="text/html; charset=utf-8">
        <link rel="icon"
              type="image/png" href="/images/favicon.ico?v=6">
        <!-- include style sheet -->
        <link rel="styleSheet" type="text/css" href="/include/css/style.css" >
        <style type="text/css">.nav { width: 190px; font-size: 10px; padding: 10px; margin: 10px;float: left; background: #D8DBD4; color: #213d30; border: 2px solid #333; }
        .navholder { width: auto; height: auto; float: left; margin: 10px 0px 0 0 }
        .tbl_cap { font-size:8pt; background-color:#DD4F05; color:white; border-style:solid; border-width:1px; border-color:white; }
        .tbl_bod { font-family:arial; font-size:8pt; color:#213d30; background-color:#D8DBD4; border-style:solid; border-color:white; border-width:1px; }
        .tbl_bod_right { font-family:arial; font-size:8pt; color:#213d30; background-color:#D8DBD4; border-style:solid; border-color:white; border-width:1px; text-align:right }
        .tbl_hdr { font-size:8pt; background-color:#CCCCCC; border-style:solid; border-width:1px; border-color:white; color:#213d30; }
        .clearer { clear: both; font-size: 1px; }
        .outer { width: auto; height: auto; padding: 0px; font-size: 10px }
        .table1 { width: 100%; margin-left: auto; margin-right: auto;border: 1px solid #CCCCCC }
        .table2 { width: 100%;text-align: center;margin-left: auto; margin-right: auto;border: 1px solid #CCCCCC }
        .table3 { width: 720px;margin-left: auto; margin-right: auto;border: 1px solid #CCCCCC }
        .div_params { height: 190px; width: 380px; overflow: auto;margin-left: auto; margin-right: auto; }
        .div_edit_params { height: 230px; width: 470px; overflow: auto;margin-left: auto; margin-right: auto; }
        .div_outer_box { width: 720px;margin-left: auto; margin-right: auto;border: 1px solid #FFFFFF }
        .txt_expl_center { font-size: 12px; text-align:center;height: auto; border: 1px solid #CCCCCC;margin-top: 5px; margin-left: 5px; margin-right: auto; }
        .label_text_right { text-align: right; border: 1px solid #CCCCCC;font-size: 10px; }
        .label_text_left { text-align: left; border: 1px solid #CCCCCC;font-size: 10px; }
        .label_text_left_alert { text-align: left; color: #7d3020; border: 1px solid #CCCCCC;font-size: 12px; }
        .link_cell_right { font-size: 10px;height: auto; border: 1px solid #CCCCCC;text-align: right;margin-top: 5px; margin-left: auto; margin-right: 25px; }
        .link_cell_left { font-size: 10px;height: auto; border: 1px solid #CCCCCC;text-align: left;margin-top: 5px; margin-left: 25px; margin-right: auto; }
        .link_cell_right_out { font-size: 12px; padding: 15px;height: auto; border: 1px solid #CCCCCC;text-align: right;margin-top: 5px; margin-left: auto; margin-right: 25px; }
        .link_cell_left_out { font-size: 12px;height: auto; border: 1px solid #CCCCCC;text-align: left; padding: 15px;margin-top: 5px; margin-left: 25px; margin-right: auto; }
        .ns_fldset { height: auto; width: 720px; border: 1px solid #7d3020; background: #CCCCCC; padding: 15px; }
        .ns_fldset_inner { border: 1px solid #7d3020; background: #CCCCCC; }
        .enp_legend { text-align: left; border: 1px solid #7d3020; background: #FFD401; }
        .input_left { border: 1px solid #CCCCCC; text-align: left; }
        .grid_config { float: left; margin-left: 50px; width: 830px; text-align: center; }
        .swimgnav { width: 160px; height: auto; font-size: 10px; padding: 10px; margin: 10px;float: left; background: #D8DBD4; color: #213d30;border: 2px solid #333; }
        .frames { clear: right; width: auto; padding: 15px; height: auto; margin-left: 200px; margin-right: auto; }
        .tbl_cap_left { font-size:8pt; background-color:#DD4F05; color:white; border-style:solid; border-width:1px; border-color:white; text-align:left; }
        .tbl_cap_center { font-size:8pt; background-color:#DD4F05; color:white; border-style:solid; border-width:1px; border-color:white; text-align:center; }
        .tbl_bod_left { font-family:arial; font-size:8pt; color:#213d30; background-color:#D8DBD4; border-style:solid; border-color:white; border-width:1px; text-align:left; }
[... cut ...]
      <div style="margin-left: auto; margin-right:auto; width: 100%; text-align: center; ">
      &copy; 2007-2018, [e] Enphase Energy, Inc. All rights reserved. |
      <a href="http://www.enphaseenergy.com/licenses">Licenses</a>
      <!-- END BOTTOM NAV CONTENT -->
      <br><br>
      </div>
      </body>
      </html>

[1/29/2021, 3:48:09 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, last inverter: undefined power: NaN W
[1/29/2021, 3:48:09 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, max inverter: undefined power: NaN W
```

hoefs, 2021-01-29:
Same html in logs with homebridge-enphase-envoy v3.5.79

hoefs, 2021-01-29:
New update shows error:
```
USER: root
DIR: /homebridge
CMD: npm install --save homebridge-enphase-envoy@latest

npm ERR! code ETARGET
npm ERR! notarget No matching version found for http@0.0.1-secutity.
npm ERR! notarget In most cases you or one of your dependencies are requesting
npm ERR! notarget a package version that doesn't exist.
npm ERR! notarget 
npm ERR! notarget It was specified as a dependency of 'homebridge-enphase-envoy'
npm ERR! notarget 

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-01-29T15_02_59_446Z-debug.log

Command failed. Please review log for details.
```

grzegorz914, 2021-01-29:
please do not update without announcement please still on 3.5.0

hoefs, 2021-01-29:
> please do not update without announcement please still on 3.5.0

How can I roll back? The updates just show up in homebridge GUI...

Please advise

hoefs, 2021-01-29:
New error for homebridge-enphase-envoy v3.5.83

`[1/29/2021, 4:11:01 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, could not read inverters production, error: TypeError [ERR_INVALID_PROTOCOL]: Protocol "http:" not supported. Expected "https:"`

hoefs, 2021-01-29:
homebridge-enphase-envoy v3.5.86

[1/29/2021, 4:32:39 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, could not read inverters production, error: Error: Request failed with status code 401
    at createError (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/createError.js:16:15)
    at settle (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/settle.js:17:12)
    at IncomingMessage.handleStreamEnd (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/adapters/http.js:260:11)
    at IncomingMessage.emit (events.js:327:22)
    at endReadableNT (internal/streams/readable.js:1327:12)
    at processTicksAndRejections (internal/process/task_queues.js:80:21) {
  config: [Object],
  request: [ClientRequest],
  response: [Object],
  isAxiosError: true,
  toJSON: [Function: toJSON]

hoefs, 2021-01-29:
homebridge-enphase-envoy v3.5.89
```
[1/29/2021, 4:44:54 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, could not read inverters production, error: Error: Request failed with status code 401
    at createError (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/createError.js:16:15)
    at settle (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/settle.js:17:12)
    at IncomingMessage.handleStreamEnd (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/adapters/http.js:260:11)
    at IncomingMessage.emit (events.js:327:22)
    at endReadableNT (internal/streams/readable.js:1327:12)
    at processTicksAndRejections (internal/process/task_queues.js:80:21) {
  config: [Object],
  request: [ClientRequest],
  response: [Object],
  isAxiosError: true,
  toJSON: [Function: toJSON]
}
```

hoefs, 2021-01-29:
homebridge-enphase-envoy v3.5.94

[1/29/2021, 5:12:23 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, could not read inverters production, error: Error: Request failed with status code 401
    at createError (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/createError.js:16:15)
    at settle (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/settle.js:17:12)
    at IncomingMessage.handleStreamEnd (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/adapters/http.js:260:11)
    at IncomingMessage.emit (events.js:327:22)
    at endReadableNT (internal/streams/readable.js:1327:12)
    at processTicksAndRejections (internal/process/task_queues.js:80:21) {
  config: [Object],
  request: [ClientRequest],
  response: [Object],
  isAxiosError: true,
  toJSON: [Function: toJSON]

hoefs, 2021-01-29:
homebridge-enphase-envoy v3.5.96

```
[1/29/2021, 5:24:11 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, could not read inverters production, error: Error: Request failed with status code 401
    at createError (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/createError.js:16:15)
    at settle (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/settle.js:17:12)
    at IncomingMessage.handleStreamEnd (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/adapters/http.js:260:11)
    at IncomingMessage.emit (events.js:327:22)
    at endReadableNT (internal/streams/readable.js:1327:12)
    at processTicksAndRejections (internal/process/task_queues.js:80:21) {
  config: [Object],
  request: [ClientRequest],
  response: [Object],
  isAxiosError: true,
  toJSON: [Function: toJSON]
}
```

hoefs, 2021-01-29:
Same html in logs with homebridge-enphase-envoy v3.5.104

grzegorz914, 2021-01-29:
v3.6.0 released

closed 2021-01-29 by grzegorz914

hoefs, 2021-01-30:
When I try to update it wants to update to 3.6.61 beta...