grzegorz914/homebridge-enphase-envoy #6: Inverter numbers don't seem right in latest version

sddgit, 2021-01-30:
Hi. Thank you for updating this plugin.

I'm having some trouble interpreting what I'm seeing. I have 46 micro inverters, which seem to be picked up OK, and are shown in the logs as producing power at startup. But when I look at the devices in the Eve app, the inverters are shown as inactive, and the logs say they are not producing power. I have put the Envoy user and password into the config, although when I go to the web interface, Diagnostic Tools, and enter those credentials, I just get a spinner. I've attached a screenshot from Eve.

Any ideas? Relevant log entries follow:

**At startup**

[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, state: Online.
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] -------- Envoy-S --------
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Manufacturer: Enphase
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Model: Envoy
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Serialnr: 121716041706
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Firmware: D5.0.48
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Inverters: 46
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Encharges: 0
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Q-Relays: 0
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Meters: 2
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] ----------------------------------
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Inverter 1: 121532051083
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Producing: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Communicating: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Provisioned: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Operating: No
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Status: envoy.global.ok
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] ----------------------------------
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Inverter 2: 121532051073
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Producing: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Communicating: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Provisioned: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Operating: No
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Status: envoy.global.ok
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] ----------------------------------
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Inverter 3: 121524045135
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Producing: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Communicating: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Provisioned: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Operating: No
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Status: envoy.global.ok
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] ----------------------------------
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Inverter 4: 121532051509
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Producing: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Communicating: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Provisioned: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Operating: No
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Status: envoy.global.ok
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] ----------------------------------
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Inverter 5: 121532051099
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Producing: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Communicating: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Provisioned: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Operating: No
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Status: envoy.global.ok
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] ----------------------------------
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Inverter 6: 121615031972
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Producing: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Communicating: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Provisioned: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Operating: No
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Status: envoy.global.ok
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] ----------------------------------
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Inverter 7: 121532051668
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Producing: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Communicating: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Provisioned: Yes
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Operating: No
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Status: envoy.global.ok
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] ----------------------------------
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Inverter 8: 121532051743
[1/30/2021, 6:02:08 PM] [enphaseEnvoy] Producing: Yes
[... 420 lines cut ...]
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121524038236 last power: 66 W
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121524038236 max power: 259 W
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532051669 producing: No
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532051669 last power: 71 W
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532051669 max power: 259 W
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532051743 producing: No
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532051743 last power: 68 W
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532051743 max power: 259 W
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532052190 last power: 64 W
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532052190 producing: No
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532052190 max power: 259 W
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121521049697 last power: 69 W
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121521049697 max power: 259 W
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121521049697 producing: No
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532051369 last power: 69 W
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532051369 max power: 259 W
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532051369 producing: No
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532052925 producing: No
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532052925 last power: 151 W
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532052925 max power: 262 W
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532051973 producing: No
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532051973 max power: 259 W
[1/30/2021, 5:59:36 PM] [enphaseEnvoy] Device: envoy.local Envoy-S, inverter: 121532051973 last power: 68 W

[image]

grzegorz914, 2021-01-30:
Update to 4.0.0 and get feedback.

grzegorz914, 2021-01-30:
All is correct innthe log, Last Power in The inverter wił be Clearex automatically ta midnigjt, Hovever The Microinverter not producig The Power stored it last Power and Max Power in memory.
If U watt to login as installer in Installer Tolkiena or web interface need to use Inspaller password not Envoy password.

sddgit, 2021-01-31:
I've updated to 4.0.0 and things look much better. All inverters are shown as producing and are connected properly in the Eve app. I'm a little suspicious that the max power is the same for almost all the inverters and that the last power is also very similar for all of them. My panels are at different angles, so produce differing amounts of power. I'll keep an eye on it when the sun is shining more than today!

In the config, I only have envoyUser (envoy) and envoyPasswd (last 6 digits of serial number). I don't know my installer password, and don't have an Android device to run that generating app. What am I missing out on if I don't have that?

Thank for the plugin, and for the help...

grzegorz914, 2021-01-31:
Right now installer settings in the config are for future expand the plugin.
Regarding 
last and Max Power please check these value in Your Web interface and compare it.
These value are refresh during inverter itself in 5 min duration and can be very similar.

sddgit, 2021-01-31:
Please excuse my ignorance, but how do I check all the inverters in the web interface?

grzegorz914, 2021-01-31:
http://envoy.local/api/v1/production/inverters

closed 2021-01-31 by grzegorz914