grzegorz914/homebridge-enphase-envoy #7: devType undefined

jr7216, 2021-01-31:
With te last update I receive the following error

TypeError: Cannot read property 'devType' of undefined
    at output (/homebridge/node_modules/homebridge-enphase-envoy/index.js:1521:55)
    at done (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/urllib/lib/urllib.js:617:5)
    at /homebridge/node_modules/homebridge-enphase-envoy/node_modules/urllib/lib/urllib.js:911:9
    at decodeContent (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/urllib/lib/urllib.js:699:14)
    at handleResponseCloseAndEnd (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/urllib/lib/urllib.js:876:7)
    at IncomingMessage.<anonymous> (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/urllib/lib/urllib.js:920:7)
    at IncomingMessage.emit (events.js:327:22)
    at endReadableNT (internal/streams/readable.js:1327:12)
    at processTicksAndRejections (internal/process/task_queues.js:80:21)

sparky742, 2021-01-31:
I'm seeing this also.  It keeps sending home bridge into a boot loop...

```
[1/31/2021, 11:16:59] TypeError: Cannot read property 'devType' of undefined
    at output (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:1294:55)
    at done (/usr/local/lib/node_modules/homebridge-enphase-envoy/node_modules/urllib/lib/urllib.js:617:5)
    at /usr/local/lib/node_modules/homebridge-enphase-envoy/node_modules/urllib/lib/urllib.js:911:9
    at decodeContent (/usr/local/lib/node_modules/homebridge-enphase-envoy/node_modules/urllib/lib/urllib.js:699:14)
    at handleResponseCloseAndEnd (/usr/local/lib/node_modules/homebridge-enphase-envoy/node_modules/urllib/lib/urllib.js:876:7)
    at IncomingMessage.<anonymous> (/usr/local/lib/node_modules/homebridge-enphase-envoy/node_modules/urllib/lib/urllib.js:920:7)
    at IncomingMessage.emit (events.js:327:22)
    at endReadableNT (_stream_readable.js:1220:12)
    at processTicksAndRejections (internal/process/task_queues.js:84:21)
[1/31/2021, 11:16:59] Got SIGTERM, shutting down Homebridge...
```

jr7216, 2021-01-31:
> I'm seeing this also. It keeps sending home bridge into a boot loop...
>
> ```
> [1/31/2021, 11:16:59] TypeError: Cannot read property 'devType' of undefined
>     at output [...]

jr7216, 2021-01-31:
The boot loop is the result of the TypeError. The boot loop end if the plugin is deleted.

grzegorz914, 2021-01-31:
Please put this in web browser and get the output:

http://envoy.local/api/v1/production/inverters

closed 2021-01-31 by grzegorz914

jr7216, 2021-01-31:
[
  {
    "serialNumber": "122022001491",
    "lastReportDate": 1612110254,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 152
  },
  {
    "serialNumber": "122021062205",
    "lastReportDate": 1612110261,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 158
  },
  {
    "serialNumber": "122022001349",
    "lastReportDate": 1612110269,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 168
  },
  {
    "serialNumber": "122022000357",
    "lastReportDate": 1612110255,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 166
  },
  {
    "serialNumber": "122021058482",
    "lastReportDate": 1612110263,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 172
  },
  {
    "serialNumber": "122022001104",
    "lastReportDate": 1612110270,
    "devType": 1,
    "lastReportWatts": -1,
    "maxReportWatts": 140
  },
  {
    "serialNumber": "122022001489",
    "lastReportDate": 1612109956,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 143
  },
  {
    "serialNumber": "122022001352",
    "lastReportDate": 1612110264,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 166
  },
  {
    "serialNumber": "122022001301",
    "lastReportDate": 1612110254,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 167
  },
  {
    "serialNumber": "122021061909",
    "lastReportDate": 1612110259,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 158
  },
  {
    "serialNumber": "122022001553",
    "lastReportDate": 1612109964,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 145
  }
]

sparky742, 2021-01-31:
Interesting….in my case, it won’t authenticate with envoy and my serial number (1219xxxxxxxx)

http://envoy.local/api/v1/production/inverters

grzegorz914, 2021-02-01:
Please post Your config.json

commit 4f4443ae (2021-01-31): fix #7