grzegorz914/homebridge-enphase-envoy #8: v4.0.0 =  Device state error: TypeError: Cannot read property '0' of undefined

hoefs, 2021-02-01:
```
[2/1/2021, 12:50:02 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, update Device state error: TypeError: Cannot read property '0' of undefined
    at envoyDevice.updateDeviceState (/homebridge/node_modules/homebridge-enphase-envoy/index.js:823:77), state: Offline
```

grzegorz914, 2021-02-01:
Please update to latest version.

hoefs, 2021-02-01:
> Please update to latest version.

Same issue with homebridge-enphase-envoy v4.0.34?

```
[2/1/2021, 1:41:00 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, update Device state error: TypeError: Cannot read property 'activeCount' of undefined
    at envoyDevice.updateDeviceState (/homebridge/node_modules/homebridge-enphase-envoy/index.js:631:74)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
```

grzegorz914, 2021-02-01:
Please getvthe output for

http://envoy.local/production.json

hoefs, 2021-02-01:
> Please get The output for
> 
> http://envoy.local/production.json

`{"production":[{"type":"inverters","activeCount":14,"readingTime":1612183759,"wNow":75,"whLifetime":4785715}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}`

grzegorz914, 2021-02-01:
OK
I know What is wrong, will update The plugin asap.
I have full system with all possible devices and with Your case The output is little different.

please also get response for:
http://Envoy.local/ivp/meters

hoefs, 2021-02-01:
> OK
> I know What is wrong, will update The plugin asap.
> I have full system with all possible devices and with Your case The output is little different.
> 
> please also get response for:
> http://Envoy.local/ivp/meters

http://192.168.2.25/ivp/meters output is just 2 brackets

`[]`

grzegorz914, 2021-02-01:
Ok 
Thanks

grzegorz914, 2021-02-01:
Make update  and test.

closed 2021-02-01 by grzegorz914

hoefs, 2021-02-01:
Same error with homebridge-enphase-envoy v4.0.35

```
[2/1/2021, 2:12:27 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, update Device state error: TypeError: Cannot read property 'count' of undefined
    at envoyDevice.updateDeviceState (/homebridge/node_modules/homebridge-enphase-envoy/index.js:633:21)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
```

grzegorz914, 2021-02-01:
Update to 4.0.36

hoefs, 2021-02-01:
> Update to 4.0.36

Still have error...

```
[2/1/2021, 2:16:00 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, update Device state error: ReferenceError: totalConsumtionMeterAvtiveCount is not defined
    at envoyDevice.updateDeviceState (/homebridge/node_modules/homebridge-enphase-envoy/index.js:705:7), state: Offline
```

grzegorz914, 2021-02-01:
Sorry,
Try again

hoefs, 2021-02-01:
> Sorry,
> Try again

Seems to be working now!!! 
Why are the betas instantly available via homebridge?

commit 455542d4 (2021-02-01): fix #8