grzegorz914/homebridge-enphase-envoy #9: Version v4.0.56: Cannot read property '0' of undefined

MchlLyn, 2021-02-02:
Hi,

I saw this issue: https://github.com/grzegorz914/homebridge-enphase-envoy/issues/8 which is closed now. I'm experiencing a similar issue using version 4.0.56, i'm assuming due to a slightly different setup/output? I tried reinstalling the pluging with a new config but this didn't help.

**The error is:**
[enphaseEnvoy] Device: 10.0.0.222 Envoy-S, update Device state error: TypeError: Cannot read property '0' of undefined
    at envoyDevice.updateDeviceState (/homebridge/node_modules/homebridge-enphase-envoy/index.js:767:139)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline

**Output from http://envoy.local/production.json:**
{"production":[{"type":"inverters","activeCount":13,"readingTime":1612259036,"wNow":478,"whLifetime":7909820}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}

**Output from http://envoy.local/ivp/meters:**
[]

Thanks in advance!

grzegorz914, 2021-02-02:
please update to 4.1.0

grzegorz914, 2021-02-02:
Fixed in 4.0.1

closed 2021-02-02 by grzegorz914

MchlLyn, 2021-02-02:
@grzegorz914 updated to 4.1.2 just now and it's fixed, that's quick!
Thanks :)

commit b12d326f (2021-02-02): fix #9