grzegorz914/homebridge-enphase-envoy #10: Wrong unit in Eve

jr7216, 2021-02-04:
The unit of measured power is kW, but the value is W.

jr7216, 2021-02-04:
Is it also posible to display the total power? All the inverters added up, and other details like precious versions? 
[image]

grzegorz914, 2021-02-04:
Please post the data of Your PV what U have installed:
Envoy type, Q-Relay, Microinverters type, Meter.
The kW unit is correct.
The Power is the TotalPower.

jr7216, 2021-02-04:
Envoy-S-Metered-EU,IQ7+no Q-relay, no meters
For the wrong unit see the screen dump in the next post. One panel producing 22 kW?
Now installed the last version from 1 view minutes ago 4.2.13. (the first screen in Eve is a complete  mesh, see next post screendump).
I cannot find the total power in the latest version. The previous screendump is of a older version.

jr7216, 2021-02-04:
[image]
[image]

jr7216, 2021-02-04:
[image]

jr7216, 2021-02-04:
[image]

jr7216, 2021-02-04:
[image]

jr7216, 2021-02-04:
[image]

grzegorz914, 2021-02-04:
Please get The output:
http://envoy.local/inventory.json
http://envoy.local/production.json
http://envoy.local/api/v1/production

jr7216, 2021-02-04:
[inventory.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/5926682/inventory.txt)
[production.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/5926685/production.txt)
[production-api.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/5926686/production-api.txt)

grzegorz914, 2021-02-04:
Please get also
http://envoy.local/ivp/meters

grzegorz914, 2021-02-04:
try update and get the feedback

jr7216, 2021-02-05:
[meters.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/5930937/meters.txt)

The last update 4.2.28 is great again, the right units en layout (position of de information). Thnx for the effort.

closed 2021-02-05 by jr7216