grzegorz914/homebridge-enphase-envoy #11: update Device state error: TypeError: Cannot read property 'type' of undefined

BeauGiles, 2021-02-05:
Hi,
Just installed this plugin for the first time, but I am getting errors in Homebridge, and can't add the accessory in the Home app.

Log has 

```
[06/02/2021, 07:59:05] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, state: Online.
[06/02/2021, 07:59:05] [enphaseEnvoy] -------- Envoy-S --------
[06/02/2021, 07:59:05] [enphaseEnvoy] Manufacturer: Enphase
[06/02/2021, 07:59:05] [enphaseEnvoy] Model: Envoy
[06/02/2021, 07:59:05] [enphaseEnvoy] Serial: XXXXXXX
[06/02/2021, 07:59:05] [enphaseEnvoy] Firmware: D5.0.49
[06/02/2021, 07:59:05] [enphaseEnvoy] Inverters: 51
[06/02/2021, 07:59:05] [enphaseEnvoy] Encharges: 12
[06/02/2021, 07:59:05] [enphaseEnvoy] Q-Relays: 0
[06/02/2021, 07:59:05] [enphaseEnvoy] Meters: 2
[06/02/2021, 07:59:05] [enphaseEnvoy] Time: 06/02/2021, 07:58:41
[06/02/2021, 07:59:05] [enphaseEnvoy] ----------------------------------
[06/02/2021, 07:59:15] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, update Device state error: TypeError: Cannot read property 'type' of undefined
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:1739:61)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
[06/02/2021, 07:59:36] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, state: Online.
[06/02/2021, 07:59:36] [enphaseEnvoy] -------- Envoy-S --------
[06/02/2021, 07:59:36] [enphaseEnvoy] Manufacturer: Enphase
[06/02/2021, 07:59:36] [enphaseEnvoy] Model: Envoy
[06/02/2021, 07:59:36] [enphaseEnvoy] Serial: XXXXXXX
[06/02/2021, 07:59:36] [enphaseEnvoy] Firmware: D5.0.49
[06/02/2021, 07:59:36] [enphaseEnvoy] Inverters: 51
[06/02/2021, 07:59:36] [enphaseEnvoy] Encharges: 12
[06/02/2021, 07:59:36] [enphaseEnvoy] Q-Relays: 0
[06/02/2021, 07:59:36] [enphaseEnvoy] Meters: 2
[06/02/2021, 07:59:36] [enphaseEnvoy] Time: 06/02/2021, 07:59:11
[06/02/2021, 07:59:36] [enphaseEnvoy] ----------------------------------
[06/02/2021, 07:59:51] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, update Device state error: TypeError: Cannot read property 'type' of undefined
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:1739:61)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
```

Config - have only entered the IP, and installer password

```
        {
            "devices": [
                {
                    "name": "Envoy-S",
                    "host": "10.0.1.243",
                    "refreshInterval": 30,
                    "installerUser": "installer",
                    "installerPasswd": "XXXXXXXXX",
                    "enchargeStorageOffset": 0,
                    "powerProductionMaxDetected": 0,
                    "energyProductionLifetimeOffset": 0,
                    "powerConsumptionTotalMaxDetected": 0,
                    "energyConsumptionTotalLifetimeOffset": 0,
                    "powerConsumptionNetMaxDetected": 0,
                    "energyConsumptionNetLifetimeOffset": 0
                }
            ],
            "platform": "enphaseEnvoy"
        }
```

grzegorz914, 2021-02-05:
Please get me output from

http://envoy.local/production.json
http://envoy.local/inventory.json

BeauGiles, 2021-02-05:
http://envoy.local/inventory.json

```
[{
	"type": "PCU",
	"devices": [{
		"part_num": "800-00539-r05",
		"installed": "1479767146",
		"serial_num": "121632035883",
		"device_status": ["envoy.global.ok"],
		"last_rpt_date": "1612559083",
		"admin_state": 1,
		"dev_type": 1,
		"created_date": "1479767146",
		"img_load_date": "1529964383",
		"img_pnum_running": "520-00066-r01-v00.34.09",
		"ptpn": "540-00105-r01-v00.34.04",
		"chaneid": 1627390225,
		"device_control": [{
			"gficlearset": false
		}],
		"producing": true,
		"communicating": true,
		"provisioned": true,
		"operating": true
	}, {
		"part_num": "800-00539-r05",
		"installed": "1479767178",
		"serial_num": "121632036078",
		"device_status": ["envoy.global.ok"],
		"last_rpt_date": "1612559084",
		"admin_state": 1,
		"dev_type": 1,
		"created_date": "1479767178",
		"img_load_date": "1529964383",
		"img_pnum_running": "520-00066-r01-v00.34.09",
		"ptpn": "540-00105-r01-v00.34.04",
		"chaneid": 1627390481,
		"device_control": [{
			"gficlearset": false
		}],
		"producing": true,
		"communicating": true,
		"provisioned": true,
		"operating": true
	}, {
		"part_num": "800-00539-r05",
		"installed": "1479767189",
		"serial_num": "121632033464",
		"device_status": ["envoy.global.ok"],
		"last_rpt_date": "1612559087",
		"admin_state": 1,
		"dev_type": 1,
		"created_date": "1479767189",
		"img_load_date": "1529964383",
		"img_pnum_running": "520-00066-r01-v00.34.09",
		"ptpn": "540-00105-r01-v00.34.04",
		"chaneid": 1627390737,
		"device_control": [{
			"gficlearset": false
		}],
		"producing": true,
		"communicating": true,
		"provisioned": true,
		"operating": true
	}, {
		"part_num": "800-00539-r05",
		"installed": "1479767222",
		"serial_num": "121632035907",
		"device_status": ["envoy.global.ok"],
[... 1368 lines cut ...]
        "phaseMode": "three",
        "phaseCount": 3,
        "meteringStatus": "normal",
        "statusFlags": []
    },
    {
        "eid": 704643584,
        "state": "enabled",
        "measurementType": "net-consumption",
        "phaseMode": "three",
        "phaseCount": 3,
        "meteringStatus": "normal",
        "statusFlags": []
    }
]
```
http://envoy.local/api/v1/production
```
{
  "wattHoursToday": 594,
  "wattHoursSevenDays": 493126,
  "wattHoursLifetime": 84506619,
  "wattsNow": 883
}
```

grzegorz914, 2021-02-05:
Please update and get Feedback

BeauGiles, 2021-02-05:
Thank you. I have updated, and am now getting a different error. It also still isn't showing up in 'Add Accessory'.

```
[06/02/2021, 09:29:36] [Homebridge UI] Starting terminal session
[06/02/2021, 09:29:36] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, state: Online.
[06/02/2021, 09:29:36] [enphaseEnvoy] -------- Envoy-S --------
[06/02/2021, 09:29:36] [enphaseEnvoy] Manufacturer: Enphase
[06/02/2021, 09:29:36] [enphaseEnvoy] Model: Envoy
[06/02/2021, 09:29:36] [enphaseEnvoy] Serial: XXXXXX
[06/02/2021, 09:29:36] [enphaseEnvoy] Firmware: D5.0.49
[06/02/2021, 09:29:36] [enphaseEnvoy] Inverters: 51
[06/02/2021, 09:29:36] [enphaseEnvoy] Encharges: 12
[06/02/2021, 09:29:36] [enphaseEnvoy] Q-Relays: 0
[06/02/2021, 09:29:36] [enphaseEnvoy] Meters: 2
[06/02/2021, 09:29:36] [enphaseEnvoy] Time: 06/02/2021, 09:29:12
[06/02/2021, 09:29:36] [enphaseEnvoy] ----------------------------------
[06/02/2021, 09:29:51] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, update Device state error: TypeError: Cannot read property 'lastReportDate' of undefined
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:1970:61)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
[06/02/2021, 09:30:08] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, state: Online.
[06/02/2021, 09:30:08] [enphaseEnvoy] -------- Envoy-S --------
[06/02/2021, 09:30:08] [enphaseEnvoy] Manufacturer: Enphase
[06/02/2021, 09:30:08] [enphaseEnvoy] Model: Envoy
[06/02/2021, 09:30:08] [enphaseEnvoy] Serial: XXXXX
[06/02/2021, 09:30:08] [enphaseEnvoy] Firmware: D5.0.49
[06/02/2021, 09:30:08] [enphaseEnvoy] Inverters: 51
[06/02/2021, 09:30:08] [enphaseEnvoy] Encharges: 12
[06/02/2021, 09:30:08] [enphaseEnvoy] Q-Relays: 0
[06/02/2021, 09:30:08] [enphaseEnvoy] Meters: 2
[06/02/2021, 09:30:08] [enphaseEnvoy] Time: 06/02/2021, 09:29:44
[06/02/2021, 09:30:08] [enphaseEnvoy] ----------------------------------
[06/02/2021, 09:30:14] [Homebridge UI] Terminal session ended.
[06/02/2021, 09:30:26] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, update Device state error: TypeError: Cannot read property 'lastReportDate' of undefined
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:1970:61)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
[06/02/2021, 09:30:34] [Homebridge UI] Starting terminal session
[06/02/2021, 09:30:37] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, state: Online.
[06/02/2021, 09:30:37] [enphaseEnvoy] -------- Envoy-S --------
[06/02/2021, 09:30:37] [enphaseEnvoy] Manufacturer: Enphase
[06/02/2021, 09:30:37] [enphaseEnvoy] Model: Envoy
[06/02/2021, 09:30:37] [enphaseEnvoy] Serial: XXXXXX
[06/02/2021, 09:30:37] [enphaseEnvoy] Firmware: D5.0.49
[06/02/2021, 09:30:37] [enphaseEnvoy] Inverters: 51
[06/02/2021, 09:30:37] [enphaseEnvoy] Encharges: 12
[06/02/2021, 09:30:37] [enphaseEnvoy] Q-Relays: 0
[06/02/2021, 09:30:37] [enphaseEnvoy] Meters: 2
[06/02/2021, 09:30:37] [enphaseEnvoy] Time: 06/02/2021, 09:30:13
[06/02/2021, 09:30:37] [enphaseEnvoy] ----------------------------------
[06/02/2021, 09:30:49] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, update Device state error: TypeError: Cannot read property 'lastReportDate' of undefined
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:1970:61)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
```

grzegorz914, 2021-02-05:
Try again

BeauGiles, 2021-02-05:
Updated plugin - same error, different line

```
[06/02/2021, 10:21:52] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, state: Online.
[06/02/2021, 10:21:52] [enphaseEnvoy] -------- Envoy-S --------
[06/02/2021, 10:21:52] [enphaseEnvoy] Manufacturer: Enphase
[06/02/2021, 10:21:52] [enphaseEnvoy] Model: Envoy
[06/02/2021, 10:21:52] [enphaseEnvoy] Serial: 121638035839
[06/02/2021, 10:21:52] [enphaseEnvoy] Firmware: D5.0.49
[06/02/2021, 10:21:52] [enphaseEnvoy] Inverters: 51
[06/02/2021, 10:21:52] [enphaseEnvoy] Encharges: 12
[06/02/2021, 10:21:52] [enphaseEnvoy] Q-Relays: 0
[06/02/2021, 10:21:52] [enphaseEnvoy] Meters: 2
[06/02/2021, 10:21:52] [enphaseEnvoy] Time: 06/02/2021, 10:21:28
[06/02/2021, 10:21:52] [enphaseEnvoy] ----------------------------------
[06/02/2021, 10:22:03] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, update Device state error: TypeError: Cannot read property 'lastReportDate' of undefined
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:1974:61)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
[06/02/2021, 10:22:14] [Homebridge UI] Terminal session ended.
[06/02/2021, 10:22:22] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, state: Online.
[06/02/2021, 10:22:22] [enphaseEnvoy] -------- Envoy-S --------
[06/02/2021, 10:22:22] [enphaseEnvoy] Manufacturer: Enphase
[06/02/2021, 10:22:22] [enphaseEnvoy] Model: Envoy
[06/02/2021, 10:22:22] [enphaseEnvoy] Serial: 121638035839
[06/02/2021, 10:22:22] [enphaseEnvoy] Firmware: D5.0.49
[06/02/2021, 10:22:22] [enphaseEnvoy] Inverters: 51
[06/02/2021, 10:22:22] [enphaseEnvoy] Encharges: 12
[06/02/2021, 10:22:22] [enphaseEnvoy] Q-Relays: 0
[06/02/2021, 10:22:22] [enphaseEnvoy] Meters: 2
[06/02/2021, 10:22:22] [enphaseEnvoy] Time: 06/02/2021, 10:21:58
[06/02/2021, 10:22:22] [enphaseEnvoy] ----------------------------------
[06/02/2021, 10:22:27] [Homebridge UI] Starting terminal session
[06/02/2021, 10:22:32] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, update Device state error: TypeError: Cannot read property 'lastReportDate' of undefined
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:1974:61)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
```

grzegorz914, 2021-02-05:
Get this

http://envoy.local/api/v1/production/inverters?

grzegorz914, 2021-02-05:
Try update again

BeauGiles, 2021-02-05:
Here is http://envoy.local/api/v1/production/inverters
```
[
  {
    "serialNumber": "121632033464",
    "lastReportDate": 1612568443,
    "devType": 1,
    "lastReportWatts": 53,
    "maxReportWatts": 270
  },
  {
    "serialNumber": "121632035868",
    "lastReportDate": 1612568450,
    "devType": 1,
    "lastReportWatts": 54,
    "maxReportWatts": 270
  },
  {
    "serialNumber": "121632036671",
    "lastReportDate": 1612568461,
    "devType": 1,
    "lastReportWatts": 53,
    "maxReportWatts": 270
  },
  {
    "serialNumber": "121632035961",
    "lastReportDate": 1612568468,
    "devType": 1,
    "lastReportWatts": 54,
    "maxReportWatts": 270
  },
  {
    "serialNumber": "121714010584",
    "lastReportDate": 1612568174,
    "devType": 1,
    "lastReportWatts": 43,
    "maxReportWatts": 270
  },
  {
    "serialNumber": "121632035907",
    "lastReportDate": 1612568444,
    "devType": 1,
    "lastReportWatts": 51,
    "maxReportWatts": 270
  },
  {
    "serialNumber": "121718026420",
    "lastReportDate": 1612568494,
    "devType": 1,
    "lastReportWatts": 60,
    "maxReportWatts": 270
  },
  {
    "serialNumber": "121718026393",
    "lastReportDate": 1612568502,
    "devType": 1,
    "lastReportWatts": 57,
    "maxReportWatts": 270
  },
  {
    "serialNumber": "121714010415",
    "lastReportDate": 1612568469,
    "devType": 1,
    "lastReportWatts": 59,
    "maxReportWatts": 270
  },
  {
    "serialNumber": "121632033274",
    "lastReportDate": 1612568448,
    "devType": 1,
[... 352 lines cut ...]
  },
  {
    "serialNumber": "121632035883",
    "lastReportDate": 1612568438,
    "devType": 1,
    "lastReportWatts": 52,
    "maxReportWatts": 270
  },
  {
    "serialNumber": "121632035809",
    "lastReportDate": 1612568454,
    "devType": 1,
    "lastReportWatts": 54,
    "maxReportWatts": 269
  },
  {
    "serialNumber": "121632033465",
    "lastReportDate": 1612568467,
    "devType": 1,
    "lastReportWatts": 51,
    "maxReportWatts": 270
  }
]

```

BeauGiles, 2021-02-05:
Updated again. :-) 

Now I get

```
[06/02/2021, 10:48:57] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, update Device state error: TypeError: Cannot read property 'devType' of undefined
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:1976:51)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
[06/02/2021, 10:49:14] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, state: Online.
[06/02/2021, 10:49:14] [enphaseEnvoy] -------- Envoy-S --------
[06/02/2021, 10:49:14] [enphaseEnvoy] Manufacturer: Enphase
[06/02/2021, 10:49:14] [enphaseEnvoy] Model: Envoy
[06/02/2021, 10:49:14] [enphaseEnvoy] Serial: XXXXXXX
[06/02/2021, 10:49:14] [enphaseEnvoy] Firmware: D5.0.49
[06/02/2021, 10:49:14] [enphaseEnvoy] Inverters: 51
[06/02/2021, 10:49:14] [enphaseEnvoy] Encharges: 12
[06/02/2021, 10:49:14] [enphaseEnvoy] Q-Relays: 0
[06/02/2021, 10:49:14] [enphaseEnvoy] Meters: 2
[06/02/2021, 10:49:14] [enphaseEnvoy] Time: 06/02/2021, 10:48:49
[06/02/2021, 10:49:14] [enphaseEnvoy] ----------------------------------
[06/02/2021, 10:49:23] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, update Device state error: TypeError: Cannot read property 'devType' of undefined
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:1976:51)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
```

grzegorz914, 2021-02-05:
Hmm,

on my site this is read correct i will look on this toomorow, now is 0:53 night

BeauGiles, 2021-02-08:
Thank you. I just updated the plugin, including just to the latest 4.3.2 beta 1, entered only the IP address, 'installer' and password in the config, and I'm still getting errors, though different again.

```
[08/02/2021, 21:42:46] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, update Device state error: TypeError: Cannot read property 'lastReportDate' of undefined
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:2178:64)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
[08/02/2021, 21:43:05] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, state: Online.
[08/02/2021, 21:43:05] [enphaseEnvoy] -------- Envoy-S --------
[08/02/2021, 21:43:05] [enphaseEnvoy] Manufacturer: Enphase
[08/02/2021, 21:43:05] [enphaseEnvoy] Model: Envoy
[08/02/2021, 21:43:05] [enphaseEnvoy] Serial: XXXXX
[08/02/2021, 21:43:05] [enphaseEnvoy] Firmware: D5.0.49
[08/02/2021, 21:43:05] [enphaseEnvoy] Inverters: 51
[08/02/2021, 21:43:05] [enphaseEnvoy] Encharges: 12
[08/02/2021, 21:43:05] [enphaseEnvoy] Q-Relays: 0
[08/02/2021, 21:43:05] [enphaseEnvoy] Meters: 2
[08/02/2021, 21:43:05] [enphaseEnvoy] Time: 08/02/2021, 21:42:40
[08/02/2021, 21:43:05] [enphaseEnvoy] ----------------------------------
[08/02/2021, 21:43:15] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, update Device state error: TypeError: Cannot read property 'lastReportDate' of undefined
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:2178:64)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
[08/02/2021, 21:43:34] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, state: Online.
[08/02/2021, 21:43:34] [enphaseEnvoy] -------- Envoy-S --------
[08/02/2021, 21:43:34] [enphaseEnvoy] Manufacturer: Enphase
[08/02/2021, 21:43:34] [enphaseEnvoy] Model: Envoy
[08/02/2021, 21:43:34] [enphaseEnvoy] Serial: XXXXX
[08/02/2021, 21:43:34] [enphaseEnvoy] Firmware: D5.0.49
[08/02/2021, 21:43:34] [enphaseEnvoy] Inverters: 51
[08/02/2021, 21:43:34] [enphaseEnvoy] Encharges: 12
[08/02/2021, 21:43:34] [enphaseEnvoy] Q-Relays: 0
[08/02/2021, 21:43:34] [enphaseEnvoy] Meters: 2
[08/02/2021, 21:43:34] [enphaseEnvoy] Time: 08/02/2021, 21:43:08
[08/02/2021, 21:43:34] [enphaseEnvoy] ----------------------------------
[08/02/2021, 21:43:44] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, update Device state error: TypeError: Cannot read property 'lastReportDate' of undefined
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:2178:64)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
[08/02/2021, 21:44:04] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, state: Online.
[08/02/2021, 21:44:04] [enphaseEnvoy] -------- Envoy-S --------
[08/02/2021, 21:44:04] [enphaseEnvoy] Manufacturer: Enphase
[08/02/2021, 21:44:04] [enphaseEnvoy] Model: Envoy
[08/02/2021, 21:44:04] [enphaseEnvoy] Serial: XXXXX
[08/02/2021, 21:44:04] [enphaseEnvoy] Firmware: D5.0.49
[08/02/2021, 21:44:04] [enphaseEnvoy] Inverters: 51
[08/02/2021, 21:44:04] [enphaseEnvoy] Encharges: 12
[08/02/2021, 21:44:04] [enphaseEnvoy] Q-Relays: 0
[08/02/2021, 21:44:04] [enphaseEnvoy] Meters: 2
[08/02/2021, 21:44:04] [enphaseEnvoy] Time: 08/02/2021, 21:43:35
[08/02/2021, 21:44:04] [enphaseEnvoy] ----------------------------------
[08/02/2021, 21:44:13] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, update Device state error: TypeError: Cannot read property 'lastReportDate' of undefined
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:2178:64)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
```

grzegorz914, 2021-02-08:
Do not install beta please, use release 4.3.1

BeauGiles, 2021-02-08:
Apologies, the Homebridge UI suggested updating to it. Downgraded back to 4.3.1

```
[08/02/2021, 22:01:15] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, update Device state error: TypeError: Cannot read property 'lastReportDate' of undefined
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:2178:64)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
```

grzegorz914, 2021-02-08:
please install beta8 and get the log output

BeauGiles, 2021-02-08:
Thank you.

Latest log from beta 8 is

`[08/02/2021, 22:35:58] [enphaseEnvoy] invertersi {
  status: 401,
  error: '',
  info: 'Authentication required',
  moreInfo: ''
}
[08/02/2021, 22:35:58] [enphaseEnvoy] indexActiveInverter -1
[08/02/2021, 22:35:58] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, update Device state error: TypeError: Cannot read property 'lastReportDate' of undefined
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:2101:64)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
[08/02/2021, 22:36:16] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, state: Online.
[08/02/2021, 22:36:17] [enphaseEnvoy] -------- Envoy-S --------
[08/02/2021, 22:36:17] [enphaseEnvoy] Manufacturer: Enphase
[08/02/2021, 22:36:17] [enphaseEnvoy] Model: Envoy
[08/02/2021, 22:36:17] [enphaseEnvoy] Serial: XXXXXX
[08/02/2021, 22:36:17] [enphaseEnvoy] Firmware: D5.0.49
[08/02/2021, 22:36:17] [enphaseEnvoy] Inverters: 51
[08/02/2021, 22:36:17] [enphaseEnvoy] Encharges: 12
[08/02/2021, 22:36:17] [enphaseEnvoy] Q-Relays: 0
[08/02/2021, 22:36:17] [enphaseEnvoy] Meters: 2
[08/02/2021, 22:36:17] [enphaseEnvoy] Time: 08/02/2021, 22:35:48
[08/02/2021, 22:36:17] [enphaseEnvoy] ----------------------------------`

Just to make sure I have. the right username/password, I went directly to http://10.0.1.243/api/v1/production/inverters and was able to log in as 'installer' and the password.

grzegorz914, 2021-02-08:
{ status: 401, error: '', info: 'Authentication required', moreInfo: '' }  here the problem

install beta10 and get the log

BeauGiles, 2021-02-08:
Thank you. The default password (last 6 digits of serial number) is being used.

I have changed this password. (For security as it was available on the internet)

Is it still possible to override the password used for the 'envoy' user? The readme says it was removed a few versions ago?

grzegorz914, 2021-02-08:
Ok, i will ad this possibility

grzegorz914, 2021-02-08:
try beta 13

BeauGiles, 2021-02-08:
Thank you. Just tried - I was able to manually set the 'envoy' password. No more Authentication error.

Different error now...  Sorry my setup is being painful

```
[08/02/2021, 23:11:55] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, update Device state error: TypeError: Cannot read property 'getCharacteristic' of undefined
    at envoyDevice.prepareEnphaseService (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:2750:43)
    at envoyDevice.prepareAccessory (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:2143:10)
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:2124:12)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
[08/02/2021, 23:12:15] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, state: Online.
[08/02/2021, 23:12:15] [enphaseEnvoy] -------- Envoy-S --------
[08/02/2021, 23:12:15] [enphaseEnvoy] Manufacturer: Enphase
[08/02/2021, 23:12:15] [enphaseEnvoy] Model: Envoy
[08/02/2021, 23:12:15] [enphaseEnvoy] Serial: XXXX
[08/02/2021, 23:12:15] [enphaseEnvoy] Firmware: D5.0.49
[08/02/2021, 23:12:15] [enphaseEnvoy] Inverters: 51
[08/02/2021, 23:12:15] [enphaseEnvoy] Encharges: 12
[08/02/2021, 23:12:15] [enphaseEnvoy] Q-Relays: 0
[08/02/2021, 23:12:15] [enphaseEnvoy] Meters: 2
[08/02/2021, 23:12:15] [enphaseEnvoy] Time: 08/02/2021, 23:11:49
[08/02/2021, 23:12:15] [enphaseEnvoy] ----------------------------------
[08/02/2021, 23:12:25] [enphaseEnvoy] Device: 10.0.1.243 Envoy-S, update Device state error: TypeError: Cannot read property 'getCharacteristic' of undefined
    at envoyDevice.prepareEnphaseService (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:2750:43)
    at envoyDevice.prepareAccessory (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:2143:10)
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:2124:12)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5), state: Offline
```

grzegorz914, 2021-02-08:
OK will fixed soon

grzegorz914, 2021-02-08:
try beta 15

grzegorz914, 2021-02-08:
test in progress

grzegorz914, 2021-02-12:
Fixed

commit 3a72088f (2021-02-07): bump to 4.3.0, fix #11, #12, and many improvements

## [4.3.0] - (07.02.2021)
## Changs
- added more characteristics for encharges
- added characteristics for Current, Voltage and Power Factor
- fixed reported bugs
closed 2021-02-12 by grzegorz914