grzegorz914/homebridge-enphase-envoy #12: Zero production on inverters and error on first 2

jr7216, 2021-02-06:
Version 4.2.39
[image]

grzegorz914, 2021-02-06:
I will fixed this all tomorrow, be patient

grzegorz914, 2021-02-07:
please get the output from

http://envoy.local/api/v1/production/inverters?locale=en

jr7216, 2021-02-07:
Unfortunately its snowing. In have zero production at the moment.

  {
    "serialNumber": "122022001491",
    "lastReportDate": 1612691674,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 152
  },
  {
    "serialNumber": "122021062205",
    "lastReportDate": 1612692350,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 158
  },
  {
    "serialNumber": "122022001349",
    "lastReportDate": 1612692723,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 168
  },
  {
    "serialNumber": "122022000357",
    "lastReportDate": 1612692324,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 166
  },
  {
    "serialNumber": "122021058482",
    "lastReportDate": 1612690808,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 172
  },
  {
    "serialNumber": "122022001104",
    "lastReportDate": 1612692701,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 140
  },
  {
    "serialNumber": "122022001489",
    "lastReportDate": 1612628783,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 143
  },
  {
    "serialNumber": "122022001352",
    "lastReportDate": 1612628789,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 166
  },
  {
    "serialNumber": "122022001301",
    "lastReportDate": 1612692771,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 167
  },
  {
    "serialNumber": "122021061909",
    "lastReportDate": 1612692755,
    "devType": 1,
    "lastReportWatts": -2,
    "maxReportWatts": 158
  },
  {
    "serialNumber": "122022001553",
    "lastReportDate": 1612628788,
    "devType": 1,
    "lastReportWatts": 0,
    "maxReportWatts": 145
  }
]

closed 2021-02-07 by grzegorz914