grzegorz914/homebridge-enphase-envoy #14: homebridge-enphase-envoy v4.3.2-beta119 - TypeError: Cannot read property 'activeCount' of undefined

hoefs, 2021-02-10:
[2/10/2021, 3:21:22 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, state: Online.
[2/10/2021, 3:21:22 PM] [enphaseEnvoy] -------- Envoy-S --------
[2/10/2021, 3:21:22 PM] [enphaseEnvoy] Manufacturer: Enphase
[2/10/2021, 3:21:22 PM] [enphaseEnvoy] Model: Envoy-S
[2/10/2021, 3:21:22 PM] [enphaseEnvoy] Meters: 0
[2/10/2021, 3:21:22 PM] [enphaseEnvoy] Q-Relays: 1
[2/10/2021, 3:21:22 PM] [enphaseEnvoy] Encharges: 0
[2/10/2021, 3:21:22 PM] [enphaseEnvoy] Inverters: 14
[2/10/2021, 3:21:22 PM] [enphaseEnvoy] Firmware: R4.10.35
[2/10/2021, 3:21:22 PM] [enphaseEnvoy] SerialNr: 121921106049
[2/10/2021, 3:21:22 PM] [enphaseEnvoy] Time: 2/10/2021, 3:21:19 PM
[2/10/2021, 3:21:22 PM] [enphaseEnvoy] ----------------------------------
[2/10/2021, 3:21:22 PM] [Occupancy Sensor] 0 monitored device(s) found. Accessory is in mode "any" so NOT triggering occupancy.
[2/10/2021, 3:21:27 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, update Device state error: TypeError: Cannot read property 'activeCount' of undefined
    at envoyDevice.updateDeviceState (/homebridge/node_modules/homebridge-enphase-envoy/index.js:1363:69)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)

hoefs, 2021-02-10:
Seems it is always in the loop:

```
[2/10/2021, 3:26:36 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, state: Online.
[2/10/2021, 3:26:36 PM] [enphaseEnvoy] Device 192.168.2.25 Envoy-S, get device status data inventory [ [Object], [Object], [Object] ] info: <?xml version='1.0' encoding='UTF-8'?>
<envoy_info>
  <time>1612967194</time>
  <device>
    <sn>121921106049</sn>
    <pn>800-00553-r02</pn>
    <software>R4.10.35</software>
    <euaid>4c8675</euaid>
    <seqnum>0</seqnum>
    <apiver>1</apiver>
    <imeter>false</imeter>
  </device>
  <package name='rootfs'>
    <pn>500-00001-r01</pn>
    <version>02.00.00</version>
    <build>945</build>
  </package>
  <package name='kernel'>
    <pn>500-00011-r01</pn>
    <version>04.00.00</version>
    <build>5bb754</build>
  </package>
  <package name='boot'>
    <pn>590-00018-r01</pn>
    <version>02.00.01</version>
    <build>426697</build>
  </package>
  <package name='app'>
    <pn>500-00002-r01</pn>
    <version>04.10.35</version>
    <build>6ed292</build>
  </package>
  <package name='devimg'>
    <pn>500-00005-r01</pn>
    <version>01.02.186</version>
    <build>d0d70f</build>
  </package>
  <package name='geo'>
    <pn>500-00008-r01</pn>
    <version>02.01.22</version>
    <build>06e201</build>
  </package>
  <package name='backbone'>
    <pn>500-00010-r01</pn>
    <version>04.10.25</version>
    <build>7b7de5</build>
  </package>
  <package name='meter'>
    <pn>500-00013-r01</pn>
    <version>03.02.07</version>
    <build>4c9d48</build>
  </package>
  <package name='agf'>
    <pn>500-00012-r01</pn>
    <version>02.02.00</version>
    <build>c00a8f</build>
  </package>
  <package name='security'>
    <pn>500-00016-r01</pn>
    <version>02.00.00</version>
    <build>54a6dc</build>
  </package>
  <build_info>
    <build_time_gmt>1542157882</build_time_gmt>
    <build_id>release-4.10.x-103-Nov-12-18-18:25:06</build_id>
  </build_info>
[... 534 lines cut ...]
  request: [ClientRequest],
  data: [Array]
}
[2/10/2021, 3:27:12 PM] [enphaseEnvoy] Debug pcuCommCheck: {
  '121932043894': 4,
  '121932016547': 4,
  '121932034199': 3,
  '121932033746': 4,
  '121932016549': 3,
  '121932039512': 3,
  '121932038960': 3,
  '121932043380': 4,
  '121932039459': 4,
  '121910052727': 3,
  '121932039006': 4,
  '121931158966': 3,
  '121932035604': 3,
  '121931157713': 4,
  '121932039565': 3
}
[2/10/2021, 3:27:12 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, update Device state error: TypeError: Cannot read property 'activeCount' of undefined
    at envoyDevice.updateDeviceState (/homebridge/node_modules/homebridge-enphase-envoy/index.js:1363:69)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)
```

grzegorz914, 2021-02-10:
Try release 4.4.0

grzegorz914, 2021-02-10:
Can u get me output from:
http://envoy.local/production.json

grzegorz914, 2021-02-10:
Looks like U have Envoy-S standard, this model does not have any meters inputs. Fixed.

closed 2021-02-10 by grzegorz914

hoefs, 2021-02-10:
> Looks like U have Envoy-S standard, this model does not have any meters inputs. Fixed.

Indeed envoy-s...

Performed the upgrade but got another error now:
[image]

grzegorz914, 2021-02-10:
Please create manually in 
/homebridge/enphaseEnvoy/productionPowerMax_192168225 
file and restart Homebridge, then get the feedback

hoefs, 2021-02-10:
> Please create manually in
> /homebridge/enphaseEnvoy/productionPowerMax_192168225
> file and restart Homebridge, then get the feedback

Do I need to create I manually? I have never had to do it. I know how to do it, but it is not user friendly.

grzegorz914, 2021-02-10:
On which devices U run Homebridge? RPI or

hoefs, 2021-02-10:
> On which devices U run Homebridge? RPI or

Docker on synology 

I have created the file. Seems like another (almost same name was existing; I have created productionPowerMax_xxxxxxxx and powerProductionMax_xxxxxxx was already existing)

[image]

grzegorz914, 2021-02-10:
OK. fixed in 4.4.4

commit 872a7005 (2021-02-10): fix #14