grzegorz914/homebridge-enphase-envoy #15: v4.4.16-beta-14  TypeError: Cannot read property 'res' of undefined

jcgillespie, 2021-02-15:
Getting a TypeError with v4.4.16-beta.14 (2021-02-15)

```
[15/02/2021, 16:10:38] [enphaseEnvoy] -------- Envoy-S --------
[15/02/2021, 16:10:38] [enphaseEnvoy] Manufacturer: Enphase
[15/02/2021, 16:10:38] [enphaseEnvoy] Model: Envoy
[15/02/2021, 16:10:38] [enphaseEnvoy] Meters: 2
[15/02/2021, 16:10:38] [enphaseEnvoy] Q-Relays: 0
[15/02/2021, 16:10:38] [enphaseEnvoy] Encharges: 0
[15/02/2021, 16:10:38] [enphaseEnvoy] Inverters: 22
[15/02/2021, 16:10:38] [enphaseEnvoy] Firmware: D5.0.49
[15/02/2021, 16:10:38] [enphaseEnvoy] SerialNr: <snipped>
[15/02/2021, 16:10:38] [enphaseEnvoy] Time: 15/02/2021, 16:10:33
[15/02/2021, 16:10:38] [enphaseEnvoy] ----------------------------------
[15/02/2021, 16:10:44] [enphaseEnvoy] Device: 192.168.9.104 Envoy-S, update Device state error: TypeError: Cannot read property 'res' of undefined
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:2094:50)
```

chris-araman, 2021-02-16:
I'm seeing the same error every 30s. Looking over the code, it seems this might happen when `installerPasswd` is not configured.

@grzegorz914, this configuration property is marked "optional", so I wouldn't expect a configuration without `installerPasswd` to generate errors. Based on the info you reference [here](https://github.com/grzegorz914/homebridge-enphase-envoy/commit/2b3bb8c50386992aac4de4fb6afa30c5c955972b#diff-be1695b1e63a508d59982601f9e1fb7f58247deecb1e427adb77bcad758ae5e5R57), It looks like the password could be derived automatically from the serial number. Would you be open to a pull request that implements that functionality?

grzegorz914, 2021-02-16:
installerPassword is other one as envoyPassword and can not be generated automatically in my code right now.
The typeError will be fixed today.

closed 2021-02-16 by grzegorz914

grzegorz914, 2021-02-16:
@chris-araman 
Pull request and help in development highly appreciated.

grzegorz914, 2021-02-16:
Please after test get feedback about functionality of this beta version.

chris-araman, 2021-02-16:
The fix works for me. Thanks for the quick response!

grzegorz914, 2021-02-16:
Nice, if U add installer password U will be able to read communications level for every micro inverters, Q-Relay and Encharges. if this password is empty or wrong the level will be always 0 but again U can read summary level of all devices in Envoy section.

commit 4375a0f0 (2021-02-16): fix #15