grzegorz914/homebridge-enphase-envoy #16: EHOSTUNREACH - Any ideas?

jcgillespie, 2021-02-17:
I'm getting the below error.  I'm running Homebridge on a Raspberry Pi, with both `wlan0` and `eth0` connected. `eth0` is on subnet 5, which is the main network. `wlan0` is on subnet 9, which is an IoT only network. As near as I can tell, HomeBridge is natively able to handle this setup, but for some reason I'm not able to successfully connect.

The Envoy and Homebridge interfaces all have static IP addresses.  

When I browse to the IP address directly from another device on that IoT subnet, it comes right up.

**Logs**
This shows up in the log every 30 seconds, per the refresh interval.
```
Device: 192.168.9.104 Envoy-S, requesting devices info eror: Error: connect EHOSTUNREACH 192.168.9.104:80
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1146:16) {
  errno: -113,
  code: 'EHOSTUNREACH',
  syscall: 'connect',
  address: '192.168.9.104',
  port: 80,
  config: [Object],
  request: [Writable],
  response: undefined,
  isAxiosError: true,
  toJSON: [Function: toJSON]
}, state: Offline trying to reconnect.
```

**Config**
```
{
            "devices": [
                {
                    "name": "Envoy-S",
                    "host": "192.168.9.104",
                    "refreshInterval": 30,
                    "disableLogInfo": false,
                    "envoyUser": "REDACTED",
                    "envoyPasswd": "REDACTED",
                    "installerUser": "installer",
                    "installerPasswd": "REDACTED",
                    "enchargeStorageOffset": 0,
                    "powerProductionMaxDetected": 7100,
                    "energyProductionLifetimeOffset": 0,
                    "powerConsumptionTotalMaxDetected": 10000,
                    "energyConsumptionTotalLifetimeOffset": 0,
                    "powerConsumptionNetMaxDetected": 10000,
                    "energyConsumptionNetLifetimeOffset": 0,
                    "manufacturer": "Enphase",
                    "modelName": "Envoy-S"
                }
            ],
            "platform": "enphaseEnvoy"
        },
```

**Environment**
    Node.js Version: v14.15.5
    NPM Version: v6.14.11
    Homebridge Version: v1.1.7 
    Operating System: Raspbian GNU/Linux Buster (10)

grzegorz914, 2021-02-17:
Looks like Your Envoy is not reached on the network, please leave the address IP empty in config and restart homebridge.

grzegorz914, 2021-02-17:
Please also set correct interface in homebridge settings.

jcgillespie, 2021-02-17:
I've tried the empty config, which defaults to to `envoy.local`. That DNS doesn't route on the IoT subnetwork, intentionally.
The Envoy is on the network, and I can reach it directly via IP address using other devices on that network.

I've tried both Homebridge version 1.1.7 and the latest beta (1.3.0-beta.61).  In 1.3.0 beta, I've tried both Bonjour and Ciao.

Interfaces are both selected.

[image]

grzegorz914, 2021-02-17:
Please try disable the eth0 interface and try again, may be some blocked rpi to communicate with Envoy, can you Ping your Envoy from RPI?

audientsound, 2021-02-23:
can the plugin work solely through the user account on enphase's servers or does it need a direct local connection to the Envoy? I can't leave local IP blank in the plugin settings. I did try deleting the local ip in the config file, and entered my enphase account name/password, but that didnt fly.

```
Device: envoy.local Envoy-S, requesting devices info eror: Error: getaddrinfo ENOTFOUND envoy.local
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:67:26) {
  errno: -3008,
  code: 'ENOTFOUND',
  syscall: 'getaddrinfo',
  hostname: 'envoy.local',
  config: [Object],
  request: [Writable],
  response: undefined,
  isAxiosError: true,
  toJSON: [Function: toJSON]
}, state: Offline trying to reconnect.
```

grzegorz914, 2021-02-23:
this plugin works with local connection realtime data, no need Enphase account. 
Just add your Envoy IP address in hostname all other U can leave empty

audientsound, 2021-02-23:
I dont have a local connection to the envoy - it's not on the same network as homebridge. I currently get its status via the iOS App, which logs into emphase's servers.
I'll to try to set up some kind of tunnel for it. thanks for replying.

grzegorz914, 2021-02-23:
U can make VPN connection or Redirect port and use external IP address

audientsound, 2021-02-23:
exactly. thanks.

closed 2021-02-24 by grzegorz914