grzegorz914/homebridge-enphase-envoy #17: homebridge-enphase-envoy v4.4.16-beta.15 - [homebridge-enphase-envoy] Error:

hoefs, 2021-02-20:
There errors are shown in homebridge in debug mode in version homebridge-enphase-envoy v4.4.16-beta.15

```
} productionCT: { production: [Array], storage: [Array] }
[2/20/2021, 10:41:20 AM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Tariff': characteristic was supplied illegal value: undefined! This might throw errors in the future! See https://git.io/JtMGR for more info.
[2/20/2021, 10:41:20 AM] [homebridge-enphase-envoy] Error: 
    at Characteristic.enphaseEnvoyTariff.Characteristic.characteristicWarning (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2063:79)
    at Characteristic.enphaseEnvoyTariff.Characteristic.validateUserInput (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1839:12)
    at Characteristic.enphaseEnvoyTariff.Characteristic.updateValue (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1338:20)
    at Service.enphaseEnvoy.Service.updateCharacteristic (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Service.ts:465:35)
    at envoyDevice.updateDeviceState (/homebridge/node_modules/homebridge-enphase-envoy/index.js:1466:34)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)
[2/20/2021, 10:41:20 AM] [enphaseEnvoy] Debug production inverters: [
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object]
]
```

grzegorz914, 2021-02-20:
Please get me Your output from:
http://envoy.local/home.json

hoefs, 2021-02-20:
update to homebridge-enphase-envoy v4.4.16

I can capture on Monday when I am back home

```
[2/20/2021, 7:36:46 PM] [enphaseEnvoy] Debug home: {
  software_build_epoch: 1542074240,
  is_nonvoy: false,
  db_size: '27 MB',
  db_percent_full: '7',
  timezone: 'Europe/Brussels',
  current_date: '02/20/2021',
  current_time: '19:36',
  network: [Object],
  tariff: 'none',
  comm: [Object],
  alerts: [],
  update_status: 'not-satisfied'
}, inventory: [ [Object], [Object], [Object] ], meters: [], production: {
  wattHoursToday: 12754,
  wattHoursSevenDays: 51900,
  wattHoursLifetime: 4918676,
  wattsNow: 0
} productionCT: { production: [Array], storage: [Array] }
```

```
[2/20/2021, 7:36:49 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Tariff': characteristic was supplied illegal value: undefined! This might throw errors in the future! See https://git.io/JtMGR for more info.
[2/20/2021, 7:36:49 PM] [homebridge-enphase-envoy] Error: 
    at Characteristic.enphaseEnvoyTariff.Characteristic.characteristicWarning (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2063:79)
    at Characteristic.enphaseEnvoyTariff.Characteristic.validateUserInput (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1839:12)
    at /usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1490:24
    at /usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/util/once.ts:9:18
    at Characteristic.enphaseEnvoyTariff.<anonymous> (/homebridge/node_modules/homebridge-enphase-envoy/index.js:2348:11)
    at Characteristic.enphaseEnvoyTariff.emit (events.js:315:20)
    at /usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1466:14
    at new Promise (<anonymous>)
    at Characteristic.enphaseEnvoyTariff.<anonymous> (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1464:12)
    at step (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:143:27)
    at Object.next (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:124:57)
    at /usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:117:75
    at new Promise (<anonymous>)
    at Object.__awaiter (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:113:16)
    at Characteristic.enphaseEnvoyTariff.Characteristic.handleGetRequest (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/dist/lib/Characteristic.js:632:24)
    at Characteristic.enphaseEnvoyTariff.<anonymous> (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2127:22)
    at step (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:143:27)
    at Object.next (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:124:57)
    at /usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:117:75
    at new Promise (<anonymous>)
    at Object.__awaiter (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:113:16)
    at Characteristic.enphaseEnvoyTariff.Characteristic.toHAP (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/dist/lib/Characteristic.js:1355:24)
```

commit 2b7abbc8 (2021-02-20): fix #17
closed 2021-02-20 by grzegorz914