grzegorz914/homebridge-enphase-envoy #18: getting warning

GentleJan, 2021-02-22:
[2/22/2021, 7:02:54 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Tariff': characteristic was supplied illegal value: undefined! This might throw errors in the future! See https://git.io/JtMGR for more info.

plugin versie 4.4.19
newest 1.3.0 Homebridge
node.js 14.15.5
npm 6.14.11

grzegorz914, 2021-02-22:
This is already fixed with #17

grzegorz914, 2021-02-22:
Make update to latest version.

grzegorz914, 2021-02-22:
Please get Your output from
http://envoy.local/home.json

GentleJan, 2021-02-22:
{"software_build_epoch":1607591607,"is_nonvoy":false,"db_size":"19 MB","db_percent_full":"5","timezone":"Europe/Amsterdam","current_date":"02/22/2021","current_time":"21:24","network":{"web_comm":true,"ever_reported_to_enlighten":true,"last_enlighten_report_time":1614024603,"primary_interface":"wlan0","interfaces":[{"type":"ethernet","interface":"eth0","mac":"00:1D:C0:6C:AA:AA","dhcp":true,"ip":"169.254.120.1","signal_strength":0,"signal_strength_max":1,"carrier":false},{"signal_strength":2,"signal_strength_max":5,"type":"wifi","interface":"wlan0","mac":"40:06:A0:8A:99:99","dhcp":true,"ip":"192.168.1.11","carrier":true,"supported":true,"present":true,"configured":true,"status":"connected"}]},"tariff":"time_of_use","comm":{"num":0,"level":0,"pcu":{"num":0,"level":0},"acb":{"num":0,"level":0},"nsrb":{"num":0,"level":0}},"alerts":[],"update_status":"not-satisfied"}

Mogelijk opzettelijk kort gehouden ivm verzending vanaf mijn iPhone

http://envoy.local/home.json

grzegorz914, 2021-02-22:
Fixed

closed 2021-02-22 by grzegorz914

GentleJan, 2021-02-22:
Solved! Thanks!

Mogelijk opzettelijk kort gehouden ivm verzending vanaf mijn iPhone

commit b1567e4a (2021-02-22): fix #18