grzegorz914/homebridge-enphase-envoy #19: homebridge-enphase-envoy v4.5.0

hoefs, 2021-02-23:
Hi,

Can you fix this?

```
[2/23/2021, 10:53:19 AM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Power max': characteristic was expected valid number and received NaN. See https://git.io/JtMGR for more info.
[2/23/2021, 10:53:19 AM] [homebridge-enphase-envoy] Error: 
    at Characteristic.enphasePowerMax.Characteristic.characteristicWarning (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2164:105)
    at Characteristic.enphasePowerMax.Characteristic.validateUserInput (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1966:16)
    at Characteristic.enphasePowerMax.<anonymous> (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1442:24)
    at step (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:143:27)
    at Object.next (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:124:57)
    at fulfilled (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:114:62)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)
```

thanks

grzegorz914, 2021-02-23:
Hmm, I don't get this warning, im tested on hb 1.3.0 and 1.3.1 and no warning here.
Can U explain more detail.

hoefs, 2021-02-23:
> Hmm, I don't get this warning, im tested on hb 1.3.0 and 1.3.1 and no warning here.
> Can U explain more detail.

Hi. 
I am using 1.3.1 and the lastest homebridge-enphase-envoy v4.5.0

These errors are apearing every minute.

This is my config:
```
        {
            "devices": [
                {
                    "name": "Envoy-S",
                    "host": "192.168.2.25",
                    "refreshInterval": 30,
                    "disableLogInfo": false,
                    "envoyUser": "envoy",
                    "envoyPasswd": "xxxxxx",
                    "installerUser": "installer",
                    "installerPasswd": "xxxxxxxx",
                    "enchargeStorageOffset": 0,
                    "powerProductionMaxDetected": 3400,
                    "energyProductionLifetimeOffset": 0,
                    "powerConsumptionTotalMaxDetected": 0,
                    "energyConsumptionTotalLifetimeOffset": 0,
                    "powerConsumptionNetMaxDetected": 0,
                    "energyConsumptionNetLifetimeOffset": 0,
                    "manufacturer": "Enphase",
                    "modelName": "Envoy-S"
                }
            ],
            "platform": "enphaseEnvoy"
        }
```
```
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932033746 last power: 102 W
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932033746 max power: 213 W
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932033746 producing: Yes
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932033746 communicating: Yes
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932033746 provisioned: Yes
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932033746 operating: Yes
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932033746 comm. level: 0
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932033746 status: Normal
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932033746 firmware: 520-00082-r01-v04.27.04
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932033746 last report: 2/23/2021, 11:34:15 AM
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932035604 last power: 103 W
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932035604 max power: 215 W
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932035604 producing: Yes
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932035604 communicating: Yes
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932035604 provisioned: Yes
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932035604 operating: Yes
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932035604 comm. level: 0
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932035604 status: Normal
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932035604 firmware: 520-00082-r01-v04.27.04
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932035604 last report: 2/23/2021, 11:34:17 AM
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121931157713 last power: 102 W
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121931157713 max power: 213 W
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121931157713 producing: Yes
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121931157713 communicating: Yes
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121931157713 provisioned: Yes
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121931157713 operating: Yes
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121931157713 comm. level: 0
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121931157713 status: Normal
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121931157713 firmware: 520-00082-r01-v04.27.04
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121931157713 last report: 2/23/2021, 11:34:19 AM
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932034199 last power: 101 W
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932034199 max power: 213 W
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932034199 producing: Yes
[2/23/2021, 11:35:57 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, microinverter: 121932034199 communicating: Yes
[... 382 lines cut ...]
  alerts: [],
  update_status: 'satisfied'
}, inventory: [ [Object], [Object], [Object] ], meters: [], production: {
  wattHoursToday: 1523,
  wattHoursSevenDays: 60288,
  wattHoursLifetime: 4943304,
  wattsNow: 1452
[... cut ...]
    at Characteristic.enphasePowerMax.Characteristic.updateValue (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1365:20)
    at Service.enphasePowerEnergyMeter.Service.updateCharacteristic (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Service.ts:465:35)
    at envoyDevice.updateDeviceState (/homebridge/node_modules/homebridge-enphase-envoy/index.js:1790:39)
[2/23/2021, 11:37:37 AM] [enphaseEnvoy] Debug production inverters: [
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object]
]
```

grzegorz914, 2021-02-23:
Please check this file in location 
./Homebridge/evphaseEnvoy/productionPowerMax_192168225
look what value is inside?

hoefs, 2021-02-23:
> ./Homebridge/evphaseEnvoy/productionPowerMax_192168225

```
/homebridge/enphaseEnvoy # ls
consumptionNetPowerMax_192168225  consumptionTotalPowerMax_192168225  devInfo_192168225  maxPowerProduction_192168225  maxPower_192168225  powerProductionMax_192168225  productionPowerMax_192168225
/homebridge/enphaseEnvoy # cat productionPowerMax_192168225 
/homebridge/enphaseEnvoy # 
```

grzegorz914, 2021-02-23:
powerProduction is old one not used any more, please install 4.5.1-beta.1 and get feedback in debug mode.

hoefs, 2021-02-23:
> powerProduction is old one not used any more, please install 4.5.1-beta.1 and get feedback in debug mode.

same issue after update to 4.5.1-beta.1

```
[2/23/2021, 12:03:55 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Power max': characteristic was expected valid number and received NaN. See https://git.io/JtMGR for more info.
[2/23/2021, 12:03:55 PM] [homebridge-enphase-envoy] Error: 
    at Characteristic.enphasePowerMax.Characteristic.characteristicWarning (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2164:105)
    at Characteristic.enphasePowerMax.Characteristic.validateUserInput (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1966:16)
    at Characteristic.enphasePowerMax.<anonymous> (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1442:24)
    at step (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:143:27)
    at Object.next (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:124:57)
    at fulfilled (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:114:62)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)
```

I have deleted the file productionPowerMax_192168225 and restarted HB and issue is gone:

```
/homebridge/enphaseEnvoy # cat productionPowerMax_192168225
1.956
```

grzegorz914, 2021-02-23:
I need the debug loog

hoefs, 2021-02-23:
> I need the debug loog

I enabled the -D but where can I find that log? (using homebridge-docker)

grzegorz914, 2021-02-23:
in homebridge console

hoefs, 2021-02-23:
> in homebridge console

This is the only log:

[homebridge.log](https://github.com/grzegorz914/homebridge-enphase-envoy/files/6028568/homebridge.log)

grzegorz914, 2021-02-23:
I see the error not more appear in Your loog , looks like was problem with read data from file.

before was:
`[2/23/2021, 11:36:43 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, production power max: NaN kW`
now is:
`[2/23/2021, 12:12:24 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, production power max: 2.016 kW`

closed 2021-02-23 by grzegorz914

hoefs, 2021-02-23:
> I see the error not more appear in Your loog , looks like was problem with read data from file.
>
> before was:
> `[2/23/2021, 11:36:43 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, production power [...]

Correct. But I deleted the file from commandline and after hb restart it recreated it and start updating it... size was 0 before deleting it. Now it is indeed working fine, but it should not need manual intervention.

hoefs, 2021-02-23:
[2/23/2021, 1:34:01 PM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, productionPowerMaxFile saved successful in: /homebridge/enphaseEnvoy/productionPowerMax_192168225 2.746 kW

4.5.1 works fine!

commit d2a908b5 (2021-02-23): fix #19