grzegorz914/homebridge-enphase-envoy #21: TypeError after configuring the plugin

tcasteli, 2021-02-25:
After installing and configuring the plugin I keep getting the following error:

[25/02/2021, 08:29:28] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, update Device state error: TypeError: Cannot read property 'getCharacteristic' of undefined
    at envoyDevice.prepareAccessory (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:3076:41)
    at envoyDevice.updateDeviceState (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:2551:14)

I only set-up the IP address in the configuration.

grzegorz914, 2021-02-25:
Plugin version?

tcasteli, 2021-02-25:
The plugin version is  v4.6.2, but I experienced the issue with previous versions too

grzegorz914, 2021-02-25:
Enable debug and get the log

grzegorz914, 2021-02-25:
Try 4.6.3-beta.1

tcasteli, 2021-02-25:
This is what I get with 4.6.3-beta.1:

[25/02/2021, 10:01:32] [enphaseEnvoy] Debug home: {
  software_build_epoch: 1542074240,
  is_nonvoy: false,
  db_size: '3 MB',
  db_percent_full: '0',
  timezone: 'Europe/Amsterdam',
  current_date: '02/25/2021',
  current_time: '10:00',
  network: [Object],
  tariff: 'none',
  comm: [Object],
  alerts: [],
  update_status: 'satisfied'
}, inventory: [ [Object], [Object], [Object] ], meters: [ [Object], [Object] ], production: {
  wattHoursToday: 454,
  wattHoursSevenDays: 73614,
  wattHoursLifetime: 73624,
  wattsNow: 414
} productionCT: { production: [Array], consumption: [Array], storage: [Array] }
[25/02/2021, 10:01:32] [enphaseEnvoy] Debug production inverters: [
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object]
]
[25/02/2021, 10:01:32] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, update Device state error: TypeError: Cannot read property 'length' of undefined
    at envoyDevice.updateDeviceState (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:1654:42)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)
[25/02/2021, 10:01:35] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, requesting devices info.

grzegorz914, 2021-02-25:
fixed

closed 2021-02-25 by grzegorz914

tcasteli, 2021-02-25:
With version 4.6.3 I get the following error: 

[25/02/2021, 10:22:34] [enphaseEnvoy] prepareEnphaseServices
[25/02/2021, 10:22:34] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, update Device state error: TypeError: Cannot read property 'getCharacteristic' of undefined
    at envoyDevice.prepareAccessory (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:2974:41)
    at envoyDevice.updateDeviceState (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:2449:14)
[25/02/2021, 10:22:34] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, requesting devices info.
[25/02/2021, 10:22:39] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, requesting devices info.

grzegorz914, 2021-02-25:
please get output from http//envoy.local/production.json

tcasteli, 2021-02-25:
{"production":[{"type":"inverters","activeCount":12,"readingTime":1614245706,"wNow":670,"whLifetime":73924},{"type":"eim","activeCount":0,"measurementType":"production","readingTime":1614245708,"wNow":-1.005,"whLifetime":0.0,"varhLeadLifetime":0.0,"varhLagLifetime":0.0,"vahLifetime":0.0,"rmsCurrent":0.24,"rmsVoltage":244.947,"reactPwr":-0.0,"apprntPwr":1.195,"pwrFactor":-1.0,"whToday":0.0,"whLastSevenDays":0.0,"vahToday":0.0,"varhLeadToday":0.0,"varhLagToday":0.0}],"consumption":[{"type":"eim","activeCount":0,"measurementType":"total-consumption","readingTime":1614245708,"wNow":4.998,"whLifetime":0.0,"varhLeadLifetime":0.0,"varhLagLifetime":0.0,"vahLifetime":0.0,"rmsCurrent":0.775,"rmsVoltage":245.646,"reactPwr":0.0,"apprntPwr":190.304,"pwrFactor":0.03,"whToday":0.0,"whLastSevenDays":0.0,"vahToday":0.0,"varhLeadToday":0.0,"varhLagToday":0.0},{"type":"eim","activeCount":0,"measurementType":"net-consumption","readingTime":1614245708,"wNow":6.003,"whLifetime":0.0,"varhLeadLifetime":0.0,"varhLagLifetime":0.0,"vahLifetime":0.0,"rmsCurrent":0.534,"rmsVoltage":246.346,"reactPwr":0.0,"apprntPwr":37.123,"pwrFactor":0.19,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}

grzegorz914, 2021-02-25:
Fixed in 4.6.4

tcasteli, 2021-02-25:
Unfortunately I'm still getting this error:

[25/02/2021, 10:44:54] [enphaseEnvoy] Debug home: {
  software_build_epoch: 1542074240,
  is_nonvoy: false,
  db_size: '3 MB',
  db_percent_full: '0',
  timezone: 'Europe/Amsterdam',
  current_date: '02/25/2021',
  current_time: '10:44',
  network: [Object],
  tariff: 'none',
  comm: [Object],
  alerts: [],
  update_status: 'satisfied'
}, inventory: [ [Object], [Object], [Object] ], meters: [ [Object], [Object] ], production: {
  wattHoursToday: 864,
  wattHoursSevenDays: 74024,
  wattHoursLifetime: 74034,
  wattsNow: 653
} productionCT: { production: [Array], consumption: [Array], storage: [Array] }
[25/02/2021, 10:44:55] [enphaseEnvoy] Debug production inverters: [
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object]
]
[25/02/2021, 10:44:55] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, savedProductionPowerMax: 2.296 kW
[25/02/2021, 10:44:55] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, productionPowerMax: 2.296 kW
[25/02/2021, 10:44:55] [enphaseEnvoy] prepareAccessory
[25/02/2021, 10:44:55] [enphaseEnvoy] prepareInformationService
[25/02/2021, 10:44:55] [enphaseEnvoy] prepareEnphaseServices
[25/02/2021, 10:44:55] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, update Device state error: TypeError: Cannot read property 'getCharacteristic' of undefined
    at envoyDevice.prepareAccessory (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:2970:41)
    at envoyDevice.updateDeviceState (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:2445:14)

grzegorz914, 2021-02-25:
Please install 4.6.5-beta.1 and get the log again

tcasteli, 2021-02-25:
[25/02/2021, 11:43:29] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, saved Device Info successful.
[25/02/2021, 11:43:29] [enphaseEnvoy] -------- Envoy-S --------
[25/02/2021, 11:43:29] [enphaseEnvoy] Manufacturer: Enphase
[25/02/2021, 11:43:29] [enphaseEnvoy] Model: Envoy
[25/02/2021, 11:43:29] [enphaseEnvoy] Meters: 2
[25/02/2021, 11:43:29] [enphaseEnvoy] Q-Relays: 1
[25/02/2021, 11:43:29] [enphaseEnvoy] Encharges: 0
[25/02/2021, 11:43:29] [enphaseEnvoy] Inverters: 12
[25/02/2021, 11:43:29] [enphaseEnvoy] Firmware: R4.10.35
[25/02/2021, 11:43:29] [enphaseEnvoy] SerialNr: 122033019515
[25/02/2021, 11:43:29] [enphaseEnvoy] Time: 25/02/2021, 11:43:02
[25/02/2021, 11:43:29] [enphaseEnvoy] ----------------------------------
[25/02/2021, 11:43:40] [enphaseEnvoy] Debug home: {
  software_build_epoch: 1542074240,
  is_nonvoy: false,
  db_size: '3 MB',
  db_percent_full: '0',
  timezone: 'Europe/Amsterdam',
  current_date: '02/25/2021',
  current_time: '11:43',
  network: [Object],
  tariff: 'none',
  comm: [Object],
  alerts: [],
  update_status: 'satisfied'
}, inventory: [ [Object], [Object], [Object] ], meters: [ [Object], [Object] ], production: {
  wattHoursToday: 2019,
  wattHoursSevenDays: 75143,
  wattHoursLifetime: 75189,
  wattsNow: 1536
} productionCT: { production: [Array], consumption: [Array], storage: [Array] }
[25/02/2021, 11:43:40] [enphaseEnvoy] Debug production inverters: [
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object]
]
[25/02/2021, 11:43:40] [enphaseEnvoy] Test 1.536
[25/02/2021, 11:43:40] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, savedProductionPowerMax: 2.296 kW
[25/02/2021, 11:43:40] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, productionPowerMax: 2.296 kW
[25/02/2021, 11:43:40] [enphaseEnvoy] Test1 1.536
[25/02/2021, 11:43:40] [enphaseEnvoy] prepareAccessory
[25/02/2021, 11:43:40] [enphaseEnvoy] prepareInformationService
[25/02/2021, 11:43:40] [enphaseEnvoy] prepareEnphaseServices
[25/02/2021, 11:43:40] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, update Device state error: TypeError: Cannot read property 'getCharacteristic' of undefined
    at envoyDevice.prepareAccessory (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:2971:41)
    at envoyDevice.updateDeviceState (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:2446:14)

grzegorz914, 2021-02-25:
OK i fount the mistake, please install beta.3 and get feedback

tcasteli, 2021-02-25:
[25/02/2021, 11:56:12] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, saved Device Info successful.
[25/02/2021, 11:56:12] [enphaseEnvoy] -------- Envoy-S --------
[25/02/2021, 11:56:12] [enphaseEnvoy] Manufacturer: Enphase
[25/02/2021, 11:56:12] [enphaseEnvoy] Model: Envoy
[25/02/2021, 11:56:12] [enphaseEnvoy] Meters: 2
[25/02/2021, 11:56:12] [enphaseEnvoy] Q-Relays: 1
[25/02/2021, 11:56:12] [enphaseEnvoy] Encharges: 0
[25/02/2021, 11:56:12] [enphaseEnvoy] Inverters: 12
[25/02/2021, 11:56:12] [enphaseEnvoy] Firmware: R4.10.35
[25/02/2021, 11:56:12] [enphaseEnvoy] SerialNr: 122033019515
[25/02/2021, 11:56:12] [enphaseEnvoy] Time: 25/02/2021, 11:55:45
[25/02/2021, 11:56:12] [enphaseEnvoy] ----------------------------------
[25/02/2021, 11:56:28] [enphaseEnvoy] Debug home: {
  software_build_epoch: 1542074240,
  is_nonvoy: false,
  db_size: '3 MB',
  db_percent_full: '0',
  timezone: 'Europe/Amsterdam',
  current_date: '02/25/2021',
  current_time: '11:55',
  network: [Object],
  tariff: 'none',
  comm: [Object],
  alerts: [],
  update_status: 'satisfied'
}, inventory: [ [Object], [Object], [Object] ], meters: [ [Object], [Object] ], production: {
  wattHoursToday: 2247,
  wattHoursSevenDays: 75332,
  wattHoursLifetime: 75417,
  wattsNow: 1886
} productionCT: { production: [Array], consumption: [Array], storage: [Array] }
[25/02/2021, 11:56:28] [enphaseEnvoy] Debug production inverters: [
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object]
]
[25/02/2021, 11:56:28] [enphaseEnvoy] Test 1.886
[25/02/2021, 11:56:28] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, savedProductionPowerMax: 2.296 kW
[25/02/2021, 11:56:28] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, productionPowerMax: 2.296 kW
[25/02/2021, 11:56:28] [enphaseEnvoy] prepareAccessory
[25/02/2021, 11:56:28] [enphaseEnvoy] prepareInformationService
[25/02/2021, 11:56:28] [enphaseEnvoy] prepareEnphaseServices
[25/02/2021, 11:56:28] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, update Device state error: TypeError: Cannot read property 'getCharacteristic' of undefined
    at envoyDevice.prepareAccessory (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:2971:41)
    at envoyDevice.updateDeviceState (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:2446:14)

grzegorz914, 2021-02-25:
U dont installed latest beta try again

tcasteli, 2021-02-25:
Yes I did, and I did a restart. Another restart gives this logging:

  timezone: 'Europe/Amsterdam',
  current_date: '02/25/2021',
  current_time: '12:01',
  network: [Object],
  tariff: 'none',
  comm: [Object],
  alerts: [],
  update_status: 'satisfied'
}, inventory: [ [Object], [Object], [Object] ], meters: [ [Object], [Object] ], production: {
  wattHoursToday: 2592,
  wattHoursSevenDays: 75626,
  wattHoursLifetime: 75762,
  wattsNow: 1969
} productionCT: { production: [Array], consumption: [Array], storage: [Array] }
[25/02/2021, 12:02:33] [enphaseEnvoy] Debug production inverters: [
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object],
  [Object], [Object]
]
[25/02/2021, 12:02:33] [enphaseEnvoy] Test 1.969
[25/02/2021, 12:02:33] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, savedProductionPowerMax: 2.296 kW
[25/02/2021, 12:02:33] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, productionPowerMax: 2.296 kW
[25/02/2021, 12:02:33] [enphaseEnvoy] prepareAccessory
[25/02/2021, 12:02:33] [enphaseEnvoy] prepareInformationService
[25/02/2021, 12:02:33] [enphaseEnvoy] prepareEnphaseServices
[25/02/2021, 12:02:33] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, update Device state error: TypeError: Cannot read property 'getCharacteristic' of undefined
    at envoyDevice.prepareAccessory (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:2971:41)
    at envoyDevice.updateDeviceState (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:2446:14)

grzegorz914, 2021-02-25:
YesU are right, my bad finally fixed in beta.4

tcasteli, 2021-02-25:
It is now reading the microinverters but a new error occurs:

[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043228 last power: 66 W
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043228 max power: 239 W
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043228 producing: Yes
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043228 communicating: Yes
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043228 provisioned: Yes
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043228 operating: No
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043228 status: Normal
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043228 firmware: 520-00082-r01-v04.18.02
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043228 last report: 25/02/2021, 12:30:19
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043231 last power: 71 W
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043231 max power: 237 W
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043231 producing: Yes
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043231 communicating: Yes
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043231 provisioned: Yes
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043231 operating: No
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043231 status: Normal
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043231 firmware: 520-00082-r01-v04.18.02
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043231 last report: 25/02/2021, 12:30:19
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043760 last power: 71 W
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043760 max power: 237 W
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043760 producing: Yes
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043760 communicating: Yes
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043760 provisioned: Yes
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043760 operating: No
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043760 status: Normal
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043760 firmware: 520-00082-r01-v04.18.02
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043760 last report: 25/02/2021, 12:30:21
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039042918 last power: 72 W
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039042918 max power: 243 W
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039042918 producing: Yes
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039042918 communicating: Yes
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039042918 provisioned: Yes
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039042918 operating: No
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039042918 status: Normal
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039042918 firmware: 520-00082-r01-v04.18.02
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039042918 last report: 25/02/2021, 12:30:23
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039039950 last power: 71 W
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039039950 max power: 238 W
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039039950 producing: Yes
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039039950 communicating: Yes
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039039950 provisioned: Yes
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039039950 operating: No
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039039950 status: Normal
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039039950 firmware: 520-00082-r01-v04.18.02
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039039950 last report: 25/02/2021, 12:30:23
[25/02/2021, 12:33:57] [enphaseEnvoy] Device: 192.168.42.229 Envoy-S, microinverter: 122039043093 last power: 70 W
[... cut ...]
    at Object.__awaiter (/usr/lib/node_modules/homebridge/node_modules/tslib/tslib.js:113:16)
    at Characteristic.enphaseMicroinverterCommLevel.Characteristic.handleGetRequest (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/dist/lib/Characteristic.js:648:24)
    at Characteristic.enphaseMicroinverterCommLevel.<anonymous> (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2229:22)
    at step (/usr/lib/node_modules/homebridge/node_modules/tslib/tslib.js:143:27)

grzegorz914, 2021-02-25:
try completely reboot hobebridge this sometimes happened from cache

grzegorz914, 2021-02-25:
are U type something in installer password

tcasteli, 2021-02-25:
I didn't type anything in the installer password field

tcasteli, 2021-02-25:
After rebooting the issue still is there

grzegorz914, 2021-02-25:
OK, try beta 5

commit 672c0fbd (2021-02-25): fix #21
commit a7242c1b (2021-02-25): fix #21, #22