grzegorz914/homebridge-enphase-envoy #24: TypeError: Cannot read property 'type' of undefined

GentleJan, 2021-02-26:
[2/26/2021, 8:30:12 AM] [enphaseEnvoy] Device: 192.168.178.11 Envoy, update Device state error: TypeError: Cannot read property 'type' of undefined
    at envoyDevice.updateDeviceState (/homebridge/node_modules/homebridge-enphase-envoy/index.js:2028:64)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)

Plugin Version 4.6.6
Homebridge 1.3.1
nodejs 14.15.5

grzegorz914, 2021-02-26:
Can U get me output from 
http://envoy.local/production.json

GentleJan, 2021-02-26:
{"production":[{"type":"inverters","activeCount":18,"readingTime":0,"wNow":0,"whLifetime":7905289}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}

grzegorz914, 2021-02-26:
Please also from
http://envoy.local/api/v1/production

GentleJan, 2021-02-26:
{
  "wattHoursToday": 19393,
  "wattHoursSevenDays": 98401,
  "wattHoursLifetime": 7905289,
  "wattsNow": 0
}

commit 6fc1e6f2 (2021-02-26): fix #24
closed 2021-02-26 by grzegorz914