grzegorz914/homebridge-enphase-envoy #25: Issue homebridge-enphase-envoy v4.6.6 - HPE_UNEXPECTED_CONTENT_LENGTH

eronnes, 2021-02-26:
homebridge-enphase-envoy v4.6.6

Any solution available?

[2/26/2021, 9:05:54 PM] [enphaseEnvoy] Device: 192.168.178.216 Envoy-S, update Device state error: Error: Parse Error: Content-Length can't be present with Transfer-Encoding
    at Socket.socketOnData (_http_client.js:509:22)
    at Socket.emit (events.js:315:20)
    at addChunk (internal/streams/readable.js:309:12)
    at readableAddChunk (internal/streams/readable.js:284:9)
    at Socket.Readable.push (internal/streams/readable.js:223:10)
    at TCP.onStreamRead (internal/stream_base_commons.js:188:23) {
  bytesParsed: 242,
  code: 'HPE_UNEXPECTED_CONTENT_LENGTH',
  reason: "Content-Length can't be present with Transfer-Encoding",
  rawPacket: <Buffer 48 54 54 50 2f 31 2e 31 20 33 30 31 20 4d 6f 76 65 64 20 50 65 72 6d 61 6e 65 6e 74 6c 79 0d 0a 44 61 74 65 3a 20 46 72 69 2c 20 32 36 20 46 65 62 20 ... 192 more bytes>,
  config: [Object],
  request: [Writable],
  response: undefined,
  isAxiosError: true,
  toJSON: [Function: toJSON]

grzegorz914, 2021-02-26:
Please enable debug mode, restart homebridge and get the log.

eronnes, 2021-02-27:
Done, result of the logging:

[enphaseEnvoy] Device: 192.168.178.216 Envoy-S, requesting devices info.
[2/27/2021, 1:55:45 PM] [enphaseEnvoy] Device: 192.168.178.216 Envoy-S, update Device state error: Error: Parse Error: Content-Length can't be present with Transfer-Encoding
    at Socket.socketOnData (_http_client.js:509:22)
    at Socket.emit (events.js:315:20)
    at addChunk (internal/streams/readable.js:309:12)
    at readableAddChunk (internal/streams/readable.js:284:9)
    at Socket.Readable.push (internal/streams/readable.js:223:10)
    at TCP.onStreamRead (internal/stream_base_commons.js:188:23) {
  bytesParsed: 242,
  code: 'HPE_UNEXPECTED_CONTENT_LENGTH',
  reason: "Content-Length can't be present with Transfer-Encoding",
  rawPacket: <Buffer 48 54 54 50 2f 31 2e 31 20 33 30 31 20 4d 6f 76 65 64 20 50 65 72 6d 61 6e 65 6e 74 6c 79 0d 0a 44 61 74 65 3a 20 53 61 74 2c 20 32 37 20 46 65 62 20 ... 192 more bytes>,
  config: [Object],
  request: [Writable],
  response: undefined,
  isAxiosError: true,
  toJSON: [Function: toJSON]
}
[2/27/2021, 1:55:45 PM] [Hue] Philips hue: request 18: GET /lights
[2/27/2021, 1:55:45 PM] [Hue] Philips hue: request 18: 200 OK
[2/27/2021, 1:55:49 PM] [enphaseEnvoy] Device: 192.168.178.216 Envoy-S, requesting devices info.
[2/27/2021, 1:55:50 PM] [Hue] Philips hue: request 19: GET /lights
[2/27/2021, 1:55:50 PM] [Hue] Philips hue: request 19: 200 OK

eronnes, 2021-02-27:
Is it necessary to enter an installer username and password in the setting?

grzegorz914, 2021-02-27:
If U do not have Installer password do not type it in settings. Installer user leave empty.

Please get the output from:

http://envoy.local/info.xml
http://envoy.local/inventory.json

eronnes, 2021-02-27:
The two links do not open after copy them into the "safari" bar....

grzegorz914, 2021-02-27:
then Your Envoy is not reached on the network, check Your Envoy IP address in router

eronnes, 2021-02-27:
info.xml
1614431790 
122038013143 
800-00654-r06 
D5.0.50 4c8675 0 1 true 500-00001-r01 02.00.00 950 500-00011-r01 04.01.15 1905ae 590-00019-r01 02.00.01 1f421b 500-00002-r01 05.00.50 20f047 500-00005-r01 01.02.252 b2d6f5 500-00008-r01 02.01.22 2faa48 500-00010-r0105.00.02 4fe435 500-00013-r01 03.02.07 276642 500-00012-r01 02.02.00 cbbb6d 500-00016-r01 02.00.00 54a6dc 1589922055 release-5.0.x-90-May-19-20-13:21:08

inventory,json
[{"type":"PCU","devices":[{"part_num":"800-00631-r02","installed":"1610369763","serial_num":"122043051735","device_status":["envoy.global.ok"],"last_rpt_date":"1614433379","admin_state":1,"dev_type":1,"created_date":"1610369763","img_load_date":"1583340413","img_pnum_running":"520-00082-r01-v04.18.02","ptpn":"540-00135-r01-v04.18.03","chaneid":1627390225,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00631-r02","installed":"1610369767","serial_num":"122043045931","device_status":["envoy.global.ok"],"last_rpt_date":"1614433381","admin_state":1,"dev_type":1,"created_date":"1610369767","img_load_date":"1583340413","img_pnum_running":"520-00082-r01-v04.18.02","ptpn":"540-00135-r01-v04.18.03","chaneid":1627390481,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00631-r02","installed":"1610369771","serial_num":"122043046071","device_status":["envoy.global.ok"],"last_rpt_date":"1614433384","admin_state":1,"dev_type":1,"created_date":"1610369771","img_load_date":"1583340413","img_pnum_running":"520-00082-r01-v04.18.02","ptpn":"540-00135-r01-v04.18.03","chaneid":1627390737,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00631-r02","installed":"1610369775","serial_num":"122043051782","device_status":["envoy.global.ok"],"last_rpt_date":"1614433385","admin_state":1,"dev_type":1,"created_date":"1610369775","img_load_date":"1583340413","img_pnum_running":"520-00082-r01-v04.18.02","ptpn":"540-00135-r01-v04.18.03","chaneid":1627390993,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00631-r02","installed":"1610369779","serial_num":"122043051623","device_status":["envoy.global.ok"],"last_rpt_date":"1614433387","admin_state":1,"dev_type":1,"created_date":"1610369779","img_load_date":"1583340413","img_pnum_running":"520-00082-r01-v04.18.02","ptpn":"540-00135-r01-v04.18.03","chaneid":1627391249,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00631-r02","installed":"1610369789","serial_num":"122043051697","device_status":["envoy.global.ok"],"last_rpt_date":"1614433391","admin_state":1,"dev_type":1,"created_date":"1610369789","img_load_date":"1583340413","img_pnum_running":"520-00082-r01-v04.18.02","ptpn":"540-00135-r01-v04.18.03","chaneid":1627391505,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00631-r02","installed":"1610369792","serial_num":"122043045688","device_status":["envoy.global.ok"],"last_rpt_date":"1614433391","admin_state":1,"dev_type":1,"created_date":"1610369792","img_load_date":"1583340413","img_pnum_running":"520-00082-r01-v04.18.02","ptpn":"540-00135-r01-v04.18.03","chaneid":1627391761,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00631-r02","installed":"1610369796","serial_num":"122043045726","device_status":["envoy.global.ok"],"last_rpt_date":"1614433394","admin_state":1,"dev_type":1,"created_date":"1610369796","img_load_date":"1583340413","img_pnum_running":"520-00082-r01-v04.18.02","ptpn":"540-00135-r01-v04.18.03","chaneid":1627392017,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00631-r02","installed":"1610369800","serial_num":"122043051709","device_status":["envoy.global.ok"],"last_rpt_date":"1614433396","admin_state":1,"dev_type":1,"created_date":"1610369800","img_load_date":"1583340413","img_pnum_running":"520-00082-r01-v04.18.02","ptpn":"540-00135-r01-v04.18.03","chaneid":1627392273,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00631-r02","installed":"1610369810","serial_num":"122043051625","device_status":["envoy.global.ok"],"last_rpt_date":"1614433397","admin_state":1,"dev_type":1,"created_date":"1610369810","img_load_date":"1583340413","img_pnum_running":"520-00082-r01-v04.18.02","ptpn":"540-00135-r01-v04.18.03","chaneid":1627392529,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00631-r02","installed":"1610369813","serial_num":"122043051557","device_status":["envoy.global.ok"],"last_rpt_date":"1614433398","admin_state":1,"dev_type":1,"created_date":"1610369813","img_load_date":"1583340413","img_pnum_running":"520-00082-r01-v04.18.02","ptpn":"540-00135-r01-v04.18.03","chaneid":1627392785,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00631-r02","installed":"1610369833","serial_num":"122043033136","device_status":["envoy.global.ok"],"last_ [... cut ...] evice_status":["envoy.global.ok"],"last_rpt_date":"1614433434","admin_state":1,"dev_type":12,"created_date":"1610369822","img_load_date":"1529083534","img_pnum_running":"520-00086-r01-v02.12.11","ptpn":"540-00139-r01-v02.12.00","chaneid":1811939601,"device_control":[{"gficlearset":false}],"producing":false,"communicating":true,"provisioned":true,"operating":true,"relay":"closed","reason_code":-1,"reason":"ok","line-count":3,"line1-connected":true,"line2-connected":true,"line3-connected":true}]}]

grzegorz914, 2021-02-27:
get also http://envoy.local/home.json

eronnes, 2021-02-27:
Looks like it was an issue on the envoy. After a restart it works again...only issue is the following error message:

[2/27/2021, 3:56:26 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Energy last 7 days': characteristic was supplied illegal value: number -368.577436 exceeded minimum of 0. See https://git.io/JtMGR for more info.

grzegorz914, 2021-02-27:
It’s nothing wrong, how is possible Your energy last 7 days is negative?

can You get output from 
http://envoy.local/production.json

grzegorz914, 2021-02-27:
Are U updated plugin to latest version?

eronnes, 2021-02-28:
Yes, I’m on the latest version. It’s running now (still giving the warning for the energy last 7 days). But that could be som installation thing which needs te fase out. I wil check in a view days.

There is one other thing. After I started the package in homebridge I am unable to acces the envoy via the homepage. I this supposed to be happening?

eronnes, 2021-02-28:
Leaves me the question how I can see the data in Apple HomeKit?

grzegorz914, 2021-02-28:
1. Try to reduce refresh time.
 2. https://github.com/grzegorz914/homebridge-enphase-envoy#info-v45x-and-above

closed 2021-03-06 by eronnes