grzegorz914/homebridge-enphase-envoy #26: homebridge-enphase-envoy v4.6.15-beta.18

hoefs, 2021-03-02:
```
[3/2/2021, 9:20:42 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, requesting devices info eror: TypeError: Cannot read property 'state' of undefined
    at envoyDevice.getDeviceInfo (/homebridge/node_modules/homebridge-enphase-envoy/index.js:1469:49), state: Offline trying to reconnect.
```

I can see in the log it connects to the Envoy-S but plugin is not working in EVE

grzegorz914, 2021-03-02:
Try beta 19

hoefs, 2021-03-02:
> Try beta 19

Thanks! Fixed

```
[3/2/2021, 11:56:50 AM] [enphaseEnvoy] Device: 192.168.2.25 Envoy-S, saved Device Info successful.
[3/2/2021, 11:56:50 AM] [enphaseEnvoy] -------- Envoy-S --------
[3/2/2021, 11:56:50 AM] [enphaseEnvoy] Manufacturer: Enphase
[3/2/2021, 11:56:50 AM] [enphaseEnvoy] Model: Envoy
[3/2/2021, 11:56:50 AM] [enphaseEnvoy] Api ver: 1
[3/2/2021, 11:56:50 AM] [enphaseEnvoy] Firmware: R4.10.35
[3/2/2021, 11:56:50 AM] [enphaseEnvoy] SerialNr: 121921106049
[3/2/2021, 11:56:50 AM] [enphaseEnvoy] Time: 3/2/2021, 11:56:39 AM
[3/2/2021, 11:56:50 AM] [enphaseEnvoy] ------------------------------
[3/2/2021, 11:56:50 AM] [enphaseEnvoy] Meters: No
[3/2/2021, 11:56:50 AM] [enphaseEnvoy] Q-Relays: 1
[3/2/2021, 11:56:50 AM] [enphaseEnvoy] Encharges: 0
[3/2/2021, 11:56:50 AM] [enphaseEnvoy] Inverters: 14
[3/2/2021, 11:56:50 AM] [enphaseEnvoy] ------------------------------
```

grzegorz914, 2021-03-02:
Beta 20 will identify your envoy correct

grzegorz914, 2021-03-02:
Just as info, if U have Mac device then the Controller app is now available also for Mac.

closed 2021-03-02 by grzegorz914

hoefs, 2021-03-02:
> Beta 20 will identify your envoy correct

Is this correct?

```
[3/2/2021, 12:06:56 PM] [enphaseEnvoy] -------- Envoy-S --------
[3/2/2021, 12:06:56 PM] [enphaseEnvoy] Manufacturer: Enphase
[3/2/2021, 12:06:56 PM] [enphaseEnvoy] Model: ENV-S-WB-230
[3/2/2021, 12:06:56 PM] [enphaseEnvoy] Api ver: 1
[3/2/2021, 12:06:56 PM] [enphaseEnvoy] Firmware: R4.10.35
[3/2/2021, 12:06:56 PM] [enphaseEnvoy] SerialNr: 121921106049
[3/2/2021, 12:06:56 PM] [enphaseEnvoy] Time: 3/2/2021, 12:06:45 PM
[3/2/2021, 12:06:56 PM] [enphaseEnvoy] ------------------------------
[3/2/2021, 12:06:56 PM] [enphaseEnvoy] Meters: No
[3/2/2021, 12:06:56 PM] [enphaseEnvoy] Q-Relays: 1
[3/2/2021, 12:06:56 PM] [enphaseEnvoy] Encharges: 0
[3/2/2021, 12:06:56 PM] [enphaseEnvoy] Inverters: 14
[3/2/2021, 12:06:56 PM] [enphaseEnvoy] ------------------------------
```

hoefs, 2021-03-02:
> Just as info, if U have Mac device then the Controller app is now available also for Mac.

I have a mac. Are you pointing to the bitbar solution with homebrew?

grzegorz914, 2021-03-02:
Y do not need home brew just look in the Mac App Store for Controller app

grzegorz914, 2021-03-02:
> Is this correct?
>
> ```
> [3/2/2021, 12:06:56 PM] [enphaseEnvoy] -------- Envoy-S --------
> [3/2/2021, 12:06:56 PM] [enphaseEnvoy] Manufacturer: Enphase
> [3/2/2021, 12:06:56 PM] [enphaseEnvoy] Model: [...]

Yes it is correct Envoy-S Standard EU

grzegorz914, 2021-03-02:
> > Just as info, if U have Mac device then the Controller app is now available also for Mac.
> 
> I have a mac. Are you pointing to the bitbar solution with homebrew?

Regarding homebrew just try the command from readme in Mac console.

hoefs, 2021-03-02:
> You do not need homebrew just look in the Mac App Store for Controller app. Since today is also available for mac

I can't find it, do you have a link?

cloudkucooland, 2021-03-02:
https://apps.apple.com/us/app/controller-for-homekit/id1198176727