grzegorz914/homebridge-enphase-envoy #27: This accessory is not currently supported by the Home app

godzero99, 2021-03-03:
Hi, 

I installed plugin and configured IP of envoy. Left other config as default. Homebridge pull data from envoy. When I add accessory to home app. It is added. But came up with error "This accessory is not currently supported by the Home app"

What do I did wrong?

plugin version: homebridge-enphase-envoy v4.6.15-beta.21
homebridge version: v1.3.1

Debug from homebridge: 

[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m [90mDevice: 192.168.x.13 Envoy-S, parse info.xml successful: {[39m
[90m  "envoy_info": {[39m
[90m    "time": [[39m
[90m      "1614741241"[39m
[90m    ],[39m
[90m    "device": [[39m
[90m      {[39m
[90m        "sn": [[39m
[90m          "xxxxxxx"[39m
[90m        ],[39m
[90m        "pn": [[39m
[90m          "800-00554-r03"[39m
[90m        ],[39m
[90m        "software": [[39m
[90m          "R4.10.35"[39m
[90m        ],[39m
[90m        "euaid": [[39m
[90m          "4c8675"[39m
[90m        ],[39m
[90m        "seqnum": [[39m
[90m          "0"[39m
[90m        ],[39m
[90m        "apiver": [[39m
[90m          "1"[39m
[90m        ],[39m
[90m        "imeter": [[39m
[90m          "true"[39m
[90m        ][39m
[90m      }[39m
[90m    ],[39m
[90m    "package": [[39m
[90m      {[39m
[90m        "$": {[39m
[90m          "name": "rootfs"[39m
[90m        },[39m
[90m        "pn": [[39m
[90m          "500-00001-r01"[39m
[90m        ],[39m
[90m        "version": [[39m
[90m          "02.00.00"[39m
[90m        ],[39m
[90m        "build": [[39m
[90m          "945"[39m
[90m        ][39m
[90m      },[39m
[90m      {[39m
[90m        "$": {[39m
[90m          "name": "kernel"[39m
[90m        },[39m
[90m        "pn": [[39m
[90m          "500-00011-r01"[39m
[90m        ],[39m
[90m        "version": [[39m
[90m          "04.00.00"[39m
[90m        ],[39m
[90m        "build": [[39m
[90m          "5bb754"[39m
[90m        ][39m
[90m      },[39m
[90m      {[39m
[90m        "$": {[39m
[90m          "name": "boot"[39m
[90m        },[39m
[90m        "pn": [[39m
[90m          "590-00018-r01"[39m
[90m        ],[39m
[90m        "version": [[39m
[90m          "02.00.01"[39m
[90m        ],[39m
[90m        "build": [[39m
[90m          "426697"[39m
[90m        ][39m
[90m      },[39m
[90m      {[39m
[90m        "$": {[39m
[90m          "name": "app"[39m
[90m        },[39m
[90m        "pn": [[39m
[90m          "500-00002-r01"[39m
[90m        ],[39m
[90m        "version": [[39m
[90m          "04.10.35"[39m
[90m        ],[39m
[90m        "build": [[39m
[90m          "6ed292"[39m
[90m        ][39m
[90m      },[39m
[90m      {[39m
[90m        "$": {[39m
[90m          "name": "devimg"[39m
[90m        },[39m
[90m        "pn": [[39m
[90m          "500-00005-r01"[39m
[90m        ],[39m
[90m        "version": [[39m
[90m          "01.02.186"[39m
[90m        ],[39m
[90m        "build": [[39m
[90m          "d0d70f"[39m
[90m        ][39m
[90m      },[39m
[90m      {[39m
[90m        "$": {[39m
[90m          "name": "geo"[39m
[90m        },[39m
[90m        "pn": [[39m
[90m          "500-00008-r01"[39m
[90m        ],[39m
[90m        "version": [[39m
[90m          "02.01.22"[39m
[90m        ],[39m
[90m        "build": [[39m
[90m          "06e201"[39m
[90m        ][39m
[90m      },[39m
[90m      {[39m
[90m        "$": {[39m
[90m          "name": "backbone"[39m
[90m        },[39m
[90m        "pn": [[39m
[90m          "500-00010-r01"[39m
[90m        ],[39m
[90m        "version": [[39m
[90m          "04.10.25"[39m
[90m        ],[39m
[90m        "build": [[39m
[90m          "7b7de5"[39m
[90m        ][39m
[90m      },[39m
[90m      {[39m
[90m        "$": {[39m
[90m          "name": "meter"[39m
[90m        },[39m
[90m        "pn": [[39m
[90m          "500-00013-r01"[39m
[90m        ],[39m
[90m        "version": [[39m
[90m          "03.02.07"[39m
[90m        ],[39m
[90m        "build": [[39m
[90m          "4c9d48"[39m
[90m        ][39m
[90m      },[39m
[90m      {[39m
[90m        "$": {[39m
[90m          "name": "agf"[39m
[90m        },[39m
[90m        "pn": [[39m
[90m          "500-00012-r01"[39m
[90m        ],[39m
[90m        "version": [[39m
[90m          "02.02.00"[39m
[90m        ],[39m
[90m        "build": [[39m
[90m          "c00a8f"[39m
[90m        ][39m
[90m      },[39m
[90m      {[39m
[90m        "$": {[39m
[90m          "name": "security"[39m
[90m        },[39m
[90m        "pn": [[39m
[90m          "500-00016-r01"[39m
[90m        ],[39m
[90m        "version": [[39m
[90m          "02.00.00"[39m
[90m        ],[39m
[90m        "build": [[39m
[90m          "54a6dc"[39m
[90m        ][39m
[90m      }[39m
[90m    ],[39m
[90m    "build_info": [[39m
[90m      {[39m
[90m        "build_time_gmt": [[39m
[90m          "1542157882"[39m
[90m        ],[39m
[90m        "build_id": [[39m
[90m          "release-4.10.x-103-Nov-12-18-18:25:06"[39m
[90m        ][39m
[90m      }[39m
[90m    ][39m
[90m  }[39m
[90m}[39m
[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m [90mDevice: 192.168.x.13 Envoy-S, saved Device Info successful.[39m
[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m -------- Envoy-S --------
[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m Manufacturer: Enphase
[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m Model: ENV-S-WM-230
[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m Api ver: 1
[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m Firmware: R4.10.35
[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m SerialNr: xxxxxxxx
[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m Time: 3/3/2021, 2:14:01 PM
[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m ------------------------------
[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m Meters: Yes
[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m Production: Enabled
[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m Consumption: Enabled
[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m ------------------------------
[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m Q-Relays: 2
[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m Encharges: 0
[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m Inverters: 18
[37m[3/3/2021, 2:14:43 PM] [39m[36m[enphaseEnvoy][39m ------------------------------

grzegorz914, 2021-03-03:
Hi,

HomeKit app do not support custom Characteristic and display this plugin as unsupported.
All version over 2.3 need other applike Controller or Eve to display correct all the data.
The Controller app is also available on Mac.

https://apps.apple.com/us/app/controller-for-homekit/id1198176727

godzero99, 2021-03-03:
oh ok. Sorry I missed that.

Thanks for your help.

closed 2021-03-03 by grzegorz914