grzegorz914/homebridge-enphase-envoy #31: Part Number and SKU...

grzegorz914, 2021-03-22:
Hi,

I want to create list off all devices available by Enphase but the part numbers is dificult to find in web.
Please post Your Part Number and SKU for all installed devices.
This can be found in Enlighten Manager >> Devices.

This help me to identify and correct present in the plugin.

Thanks

cloudkucooland, 2021-03-24:
IQ Envoy - SKU: X-IQ-AM1-240-3
IQ7 Microinverter - SKU: IQ7-60-2-US

That's all I have

grzegorz914, 2021-03-24:
IQ Envoy is the Device Name not Part Number. SKU is OK.

KevinBlom, 2021-03-31:
**Envoy-S-Standard EU**
SKU: ENV-S-WB-230-F

**IQ7+**
SKU: IQ7PLUS-72-2-INT

grzegorz914, 2021-03-31:
> **Envoy-S-Standard EU**
> SKU: ENV-S-WB-230-F
> 
> **IQ7+**
> SKU: IQ7PLUS-72-2-INT

Part Number ?

KevinBlom, 2021-03-31:
Let me check if SN is something I can share without implications.

grzegorz914, 2021-03-31:
Part Number can be found in Enlighten Manager >> Devices. 
Not in the Enlighten.app

KevinBlom, 2021-03-31:
**Envoy-S-Standard EU**
SKU: ENV-S-WB-230-F
Part number: 800-00553-r02

**IQ7+**
SKU: IQ7PLUS-72-2-INT

I have no part numer for the IQ7+ unfortunately.

YouCanNotBeSerious, 2021-07-03:
You may already know this, but posting on the dev forum might get you a complete list from Enphase themselves - they are usually quite helpful since projects like this plugin are just another way for thier systems to get visibility :-)

https://developer.enphase.com/docs

cscott365, 2021-08-09:
Battery SKU: B03-A01-US00-1-3
Enpower SKU: EP200G101-M240US00
Envoy SKU: X-IQ-AM1-240-3-ES
Microinverters IQ7+ SKU: IQ7PLUS-72-2-US

grzegorz914, 2021-08-10:
@cscott365 
Any part Number?

cscott365, 2021-08-12:
Part numbers: 

Envoy: 800-00655-r09
Enpower: 860-00276-r48
Encharge: 830-00703-r75
Microinverters: 883-00852-r09

schlosrat, 2023-01-07:
Not sure where to get PN, but I can share my SKUs
IQ Gateway SKU: X-IQ-AM1-240-3-ES
IQ7+ Microinverter SKU: IQ7PLUS-72-M-US

holstweb, 2023-04-05:
Envoy-S-Metered-EU
SKU: ENV-S-EM-230

IQ7+ microinverter
SKU: IQ7PLUS-72-2-INT

xorguy, 2023-07-14:
Envoy-S-Metered-EU
SKU: ENV-S-WM-230

IQ7+ microinverter
SKU: IQ7PLUS-72-2-FR

montyc123, 2024-06-29:
https://github.com/montyc123/facebook-sdk-for-unity/blob/40fe97a49d543b1ff3f6d9f9c05fb22d9e80c053/README.mdown
[image]
[](url)``@montyc123>{}

eweiman, 2024-12-12:
This is what I've got. 

```
IQ Gateway
SKU: X-IQ-AM1-240-5

IQ8X
SKU: IQ8X-80-M-US

steve-lad, 2025-03-20:
Envoy S metered EU
Part #: 800-00654-r08

IQ8AC-72-M-INT
Part #: 800-01395-r03

IQ 5P Battery
Part#: 836-01890-r24

Q-RELAY-1P-INT
Part#: 800-00598-r04

CT
Part #: 800-00654-r08

Comms Kit 2
Part #: 834-01927-r03

bedwardsnet, 2025-03-29:
Envoy S metered EU
SKU: ENV-S-WM-230

IQ8HC
SKU: IQ8HC-72-M-INT

closed 2025-05-06 by github-actions[bot]