grzegorz914/homebridge-enphase-envoy #32: Accessory doesn't popup anywhere

dannydeut, 2021-03-27:
I have already had installed and linked HomeBridge to my HomeKit and have serveral plugins running. Installed this enphase plugin, but I can't figure out how to actually add the accessory. When I (re)add HB, it says it is already connected. I even installed the EVE and Controller App, but no luck. 

Can somebody give some hints on how to add the accessory? Or has it something todo with the error below? Also tried re-installing everything, but no luck. 

I do get now and then get some error in the logs.

```
[27-3-2021 09:50:57] [enphaseEnvoy] -------- Envoy-S --------
[27-3-2021 09:50:57] [enphaseEnvoy] Manufacturer: Enphase
[27-3-2021 09:50:57] [enphaseEnvoy] Model: Envoy
[27-3-2021 09:50:57] [enphaseEnvoy] Api ver: 1
[27-3-2021 09:50:57] [enphaseEnvoy] Firmware: R5.0.55
[27-3-2021 09:50:57] [enphaseEnvoy] SerialNr: **********
[27-3-2021 09:50:57] [enphaseEnvoy] Time: 27-3-2021 09:52:36
[27-3-2021 09:50:57] [enphaseEnvoy] ------------------------------
[27-3-2021 09:50:57] [enphaseEnvoy] Meters: Yes
[27-3-2021 09:50:57] [enphaseEnvoy] Production: Enabled
[27-3-2021 09:50:57] [enphaseEnvoy] Consumption: Enabled
[27-3-2021 09:50:57] [enphaseEnvoy] ------------------------------
[27-3-2021 09:50:57] [enphaseEnvoy] Q-Relays: 1
[27-3-2021 09:50:57] [enphaseEnvoy] Encharges: 0
[27-3-2021 09:50:57] [enphaseEnvoy] Inverters: 15
[27-3-2021 09:50:57] [enphaseEnvoy] ------------------------------
[27-3-2021 09:51:01] [enphaseEnvoy] Device: 192.168.1.55 Envoy-S, update Device state error: TypeError: Cannot read property 'lastReportDate' of undefined
    at envoyDevice.updateDeviceState (C:\Users\VDG\AppData\Roaming\npm\node_modules\homebridge-enphase-envoy\index.js:2482:81)
```

grzegorz914, 2021-03-27:
Read readme please:
https://github.com/grzegorz914/homebridge-enphase-envoy#adding-to-homekit
https://github.com/grzegorz914/homebridge-enphase-envoy#info-v4xx-and-above

dannydeut, 2021-03-27:
@grzegorz914 Thanks for the quick reply, but I read the readme 10 times already. And have no idea what is going wrong.

grzegorz914, 2021-03-27:
https://github.com/grzegorz914/homebridge-enphase-envoy#adding-to-homekit

dannydeut, 2021-03-27:
I tried and read that many times, but the accessory does not popup.
1. When I follow these steps, I get a spinner in iOS that it is looking for accessories. 
2. Then I press "My Accessory Isn't Shown Here"
3. Then I choose to manually enter a code from HB
4. It returns be mack to step 1
5. Repeat step 2 and 3, then I choose "use Camera".
6. It says it found a Bridge and I click "Add to Home"
7. Then I get the message "Accessory Already Added"

dannydeut, 2021-03-27:
What about the error that shows up constantly in the logs? Doesn't that prevent it somehow to add the accessory?

grzegorz914, 2021-03-27:
get your config

dannydeut, 2021-03-27:
My config
```
        {
            "devices": [
                {
                    "name": "Envoy-S",
                    "host": "192.168.1.55",
                    "refreshInterval": 5,
                    "disableLogInfo": false,
                    "envoyPasswd": "*******",
                    "installerPasswd": "********",
                    "enchargeStorageOffset": 0,
                    "powerProductionMaxDetected": 0,
                    "energyProductionLifetimeOffset": 0,
                    "powerConsumptionTotalMaxDetected": 0,
                    "energyConsumptionTotalLifetimeOffset": 0,
                    "powerConsumptionNetMaxDetected": 0,
                    "energyConsumptionNetLifetimeOffset": 0
                }
            ],
            "platform": "enphaseEnvoy"
        }
```

grzegorz914, 2021-03-27:
have U changed Your Envoy Password already?

dannydeut, 2021-03-27:
I did

dannydeut, 2021-03-27:
I think the password maybe the problem. Somehow my password manager saved the wrong password

grzegorz914, 2021-03-27:
https://github.com/grzegorz914/homebridge-enphase-envoy#configuration-values

leave empty if not changed

dannydeut, 2021-03-27:
I did change the password, but somehow my password manager did save the wrong password. Any idea how to reset the password of the envoy?

grzegorz914, 2021-03-27:
login to envoy and change the password, Bottom press Installer Login, then U have possibility to change the envoy password

grzegorz914, 2021-03-27:
Please update the plugin to the latest version, I have added some fix to prevent read inverters data if password is wrong

dannydeut, 2021-03-27:
I really appreciate the help! I updated to the latest version (4.8.11). But unfortunately no luck :-(

I also can't figure out, how I can reset the password for the 'envoy' user. I can login as an installer, but can't find any options. I feel really stupid as an engineer.... Do you have a direct link where I can reset the password?

grzegorz914, 2021-03-27:
what is the log after update

dannydeut, 2021-03-27:
Same as in the original issue at the top

grzegorz914, 2021-03-27:
are U sure U have updated to .11

dannydeut, 2021-03-27:
Yep: homebridge-enphase-envoy v4.8.11

grzegorz914, 2021-03-27:
open Enlighten Manager on iPhone and go to menu>>devices>>Envoy>>connectLocally>>Settings>>change Password
[image]

dannydeut, 2021-03-27:
Found that one. The problem is I don't know what the current password is. My password manager did not save it when I changed it. That's why I want to reset my password instead of changing it.

grzegorz914, 2021-03-27:
probasbbly U need to reset complete envoy, but before U did this try to find the right password

grzegorz914, 2021-03-27:
Please also enable debug mode and restart homebridge and post the envoy log

dannydeut, 2021-03-27:
I probably need to contact my installer to reset everything :-(

Here are some logs in debug mode. It stats "Authentication required" at some point...
```

[27-3-2021 11:31:10] [enphaseEnvoy] didFinishLaunching
[27-3-2021 11:31:10] [enphaseEnvoy] Device: 192.168.1.55 Envoy-S, requesting devices info.
[27-3-2021 11:31:12] [enphaseEnvoy] Device 192.168.1.55 Envoy-S, get devices data inventory: [ [Object], [Object], [Object] ] info: <?xml version='1.0' encoding='UTF-8'?>
<envoy_info>
  <time>1616841180</time>
  <device>
    <sn>********</sn>
    <pn>800-00654-r06</pn>
    <software>R5.0.55</software>
    <euaid>4c8675</euaid>
    <seqnum>0</seqnum>
    <apiver>1</apiver>
    <imeter>true</imeter>
  </device>
  <package name='rootfs'>
    <pn>500-00001-r01</pn>
    <version>02.00.00</version>
    <build>950</build>
  </package>
  <package name='kernel'>
    <pn>500-00011-r01</pn>
    <version>04.01.15</version>
    <build>1905ae</build>
  </package>
  <package name='boot'>
    <pn>590-00019-r01</pn>
    <version>02.00.01</version>
    <build>1f421b</build>
  </package>
  <package name='app'>
    <pn>500-00002-r01</pn>
    <version>05.00.55</version>
    <build>4f2662</build>
  </package>
  <package name='devimg'>
    <pn>500-00005-r01</pn>
    <version>01.02.293</version>
    <build>9cf065</build>
  </package>
  <package name='geo'>
    <pn>500-00008-r01</pn>
    <version>02.01.22</version>
    <build>2faa48</build>
  </package>
  <package name='backbone'>
    <pn>500-00010-r01</pn>
    <version>05.00.02</version>
    <build>4fe435</build>
  </package>
  <package name='meter'>
    <pn>500-00013-r01</pn>
    <version>03.02.07</version>
    <build>276642</build>
  </package>
  <package name='agf'>
    <pn>500-00012-r01</pn>
    <version>02.02.00</version>
    <build>8969f6</build>
  </package>
  <package name='security'>
    <pn>500-00016-r01</pn>
    <version>02.00.00</version>
    <build>54a6dc</build>
  </package>
  <package name='full'>
[... 231 lines cut ...]
}
[27-3-2021 11:31:16] [enphaseEnvoy] Debug home: {
  software_build_epoch: 1607591706,
  is_nonvoy: false,
  db_size: '3 MB',
  db_percent_full: '0',
  timezone: 'Europe/Amsterdam',
  current_date: '03/27/2021',
  current_time: '11:33',
  network: [Object],
  tariff: 'none',
  comm: [Object],
  alerts: [],
  update_status: 'satisfied'
}, inventory: [ [Object], [Object], [Object] ], meters: [ [Object], [Object] ], production: {
  wattHoursToday: 5361,
  wattHoursSevenDays: 7477,
  wattHoursLifetime: 7482,
  wattsNow: 1151
}, productionCT: { production: [Array], consumption: [Array], storage: [Array] }
[27-3-2021 11:31:16] [enphaseEnvoy] Debug meterReading: [ [Object], [Object] ]
[27-3-2021 11:31:16] [enphaseEnvoy] Device: 192.168.1.55 Envoy-S, savedProductionPowerMax: 0.0 kW
[27-3-2021 11:31:16] [enphaseEnvoy] Device: 192.168.1.55 Envoy-S, update Device state error: TypeError: Cannot read property 'lastReportDate' of undefined
    at envoyDevice.updateDeviceState (C:\Users\VDG\AppData\Roaming\npm\node_modules\homebridge-enphase-envoy\index.js:2499:81)
```

grzegorz914, 2021-03-27:
update to .12

grzegorz914, 2021-03-27:
What is Your Envoy SKU:

dannydeut, 2021-03-27:
Yes, that seems to work! Thank you very much for all the help.

dannydeut, 2021-03-27:
My SKU is ENV-S-WM-230

grzegorz914, 2021-03-27:
OK thanks will be added to correct identify in.14

grzegorz914, 2021-03-27:
I close this as resolved, for correct read detailed inverters power U need to type correct envoy password in plugin settings

grzegorz914, 2021-03-27:
For reset envoy password contact Enphase support, the password can by reset to default online

closed 2021-03-27 by grzegorz914

dannydeut, 2021-03-27:
Thanks again. Will do! Enjoy your weekend!