grzegorz914/homebridge-enphase-envoy #33: Cannot read property 'updateValue' of undefined

YouCanNotBeSerious, 2021-07-03:
I have only found and tried this plugin today - i have this error.  What other info can i gather to help troubleshoot it?

```
[04/07/2021, 08:38:01] [homebridge-enphase-envoy] Device: Envoy.local Envoy, update Device state error: TypeError: Cannot read property 'updateValue' of undefined
    at Service.enphaseEncharge.Service.updateCharacteristic (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Service.ts:465:34)
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:2335:18)
[04/07/2021, 08:38:48] [homebridge-enphase-envoy] Device: Envoy.local Envoy, requesting devices info.
[04/07/2021, 08:38:49] [homebridge-enphase-envoy] Device Envoy.local Envoy, debug infoData: <?xml version='1.0' encoding='UTF-8'?>
<envoy_info>
  <time>1625344729</time>
  <device>
    <sn>121548040020</sn>
    <pn>800-00552-r01</pn>
    <software>R5.0.54</software>
    <euaid>4c8675</euaid>
    <seqnum>0</seqnum>
    <apiver>1</apiver>
    <imeter>true</imeter>
  </device>
  <package name='rootfs'>
    <pn>500-00001-r01</pn>
    <version>02.00.00</version>
    <build>950</build>
  </package>
  <package name='kernel'>
    <pn>500-00011-r01</pn>
    <version>04.01.15</version>
    <build>8f3564</build>
  </package>
  <package name='boot'>
    <pn>590-00018-r01</pn>
    <version>02.00.01</version>
    <build>426697</build>
  </package>
  <package name='app'>
    <pn>500-00002-r01</pn>
    <version>05.00.54</version>
    <build>7c95f4</build>
  </package>
  <package name='devimg'>
    <pn>500-00005-r01</pn>
    <version>01.02.293</version>
    <build>6ceafd</build>
  </package>
  <package name='geo'>
    <pn>500-00008-r01</pn>
    <version>02.01.22</version>
    <build>2faa48</build>
  </package>
  <package name='backbone'>
    <pn>500-00010-r01</pn>
    <version>05.00.02</version>
    <build>4fe435</build>
  </package>
  <package name='meter'>
    <pn>500-00013-r01</pn>
    <version>03.02.07</version>
    <build>276642</build>
  </package>
  <package name='agf'>
    <pn>500-00012-r01</pn>
    <version>02.02.00</version>
    <build>fa5320</build>
  </package>
  <package name='security'>
    <pn>500-00016-r01</pn>
    <version>02.00.00</version>
    <build>54a6dc</build>
  </package>
  <package name='full'>
[... 548 lines cut ...]
  db_percent_full: '8',
  timezone: 'Pacific/Auckland',
  current_date: '07/04/2021',
  current_time: '08:39',
  network: [Object],
  tariff: 'time_of_use',
  comm: [Object],
  alerts: [],
  update_status: 'satisfied'
}, inventoryData: [ [Object], [Object], [Object] ]
[04/07/2021, 08:39:55] [homebridge-enphase-envoy] Debug productionData: {
  wattHoursToday: 14,
  wattHoursSevenDays: 75596,
  wattHoursLifetime: 36678938,
  wattsNow: 46
}, productionCtData: { production: [Array], consumption: [Array], storage: [Array] }, meterReadingData: [ [Object], [Object] ]
[04/07/2021, 08:39:55] [homebridge-enphase-envoy] Device: Envoy.local Envoy, savedProductionPowerMax: 0.065749 kW
[04/07/2021, 08:39:55] [homebridge-enphase-envoy] Device: Envoy.local Envoy, productionPowerMaxFile saved successful in: /var/lib/homebridge/enphaseEnvoy/productionPowerMax_Envoylocal 0.069802 kW
[04/07/2021, 08:39:55] [homebridge-enphase-envoy] Device: Envoy.local Envoy, savedProductionPowerMax: 2.231176 kW
[04/07/2021, 08:39:55] [homebridge-enphase-envoy] Device: Envoy.local Envoy, savedProductionPowerMax: 2.231176 kW
[04/07/2021, 08:39:55] [homebridge-enphase-envoy] Device: Envoy.local Envoy, update Device state error: TypeError: Cannot read property 'updateValue' of undefined
    at Service.enphaseEncharge.Service.updateCharacteristic (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Service.ts:465:34)
    at envoyDevice.updateDeviceState (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:2335:18)

```

YouCanNotBeSerious, 2021-07-03:
Maybe this is the relevant part - something about battery SoC?
```

[04/07/2021, 09:08:52] [homebridge-enphase-envoy] This plugin threw an error from the characteristic 'Percent full': Unhandled error thrown inside read handler for characteristic: Cannot read property '6' of undefined. See https://git.io/JtMGR for more info.
[04/07/2021, 09:08:52] [homebridge-enphase-envoy] TypeError: Cannot read property '6' of undefined
    at Characteristic.enphaseEnchargePercentFull.getHandler (/usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:3309:53)
    at Characteristic.enphaseEnchargePercentFull.<anonymous> (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1450:32)
    at step (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:143:27)
    at Object.next (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:124:57)
    at /usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:117:75
    at new Promise (<anonymous>)
    at Object.__awaiter (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:113:16)
    at Characteristic.enphaseEnchargePercentFull.Characteristic.handleGetRequest (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/dist/lib/Characteristic.js:656:24)
    at Characteristic.enphaseEnchargePercentFull.<anonymous> (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2103:22)
    at step (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:143:27)
```

YouCanNotBeSerious, 2021-07-04:
Hi - i updated to fix #33, now the error is:

[05/07/2021, 10:03:17] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Percent full': characteristic value expected valid finite number and received "58,58,57,55,55,56,60" (object). See https://git.io/JtMGR for more info.
[05/07/2021, 10:03:17] [homebridge-enphase-envoy] Error: 
    at Characteristic.enphaseEnchargePercentFull.Characteristic.characteristicWarning (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2038:105)
    at Characteristic.enphaseEnchargePercentFull.Characteristic.validateUserInput (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1931:16)
    at Characteristic.enphaseEnchargePercentFull.<anonymous> (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1456:24)
    at step (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:143:27)
    at Object.next (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:124:57)
    at fulfilled (/usr/local/lib/node_modules/homebridge/node_modules/tslib/tslib.js:114:62)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)

grzegorz914, 2021-07-04:
Ok, I have finally fixed this but right now have no possibility publish it to npm (I’m on holidays)

YouCanNotBeSerious, 2021-07-04:
Hey man I get a free and very excellent plugin, I have no right to say anything other than thank you, and i look forward to it when you get around to publishing :-)

grzegorz914, 2021-07-09:
Try update and let me know

YouCanNotBeSerious, 2021-07-09:
Seems fine now, no errors in log, values in app are correct.....thanks!

commit a482903e (2021-07-04): fix #33
commit 6649f55c (2021-07-04): fix #33, update envoys part
commit 661a9895 (2021-07-04): again fix #33
commit b985979a (2021-07-05): next changes to fix #33
closed 2021-07-10 by grzegorz914