grzegorz914/homebridge-enphase-envoy #34: DEPTH_ZERO_SELF_SIGNED_CERT

jclendon, 2021-07-12:
Hi,

Great plugin thanks.

I've tried installing it and get an error. 'DEPTH_ZERO_SELF_SIGNED_CERT' 

I'm using Homebridge v1.3.4 with Node.js Version v14.17.2 on Raspbian GNU/Linux Buster (10).

The log file shows...

Device: 192.168.XXX.XXX XXXXXXXXXX, requesting devices info eror: Error: self signed certificate
    at TLSSocket.onConnectSecure (_tls_wrap.js:1514:34)
    at TLSSocket.emit (events.js:375:28)
    at TLSSocket._finishInit (_tls_wrap.js:936:8)
    at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:708:12) {
  code: 'DEPTH_ZERO_SELF_SIGNED_CERT',
  config: [Object],
  request: [Writable],
  response: undefined,
  isAxiosError: true,
  toJSON: [Function: toJSON]
}, state: Offline trying to reconnect.

It seems to cycle around and fail to connect to the envoy.

Any thoughts on what I might have wrong?

Thanks

grzegorz914, 2021-07-12:
Please post Your config here. Looks like this is specific problem of Your network configuration.

jclendon, 2021-07-12:
I will clean out the passwords etc and post asap. I think it might have something to do with the local connection to the envoy. The local page now requires authentication either with a token or login with Enphase. I do recall this in the past. The web page shows...

[image]

grzegorz914, 2021-07-12:
What a Envoy model You have installed?

jclendon, 2021-07-12:
Envoy-S-Metered-EU. SKU: ENV-S-WM-230

The part number is 800-00554-r03

grzegorz914, 2021-07-13:
Must be workin, with this device, can U check Envoy Firmware Version? 
Looks like Your Envoy uses Special authentication method.

jclendon, 2021-07-13:
Software Version D7.0.43 (8ed8ea)
Software Build Date 28 May, 2021 6:21 PM

I had it working a while ago but upgraded my version of Raspbian from Stretch to Buster and installed the latest version of Homebridge so I'm not sure if this problem is due to a change in the authentication method or something else.

grzegorz914, 2021-07-13:
Your software I very new, the plug-in was tested to firmware D5.0.55.
Must be something new with authentication in this version.

it will be nice to enable debug mode and post here the complete log from Envoy, I can check iahe api response and look for something new added in this firmware.

jclendon, 2021-07-13:
Thanks grzegorz914,

Running HomeBridge in debug I only get the same info that comes up every 30sec when it tries to connect. Do you mean run your homebridge-enphase-envoy plugin in debug? I'm just using the Homebridge UI.

[13/07/2021, 7:20:58 pm] [enphaseEnvoy] Device: envoy.local Archers21, requesting devices info.
[13/07/2021, 7:20:59 pm] [enphaseEnvoy] Device: envoy.local Archers21, requesting devices info eror: Error: self signed certificate
    at TLSSocket.onConnectSecure (_tls_wrap.js:1514:34)
    at TLSSocket.emit (events.js:375:28)
    at TLSSocket._finishInit (_tls_wrap.js:936:8)
    at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:708:12) {
  code: 'DEPTH_ZERO_SELF_SIGNED_CERT',
  config: [Object],
  request: [Writable],
  response: undefined,
  isAxiosError: true,
  toJSON: [Function: toJSON]
}, state: Offline trying to reconnect.

grzegorz914, 2021-07-13:
OK
loks like the new software uses new authentication method,  I have on my Envoy-S firmware D5.0.55 which’s working fine.

Tuckie, 2021-07-14:
I would guess this is related to: https://community.enphase.com/s/question/0D53m00006ySLuRCAW/unimpressed-with-loss-of-local-api-connectivity-to-envoys

grzegorz914, 2021-07-14:
Right now Enphase do not provide any documentation about the new authentication  method. 
Maybe the firmware cab be downgraded using Installer Toolkit.

enkrypt3d, 2022-01-02:
Following issue... having the same problem.

grzegorz914, 2022-05-25:
Support and test for new authentication method will not be added at this time.

closed 2022-05-25 by grzegorz914