grzegorz914/homebridge-enphase-envoy #35: No accessories displayed in HomeBridge or HomeKit

vaficionado, 2021-08-04:
Hi there! I installed this plugin via Homebridge UI X without a hitch, and can verify that all of my Envoy devices and IQ7 Microinverters are discovered and show their status regularly in the Homebridge logs. I also manually added the Envoy device to my HomeKit instance per the instructions, but it always shows up as  "Not Supported", and no accessories are generated from it.

What additional data or logs can I provide to help see why this might happen? I know the credentials are good (both for `envoy` and `installer`), communications are working and the devices are discovered (HB logs confirm), etc.

Thanks so much!

vaficionado, 2021-08-04:
Example log excerpt:
`[04/08/2021, 15:27:16] [enphaseEnvoy] Device: 192.168.<redacted> Envoy, microinverter: <redacted> last power: 164 W
[04/08/2021, 15:27:16] [enphaseEnvoy] Device: 192.168.<redacted> Envoy, microinverter: <redacted> max power: 251 W
[04/08/2021, 15:27:16] [enphaseEnvoy] Device: 192.168.<redacted> Envoy, microinverter: <redacted> producing: Yes
[04/08/2021, 15:27:16] [enphaseEnvoy] Device: 192.168.<redacted> Envoy, microinverter: <redacted> communicating: Yes
[04/08/2021, 15:27:16] [enphaseEnvoy] Device: 192.168.<redacted> Envoy, microinverter: <redacted> provisioned: Yes
[04/08/2021, 15:27:16] [enphaseEnvoy] Device: 192.168.<redacted> Envoy, microinverter: <redacted> operating: No
[04/08/2021, 15:27:16] [enphaseEnvoy] Device: 192.168.<redacted> Envoy, microinverter: <redacted> comm. level: 0
[04/08/2021, 15:27:16] [enphaseEnvoy] Device: 192.168.<redacted> Envoy, microinverter: <redacted> status: Normal
[04/08/2021, 15:27:16] [enphaseEnvoy] Device: 192.168.<redacted> Envoy, microinverter: <redacted> firmware: 520-00082-r01-v04.18.02
[04/08/2021, 15:27:16] [enphaseEnvoy] Device: 192.168.<redacted> Envoy, microinverter: <redacted> last report: 04/08/2021, 15:20:19
`

Screenshot of what shows up in HomeKit (no accessories show at all in Homebridge)

[image]

grzegorz914, 2021-08-05:
For best experiences please use Controller App or EVE app for iOS. 
The Home.app doesn’t support custom characteristics.

closed 2021-08-13 by grzegorz914