grzegorz914/homebridge-enphase-envoy #36: Does not work with Envoy-S Firmware D7.0.43 (8ed8ea)

mikeykau, 2021-08-07:
Since Envoy auto updated to Firmware D7.0.43 (8ed8ea) the Envoy now only accepts https connections and the authentication method has changed.

Part Number
800-00554-r03
Software Version
D7.0.43 (8ed8ea)
Software Build Date
28 May, 2021 4:21 PM

grzegorz914, 2021-08-07:
Yes, I know this problem was reported already #34 
As long as Enphase do not provide any info about this new authentication method I cannot I fix this.

cscott365, 2021-08-20:
https://community.enphase.com/s/question/0D53m00006ySLuR/unimpressed-with-loss-of-local-api-connectivity-to-envoys

A discussion on the loss of local API access and being pushed to the cloud. Would be good to get anyone using local API to call support and point out how much we want access to OUR DATA.

grzegorz914, 2021-08-22:
@cscott365 it is related to #40

thk-socal, 2021-08-23:
Not related at all. My system has not updated to latest firmware yet.

grzegorz914, 2022-05-25:
Support and test for new authentication method will not be added at this time.

closed 2022-05-25 by grzegorz914