grzegorz914/homebridge-enphase-envoy #37: Not All Components Detected

cscott365, 2021-08-09:
Currently not detecting all of my components:
[09/08/2021, 15:22:19] [homebridge-enphase-envoy] Inverters: 24
[09/08/2021, 15:22:19] [homebridge-enphase-envoy] Batteries: 0
[09/08/2021, 15:22:19] [homebridge-enphase-envoy] --------------------------------
[09/08/2021, 15:22:19] [homebridge-enphase-envoy] Meters: Yes
[09/08/2021, 15:22:19] [homebridge-enphase-envoy] Production: Enabled
[09/08/2021, 15:22:19] [homebridge-enphase-envoy] Consumption: Enabled
[09/08/2021, 15:22:19] [homebridge-enphase-envoy] --------------------------------
[09/08/2021, 15:22:19] [homebridge-enphase-envoy] Ensemble: No
[09/08/2021, 15:22:19] [homebridge-enphase-envoy] --------------------------------

I have 6 Encharge batteries and my system is Ensemble. Envoy, Enpower, Encharge

Any thoughts on how to get it to detect all of the components?

cscott365, 2021-08-09:
Added all my SKUs to the requested documentation phase to see if that would help you as well.

grzegorz914, 2021-08-10:
Please enable debug mode and get the log please.

grzegorz914, 2021-08-13:
It will be very helpful if You run this 3 links on Your local network and get the output here:
http://envoy.local/home.json
http://envoy.local/ivp/ensemble/inventory
http://envoy.local/ivp/ensemble/status

cscott365, 2021-08-15:
The "inventory" and "status" links are requesting a sign-in username and password to access.

grzegorz914, 2021-08-15:
Yes, user: installer password: installer password need to be generated https://thecomputerperson.wordpress.com/2016/08/28/reverse-engineering-the-enphase-installer-toolkit/

cscott365, 2021-08-15:
Ah, username is installer. I already had the password as part of the plugin install. Give me just a bit.

cscott365, 2021-08-15:
Looking at the details, I will need to go through and remove all the serial numbers, as I will not supply those to you. Is there anything specific you are looking for?

grzegorz914, 2021-08-15:
I want  check the api response and compare with my code, the serial numbers can be removed. I don’t have ensemble and write the code basis on user response.

cscott365, 2021-08-15:
[home-upload.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/6988724/home-upload.txt)
[inventory-upload.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/6988725/inventory-upload.txt)
[status-upload.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/6988726/status-upload.txt)

grzegorz914, 2021-08-15:
Thanks, please update to 5.2.1 and get feedback

cscott365, 2021-08-15:
Updated to 5.2.2 and have the following error: 

Device: 192.168.7.200 Envoy-13012, enpowerStatusData error: Error: Request failed with status code 401
    at createError (/usr/lib/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/createError.js:16:15)
    at settle (/usr/lib/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/settle.js:17:12)
    at IncomingMessage.handleStreamEnd (/usr/lib/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/adapters/http.js:260:11)
    at IncomingMessage.emit (events.js:412:35)
    at endReadableNT (internal/streams/readable.js:1317:12)
    at processTicksAndRejections (internal/process/task_queues.js:82:21) {
  config: [Object],
  request: [ClientRequest],
  response: [Object],
  isAxiosError: true,
  toJSON: [Function: toJSON]
}

Also noticed that the URLs you asked get forwarded to a different URL.

Home remains the same, but the others end up at:
http://envoy.local/ivp/ensemble/inventory
http://envoy.local/ivp/ensemble/status

grzegorz914, 2021-08-15:
401 authorization error, check please that you typed correct installer password in config

cscott365, 2021-08-15:
Definitely correct, as I use it to access the JSON files you asked for. I will remove the config and recreate it all from scratch to make sure it is all clear.

grzegorz914, 2021-08-15:
Fixed in 5.2.3

cscott365, 2021-08-15:
Past that error, new one is: 
Device: 192.168.7.200 Envoy-IQ, update Device state error: TypeError: Cannot read property 'secctrl' of undefined
    at envoyDevice.updateDeviceState (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:3938:60)

grzegorz914, 2021-08-15:
OK 5.2.4 ready

cscott365, 2021-08-15:
New line for error: 
Device: 192.168.7.200 Envoy-IQ, update Device state error: TypeError: Cannot read property 'secctrl' of undefined
    at envoyDevice.updateDeviceState (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:3938:66)

grzegorz914, 2021-08-15:
This time must working 5.2.5

cscott365, 2021-08-15:
Nope, not happy with that line still:

Device: 192.168.7.200 Envoy-IQ, update Device state error: TypeError: Cannot read property 'freq_bias_hz' of undefined
    at envoyDevice.updateDeviceState (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:3938:64)

cscott365, 2021-08-15:
Check line 2920:

      const ensembleStatusData = this.ensembleInventoryData;

I think that is not what you want.

grzegorz914, 2021-08-15:
That is it, 5.2.7 ready

cscott365, 2021-08-15:
Past that, now the next set of errors start here:

Device: 192.168.7.200 Envoy-IQ, update Device state error: TypeError: Cannot read property 'updateCharacteristic' of undefined
    at envoyDevice.updateDeviceState (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:3893:16)

grzegorz914, 2021-08-15:
5.2.8 ready

cscott365, 2021-08-15:
Few more to work through, these first ones popped right at the start, the others repeat

Device: 192.168.7.200 Envoy-IQ, update Device state error: ReferenceError: i is not defined
    at envoyDevice.prepareAccessory (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:5096:136)
    at envoyDevice.updateDeviceState (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:3990:71)

Device: 192.168.7.200 Envoy-IQ, update Device state error: TypeError: Cannot read property 'updateCharacteristic' of undefined
    at envoyDevice.updateDeviceState (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:3894:16)

grzegorz914, 2021-08-15:
5.2.10 ready

grzegorz914, 2021-08-15:
5.2.11 fixed enpower status data

cscott365, 2021-08-15:
Appreciate all the work on this...next errors are: 

Device: 192.168.7.200 Envoy-IQ, update Device state error: TypeError: Cannot read property 'UUID' of undefined
    at new enphaseEnpowerStatusService (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:2240:12)
    at envoyDevice.prepareAccessory (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:5198:43)
    at envoyDevice.updateDeviceState (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:3990:71)

Device: 192.168.7.200 Envoy-IQ, update Device state error: TypeError: Cannot read property 'updateValue' of undefined
    at envoyDevice.updateDeviceState (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:3904:16)

grzegorz914, 2021-08-15:
Fixed 5.2.12

grzegorz914, 2021-08-15:
Other fix 5.2.13

grzegorz914, 2021-08-15:
5.2.14 next fixes

cscott365, 2021-08-15:
Device: 192.168.7.200 Envoy-IQ, update Device state error: TypeError: Cannot read property 'updateValue' of undefined
    at envoyDevice.updateDeviceState (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:3906:16)

This is with 5.2.13, will check 5.2.14 shortly.

cscott365, 2021-08-15:
and

Device: 192.168.7.200 Envoy-IQ, update Device state error: TypeError: Cannot read property 'updateValue' of undefined
    at enphaseEnpowerService.Service.updateCharacteristic (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Service.ts:465:34)
    at envoyDevice.updateDeviceState (/usr/lib/node_modules/homebridge-enphase-envoy/index.js:3906:16)

grzegorz914, 2021-08-15:
Try 5.2.14

cscott365, 2021-08-15:
Information is flowing into Homebridge. I was expecting to see an item in the Homebridge Accessories screen, but perhaps that does not happen. There was one warning from Homebridge and I have moved this to a child bridge, as I do with all my plugins, 

This plugin generated a warning from the characteristic 'Status': characteristic value expected string and received undefined. See https://git.io/JtMGR for more info.

cscott365, 2021-08-15:
Data is flowing and looking good, closing this issue and thanks for all the great work!

closed 2021-08-15 by cscott365

grzegorz914, 2021-08-16:
Use Controller app or EVE app to display all data