grzegorz914/homebridge-enphase-envoy #40: Ensemble Detection and Encharge Status Gone in 5.3.0

cscott365, 2021-08-17:
No longer detected as Ensemble and the Encharge batteries are no longer in being reported.

grzegorz914, 2021-08-18:
try 5.3.1-beta7

cscott365, 2021-08-18:
No go with 5.3.1 beta 17

grzegorz914, 2021-08-18:
can You get the debug log?

grzegorz914, 2021-08-18:
try beta 19

cscott365, 2021-08-18:
installing and trying....one separate question for you. what is the value "max power" that comes back for my microinverters? is that what they are configured for or the maximum they have received?

grzegorz914, 2021-08-18:
beta 20 fixed the issue

grzegorz914, 2021-08-18:
max power is the peak power over the day from the microinverter

cscott365, 2021-08-18:
thank you for that...still not getting the encharges or the detection for ensemble....let me see about pulling the log data for you.

cscott365, 2021-08-18:
they are not even showing up in the inventory, is there a debug feature for your plugin I can turn on to get additional logs?

grzegorz914, 2021-08-18:
Yes, enable debug mode in HB and restart it

grzegorz914, 2021-08-23:
Any log?

thk-socal, 2021-08-24:
^[[37m[23/08/2021, 16:54:35] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDevice: 192.168.7.200 Envoy-S, requesting homeData.^[[39m
^[[37m[23/08/2021, 16:54:43] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDevice 192.168.7.200 Envoy-S, debug homeData: {^[[39m
^[[37m[23/08/2021, 16:54:43] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDevice: 192.168.7.200 Envoy-S, requesting powerModeData.^[[39m
^[[37m[23/08/2021, 16:54:43] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDebug powerModeData: { powerForcedOff: false }^[[39m
^[[37m[23/08/2021, 16:54:43] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDevice: 192.168.7.200 Envoy-S, requesting inventoryData.^[[39m
^[[37m[23/08/2021, 16:54:45] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDevice 192.168.7.200 Envoy-S, debug metersData: [ [Object], [Object], [Object] ]^[[39m
^[[37m[23/08/2021, 16:54:45] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDevice: 192.168.7.200 Envoy-S, requesting metersData.^[[39m
^[[37m[23/08/2021, 16:54:45] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDevice 192.168.7.200 Envoy-S, debug metersData: [ [Object], [Object] ]^[[39m
^[[37m[23/08/2021, 16:54:45] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDevice: 192.168.7.200 Envoy-S, requesting productionData.^[[39m
^[[37m[23/08/2021, 16:54:45] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDebug productionData: {^[[39m
^[[37m[23/08/2021, 16:54:45] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDevice: 192.168.7.200 Envoy-S, requesting productionCtData.^[[39m
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDebug productionCtData: { production: [Array], consumption: [Array], storage: [Array] }^[[39m
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDevice: 192.168.7.200 Envoy-S, requesting metersReadingData.^[[39m
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDebug metersReadingData: [ [Object], [Object] ]^[[39m
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDevice: 192.168.7.200 Envoy-S, requesting getDeviceInfo.^[[39m
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m -------- Envoy-S --------
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m Manufacturer: Enphase
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m Model: X-IQ-AM1-240-3-ES
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m Firmware: D6.0.98
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m SerialNr: <redacted>
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m Time: 23/08/2021, 16:54:30
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ------------------------------
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m Q-Relays: 0
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m Inverters: 24
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m Batteries: 0
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m --------------------------------
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m Meters: Yes
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m Production: Enabled
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m Consumption: Enabled
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m --------------------------------
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m Ensemble: Not installed
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m --------------------------------
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDevice: 192.168.7.200 Envoy-S, update device state.^[[39m
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDevice: 192.168.7.200 Envoy-S, savedProductionPowerMax: 7.09279 kW^[[39m
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDevice: 192.168.7.200 Envoy-S, savedProductionPowerMax: 14.114968000000001 kW^[[39m
^[[37m[23/08/2021, 16:54:47] ^[[39m^[[36m[homebridge-enphase-envoy]^[[39m ^[[90mDevice: 192.168.7.200 Envoy-S, savedProductionPowerMax: 14.

thk-socal, 2021-08-24:
that is the debug output for initial discovery where the Ensemble is not detected. not sure what logic changed in the inventory process that you created a bit ago, but it is not seeing an Ensemble device anymore since 5.3.0

grzegorz914, 2021-08-24:
please install 5.4.5-beta1 and get me the log, not enable debug mode

closed 2021-08-24 by grzegorz914

grzegorz914, 2021-08-24:
fixed

commit de8843c3 (2021-08-24): fix #40