grzegorz914/homebridge-enphase-envoy #42: Error in Homebridge log

euphguy, 2021-08-30:
I've had this up and running for a few days normally. This morning I have the following error in the logs and no data is flowing to HomeKit. 

Connection to the envoy homepage is slow but it loads. 

[8/30/2021, 9:22:24 AM] [enphaseEnvoy] Device: 192.168.1.177 Envoy Solar, homeData error: Error: Parse Error: Content-Length can't be present with Transfer-Encoding
    at HTTPParser.execute (<anonymous>)
    at Socket.socketOnData (_http_client.js:515:22)
    at Socket.emit (events.js:375:28)
    at addChunk (internal/streams/readable.js:290:12)
    at readableAddChunk (internal/streams/readable.js:265:9)
    at Socket.Readable.push (internal/streams/readable.js:204:10)
    at TCP.onStreamRead (internal/stream_base_commons.js:188:23) {
  bytesParsed: 242,
  code: 'HPE_UNEXPECTED_CONTENT_LENGTH',
  reason: "Content-Length can't be present with Transfer-Encoding",
  rawPacket: <Buffer 48 54 54 50 2f 31 2e 31 20 33 30 31 20 4d 6f 76 65 64 20 50 65 72 6d 61 6e 65 6e 74 6c 79 0d 0a 44 61 74 65 3a 20 4d 6f 6e 2c 20 33 30 20 41 75 67 20 ... 192 more bytes>,
  config: [Object],
  request: [Writable],
  response: undefined,
  isAxiosError: true,
  toJSON: [Function: toJSON]
}

grzegorz914, 2021-08-30:
Check please Your Envoy Firmware #36 , may be was updated over the night?

euphguy, 2021-08-30:
I don't believe it was updated. Version D5.0.55

grzegorz914, 2021-08-30:
Try to restart Envoy or check network Connecrivity

euphguy, 2021-08-30:
Thank you for helping. 

I power cycled the Envoy. I can connect to the local web server (very slowly). I Still have the same error. 

I am going to see if I can improve the wifi connection and I will report back.

euphguy, 2021-08-30:
I've improved the wifi connection slightly and I have also completely removed the plugin and config, restarted the server, added the plugin back in. With debug mode on I see this every 10 seconds: 

[8/30/2021, 4:31:04 PM] [enphaseEnvoy] Device: 192.168.1.177 Envoy Solar, requesting infoData.
[8/30/2021, 4:31:07 PM] [enphaseEnvoy] Device 192.168.1.177 Envoy Solar, debug infoData: <?xml version='1.0' encoding='UTF-8'?>
<envoy_info>
  <time>1630355467</time>
  <device>
    <sn>202123025581</sn>
    <pn>800-00655-r09</pn>
    <software>D5.0.55</software>
    <euaid>4c8675</euaid>
    <seqnum>0</seqnum>
    <apiver>1</apiver>
    <imeter>true</imeter>
  </device>
  <package name='rootfs'>
    <pn>500-00001-r01</pn>
    <version>02.00.00</version>
    <build>950</build>
  </package>
  <package name='kernel'>
    <pn>500-00011-r01</pn>
    <version>04.01.15</version>
    <build>1905ae</build>
  </package>
  <package name='boot'>
    <pn>590-00019-r01</pn>
    <version>02.00.01</version>
    <build>1f421b</build>
  </package>
  <package name='app'>
    <pn>500-00002-r01</pn>
    <version>05.00.55</version>
    <build>4f2662</build>
  </package>
  <package name='devimg'>
    <pn>500-00004-r01</pn>
    <version>01.02.293</version>
    <build>9cf065</build>
  </package>
  <package name='geo'>
    <pn>500-00008-r01</pn>
    <version>02.01.22</version>
    <build>2faa48</build>
  </package>
  <package name='backbone'>
    <pn>500-00010-r01</pn>
    <version>05.00.02</version>
    <build>4fe435</build>
  </package>
  <package name='meter'>
    <pn>500-00013-r01</pn>
    <version>03.02.07</version>
    <build>276642</build>
  </package>
  <package name='agf'>
    <pn>500-00012-r01</pn>
    <version>02.02.00</version>
    <build>8969f6</build>
  </package>
  <package name='security'>
    <pn>500-00016-r01</pn>
    <version>02.00.00</version>
    <build>54a6dc</build>
  </package>
  <build_info>
    <build_time_gmt>1607601617</build_time_gmt>
    <build_id>release-5.0.x-106-Dec-10-20-02:30:12</build_id>
  </build_info>
</envoy_info>

[8/30/2021, 4:31:07 PM] [enphaseEnvoy] Device: 192.168.1.177 Envoy Solar, parse info.xml successful: {
  "envoy_info": {
    "time": [
      "1630355467"
    ],
    "device": [
      {
        "sn": [
          "202123025581"
        ],
        "pn": [
          "800-00655-r09"
        ],
        "software": [
          "D5.0.55"
        ],
        "euaid": [
          "4c8675"
        ],
        "seqnum": [
          "0"
        ],
        "apiver": [
          "1"
        ],
        "imeter": [
          "true"
        ]
      }
    ],
    "package": [
      {
        "$": {
          "name": "rootfs"
        },
        "pn": [
          "500-00001-r01"
        ],
        "version": [
          "02.00.00"
        ],
        "build": [
          "950"
        ]
      },
      {
        "$": {
          "name": "kernel"
        },
        "pn": [
          "500-00011-r01"
        ],
        "version": [
          "04.01.15"
        ],
        "build": [
          "1905ae"
        ]
      },
      {
        "$": {
          "name": "boot"
        },
        "pn": [
          "590-00019-r01"
        ],
        "version": [
          "02.00.01"
        ],
        "build": [
          "1f421b"
        ]
      },
      {
        "$": {
          "name": "app"
        },
        "pn": [
          "500-00002-r01"
        ],
        "version": [
          "05.00.55"
        ],
        "build": [
          "4f2662"
        ]
      },
      {
        "$": {
          "name": "devimg"
        },
        "pn": [
          "500-00004-r01"
        ],
        "version": [
          "01.02.293"
        ],
        "build": [
          "9cf065"
        ]
      },
      {
        "$": {
          "name": "geo"
        },
        "pn": [
          "500-00008-r01"
        ],
        "version": [
          "02.01.22"
        ],
        "build": [
          "2faa48"
        ]
      },
      {
        "$": {
          "name": "backbone"
        },
        "pn": [
          "500-00010-r01"
        ],
        "version": [
          "05.00.02"
        ],
        "build": [
          "4fe435"
        ]
      },
      {
        "$": {
          "name": "meter"
        },
        "pn": [
          "500-00013-r01"
        ],
        "version": [
          "03.02.07"
        ],
        "build": [
          "276642"
        ]
      },
      {
        "$": {
          "name": "agf"
        },
        "pn": [
          "500-00012-r01"
        ],
        "version": [
          "02.02.00"
        ],
        "build": [
          "8969f6"
        ]
      },
      {
        "$": {
          "name": "security"
        },
        "pn": [
          "500-00016-r01"
        ],
        "version": [
          "02.00.00"
        ],
        "build": [
          "54a6dc"
        ]
      }
    ],
    "build_info": [
      {
        "build_time_gmt": [
          "1607601617"
        ],
        "build_id": [
          "release-5.0.x-106-Dec-10-20-02:30:12"
        ]
      }
    ]
  }

grzegorz914, 2021-08-30:
All OK in this log

euphguy, 2021-08-30:
OK, so with debug mode off the plugin never seems to load fully. No mention of the plugin in the logs other than "[enphaseEnvoy] Initializing enphaseEnvoy platform..."

grzegorz914, 2021-08-30:
Can You post comlete debug log?

euphguy, 2021-08-30:
`[8/30/2021, 5:04:07 PM] [HB Supervisor] OS: Linux 5.10.17-v7+ arm
[8/30/2021, 5:04:07 PM] [HB Supervisor] Node.js v14.17.2 /usr/bin/node
[8/30/2021, 5:04:07 PM] [HB Supervisor] Homebridge Path: /usr/lib/node_modules/homebridge/bin/homebridge
[8/30/2021, 5:04:07 PM] [HB Supervisor] UI Path: /usr/lib/node_modules/homebridge-config-ui-x/dist/bin/standalone.js
[8/30/2021, 5:04:45 PM] [Homebridge UI] Homebridge Config UI X v4.41.2 is listening on :: port 8581
[8/30/2021, 5:04:48 PM] [HB Supervisor] Starting Homebridge with extra flags: -I -D
[8/30/2021, 5:04:48 PM] [HB Supervisor] Started Homebridge v1.3.4 with PID: 10600
Initializing HAP-NodeJS v0.9.4...
[8/30/2021, 5:04:53 PM] [Homebridge UI] Starting terminal session
[8/30/2021, 5:04:55 PM] Loaded config.json with 1 accessories and 5 platforms.
[8/30/2021, 5:04:56 PM] Loaded 0 cached accessories from cachedAccessories.
[8/30/2021, 5:04:56 PM] ---
[8/30/2021, 5:05:04 PM] Plugin /usr/lib/node_modules/homebridge-platform-helper package.json does not contain the keyword 'homebridge-plugin'.
[8/30/2021, 5:05:04 PM] Plugin /usr/lib/node_modules/homebridge-weather-plus does not contain a package.json.
[8/30/2021, 5:05:04 PM] Loaded plugin: homebridge-august-smart-locks@5.0.9
[8/30/2021, 5:05:04 PM] Registering platform 'homebridge-august-smart-locks.AugustLocks'
[8/30/2021, 5:05:04 PM] ---
[8/30/2021, 5:05:11 PM] [Homebridge UI] Terminal session ended.
[8/30/2021, 5:05:27 PM] Loaded plugin: homebridge-broadlink-rm-pro@4.4.7-beta.1
[8/30/2021, 5:05:27 PM] Plugin 'homebridge-broadlink-rm-pro' tried to register with an incorrect plugin identifier: 'homebridge-broadlink-rm'. Please report this to the developer!
[8/30/2021, 5:05:27 PM] Registering platform 'homebridge-broadlink-rm-pro.BroadlinkRM'
[8/30/2021, 5:05:27 PM] ---
[8/30/2021, 5:05:28 PM] Loaded plugin: homebridge-config-ui-x@4.41.2
[8/30/2021, 5:05:28 PM] Registering platform 'homebridge-config-ui-x.config'
[8/30/2021, 5:05:28 PM] ---
[8/30/2021, 5:05:28 PM] Loaded plugin: homebridge-enphase-envoy@5.4.12
[8/30/2021, 5:05:28 PM] Registering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[8/30/2021, 5:05:28 PM] ---
[8/30/2021, 5:05:28 PM] Loaded plugin: homebridge-fakeswitch@0.0.3
[8/30/2021, 5:05:28 PM] Registering accessory 'homebridge-fakeswitch.FakeSwitch'
[8/30/2021, 5:05:28 PM] ---
[8/30/2021, 5:05:29 PM] Loaded plugin: homebridge-platform-orbit@1.5.2
[8/30/2021, 5:05:29 PM] Registering platform 'homebridge-platform-orbit.orbit'
[8/30/2021, 5:05:29 PM] ---
[8/30/2021, 5:05:29 PM] Loading 5 platforms...
[8/30/2021, 5:05:29 PM] [Broadlink RM] Initializing BroadlinkRM platform...
[8/30/2021, 5:05:29 PM] [Broadlink RM] [INFO] Automatically discovering Broadlink RM devices.
[8/30/2021, 5:05:29 PM] [Broadlink RM] TV Accessory Ready
[8/30/2021, 5:05:29 PM] [Broadlink RM] HVAC Off Accessory Ready
[8/30/2021, 5:05:29 PM] [Broadlink RM] AC Heat (day) Accessory Ready
[8/30/2021, 5:05:29 PM] [Broadlink RM] AC Heat (night) Accessory Ready
[8/30/2021, 5:05:29 PM] [Broadlink RM] AC Cool (72°) Accessory Ready
[8/30/2021, 5:05:29 PM] [Broadlink RM] AC (night) Accessory Ready
[8/30/2021, 5:05:29 PM] [Broadlink RM] AC (Auto) Accessory Ready
[8/30/2021, 5:05:29 PM] [Broadlink RM] AC Off Accessory Ready
[8/30/2021, 5:05:29 PM] [Broadlink RM] AC Power Accessory Ready
[8/30/2021, 5:05:29 PM] [Broadlink RM] AC Energy Saver Accessory Ready
[8/30/2021, 5:05:29 PM] [Broadlink RM] 
[8/30/2021, 5:05:29 PM] [Broadlink RM] **************************************************************************************************************
[8/30/2021, 5:05:29 PM] [Broadlink RM] You added TVs in your configuration!
[8/30/2021, 5:05:29 PM] [Broadlink RM] Due to a HomeKit limitation you need to add any TVs to the Home app by using the Add Accessory function.
[8/30/2021, 5:05:29 PM] [Broadlink RM] There you'll find your TVs and you can use the same PIN as you using for this HomeBridge instance.
[8/30/2021, 5:05:29 PM] [Broadlink RM] **************************************************************************************************************
[8/30/2021, 5:05:29 PM] [Broadlink RM] 
[8/30/2021, 5:05:29 PM] [Config] Initializing config platform...
[8/30/2021, 5:05:29 PM] [Config] Running in Service Mode
[8/30/2021, 5:05:29 PM] [AugustLocks] Initializing AugustLocks platform...
[8/30/2021, 5:05:29 PM] [AugustLocks] Initializing child bridge 0E:A4:A6:92:84:60
[8/30/2021, 5:05:29 PM] [orbit] Initializing orbit platform...
[8/30/2021, 5:05:29 PM] [orbit] Initializing child bridge 0E:2E:67:A8:B3:69
[8/30/2021, 5:05:29 PM] [enphaseEnvoy] Initializing enphaseEnvoy platform...
[8/30/2021, 5:05:29 PM] Loading 1 accessories...
[8/30/2021, 5:05:29 PM] [Max Power] Initializing FakeSwitch accessory...
[8/30/2021, 5:05:29 PM] [Max Power] Setting switch to false
[8/30/2021, 5:05:30 PM] [enphaseEnvoy] didFinishLaunching
[8/30/2021, 5:05:30 PM] [enphaseEnvoy] Device: 192.168.1.177 Envoy Solar, requesting infoData.
Preparing Advertiser for 'TV C584' using ciao backend!
[8/30/2021, 5:05:30 PM] Initializing platform accessory 'HVAC Off'...
[8/30/2021, 5:05:30 PM] Initializing platform accessory 'AC Heat (day)'...
[8/30/2021, 5:05:30 PM] Initializing platform accessory 'AC Heat (night)'...
[... 1662 lines cut ...]
        },
        "pn": [
          "500-00016-r01"
        ],
        "version": [
          "02.00.00"
        ],
        "build": [
          "54a6dc"
        ]
      }
    ],
    "build_info": [
      {
        "build_time_gmt": [
          "1607601617"
        ],
        "build_id": [
          "release-5.0.x-106-Dec-10-20-02:30:12"
        ]
      }
    ]
  }
}
[8/30/2021, 5:06:55 PM] [orbit] TX `{"event":"ping"}```

grzegorz914, 2021-08-30:
Mus be something wrong on your side, plug-in no report any error. Think what You changed before this stopped working.

euphguy, 2021-08-30:
Oh well, thanks for reading through those logs

closed 2021-08-30 by euphguy

euphguy, 2021-08-30:
After much searching I finally found a fix. Nothing directly to do with the plug-in. For future reference: [The fix was to use the old Toolkit app.](https://community.enphase.com/s/question/0D52G00004a7uhHSAQ/having-problem-with-envoy-home-page-after-software-update)