grzegorz914/homebridge-enphase-envoy #43: Microinverter current power and lifetime power is zero

JimmyDaGeek, 2021-09-22:
Note that all system level data is sane.
Envoy-S 9170
Firmware R4.10.35
Production metering is enabled and readings are reasonable.
All 18 microinverters (IQ7X) are reporting and indicate producing and provisioned, and communicating.
All 18 microinverters show Power 0 W and power max 0W. Enlighten Manager shows both figures per inverter (eg.  181 W, 799kWh)

Also, Voltage in Power and energy production shows up as 80.4 V, vs. 243.2 V (Enlighten Manager Microinverter AC Voltage)

grzegorz914, 2021-09-22:
Please enable debug mode and get the log.

JimmyDaGeek, 2021-09-27:
Regarding issue 1 - sanitized log added

Regarding voltage - After upgrade to current version this weekend, voltage is now 120ish instead of 80ish.  Enphase Envoy still reports 243
[homebridge-log-enphase-2.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/7237818/homebridge-log-enphase-2.txt)
.2

JimmyDaGeek, 2021-09-27:
No - the issue is not fixed.  The Microinverter current power and Microinverters lifetime power is still reporting as zero, despite the status being correct.

Sorry to confuse.  I do not see any debug info in the log related to the Microinverters, just the resulting reported data.

grzegorz914, 2021-09-27:
May be Envoy password is not correct, please check it again. Only if was changed by User, if not leave the Envoy Password field empty.

JimmyDaGeek, 2021-09-27:
I am getting the following (repeating) error sequence... 

```
[9/27/2021, 4:39:27 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Power': characteristic was supplied illegal value: number -13 exceeded minimum of 0. See https://git.io/JtMGR for more info.
[9/27/2021, 4:39:27 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Power': characteristic was supplied illegal value: number -16 exceeded minimum of 0. See https://git.io/JtMGR for more info.
[9/27/2021, 4:39:27 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Power': characteristic was supplied illegal value: number -4 exceeded minimum of 0. See https://git.io/JtMGR for more info.
[9/27/2021, 4:39:27 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Power': characteristic was supplied illegal value: number -5 exceeded minimum of 0. See https://git.io/JtMGR for more info.
[9/27/2021, 4:39:27 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Power': characteristic was supplied illegal value: number -7 exceeded minimum of 0. See https://git.io/JtMGR for more info.
[9/27/2021, 4:39:27 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Power': characteristic was supplied illegal value: number -15 exceeded minimum of 0. See https://git.io/JtMGR for more info.
[9/27/2021, 4:39:27 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Power': characteristic was supplied illegal value: number -30 exceeded minimum of 0. See https://git.io/JtMGR for more info.
[9/27/2021, 4:39:27 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Power': characteristic was supplied illegal value: number -19 exceeded minimum of 0. See https://git.io/JtMGR for more info.
[9/27/2021, 4:39:28 PM] [HB Supervisor] Homebridge Process Ended. Code: null, Signal: SIGBUS
[9/27/2021, 4:39:33 PM] [HB Supervisor] Restarting Homebridge...
[9/27/2021, 4:39:33 PM] [HB Supervisor] Starting Homebridge with extra flags: -I
[9/27/2021, 4:39:33 PM] [HB Supervisor] Started Homebridge v1.3.4 with PID: 2727
Initializing HAP-NodeJS v0.9.4...
```

Yes the values for power do appear, but the Hombridge now restarts each time it reads the data set from the Enphase.

JimmyDaGeek, 2021-09-27:
I reinstalled the prior version and Homebridge is still restarting.  I'll do more troubleshooting on my end, because the restart loop may be a coincidence.  I did turn debug mode on and I'll clear up the issue on my end.

In any case, the range error on the 'Power' characteristic is probably not casing the Homebridge to restart.

JimmyDaGeek, 2021-09-28:
Here is the relevant part of the log for the characteristic 'power' warning.

```
[9/27/2021, 4:43:12 PM] [enphaseEnvoy] Device: 192.168.58.164 Envoy-S, microinverter: 121941163572, last power: 177 W
[9/27/2021, 4:43:12 PM] [enphaseEnvoy] Device: 192.168.58.164 Envoy-S, microinverter: 121941163572, max power: 307 W
[9/27/2021, 4:43:12 PM] [enphaseEnvoy] Device: 192.168.58.164 Envoy-S, microinverter: 121941163572, producing: Yes
[9/27/2021, 4:43:12 PM] [enphaseEnvoy] Device: 192.168.58.164 Envoy-S, microinverter: 121941163572, communicating: Yes
[9/27/2021, 4:43:12 PM] [enphaseEnvoy] Device: 192.168.58.164 Envoy-S, microinverter: 121941163572, provisioned: Yes
[9/27/2021, 4:43:12 PM] [enphaseEnvoy] Device: 192.168.58.164 Envoy-S, microinverter: 121941163572, operating: No
[9/27/2021, 4:43:12 PM] [enphaseEnvoy] Device: 192.168.58.164 Envoy-S, microinverter: 121941163572, comm. level: 0
[9/27/2021, 4:43:12 PM] [enphaseEnvoy] Device: 192.168.58.164 Envoy-S, microinverter: 121941163572, status: Normal
[9/27/2021, 4:43:12 PM] [enphaseEnvoy] Device: 192.168.58.164 Envoy-S, microinverter: 121941163572, firmware: 520-00082-r01-v04.27.04
[9/27/2021, 4:43:12 PM] [enphaseEnvoy] Device: 192.168.58.164 Envoy-S, microinverter: 121941163572 last report: 9/27/2021, 4:31:53 PM
[9/27/2021, 4:43:12 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Power': characteristic was supplied illegal value: number -13 exceeded minimum of 0. See https://git.io/JtMGR for more info.
[9/27/2021, 4:43:12 PM] [homebridge-enphase-envoy] Error: 
    at enphaseMicroinverterPower.Characteristic.characteristicWarning (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2038:105)
    at enphaseMicroinverterPower.Characteristic.validateUserInput (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1946:16)
    at enphaseMicroinverterPower.<anonymous> (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1456:24)
    at step (/usr/lib/node_modules/homebridge/node_modules/tslib/tslib.js:143:27)
    at Object.next (/usr/lib/node_modules/homebridge/node_modules/tslib/tslib.js:124:57)
    at fulfilled (/usr/lib/node_modules/homebridge/node_modules/tslib/tslib.js:114:62)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
[9/27/2021, 4:43:12 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Power': characteristic was supplied illegal value: number -16 exceeded minimum of 0. See https://git.io/JtMGR for more info.
[9/27/2021, 4:43:12 PM] [homebridge-enphase-envoy] Error: 
    at enphaseMicroinverterPower.Characteristic.characteristicWarning (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2038:105)
    at enphaseMicroinverterPower.Characteristic.validateUserInput (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1946:16)
    at enphaseMicroinverterPower.<anonymous> (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1456:24)
    at step (/usr/lib/node_modules/homebridge/node_modules/tslib/tslib.js:143:27)
    at Object.next (/usr/lib/node_modules/homebridge/node_modules/tslib/tslib.js:124:57)
    at fulfilled (/usr/lib/node_modules/homebridge/node_modules/tslib/tslib.js:114:62)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
[9/27/2021, 4:43:12 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Power': characteristic was supplied illegal value: number -4 exceeded minimum of 0. See https://git.io/JtMGR for more info.
[9/27/2021, 4:43:12 PM] [homebridge-enphase-envoy] Error: 
    at enphaseMicroinverterPower.Characteristic.characteristicWarning (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2038:105)
    at enphaseMicroinverterPower.Characteristic.validateUserInput (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1946:16)
    at enphaseMicroinverterPower.<anonymous> (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1456:24)
    at step (/usr/lib/node_modules/homebridge/node_modules/tslib/tslib.js:143:27)
    at Object.next (/usr/lib/node_modules/homebridge/node_modules/tslib/tslib.js:124:57)
    at fulfilled (/usr/lib/node_modules/homebridge/node_modules/tslib/tslib.js:114:62)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
[9/27/2021, 4:43:12 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Power': characteristic was supplied illegal value: number -5 exceeded minimum of 0. See https://git.io/JtMGR for more info.
[9/27/2021, 4:43:12 PM] [homebridge-enphase-envoy] Error: 
    at enphaseMicroinverterPower.Characteristic.characteristicWarning (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2038:105)
    at enphaseMicroinverterPower.Characteristic.validateUserInput (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1946:16)
    at enphaseMicroinverterPower.<anonymous> (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1456:24)
    at step (/usr/lib/node_modules/homebridge/node_modules/tslib/tslib.js:143:27)
[... cut ...]
    at enphaseMicroinverterPower.<anonymous> (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1456:24)
    at step (/usr/lib/node_modules/homebridge/node_modules/tslib/tslib.js:143:27)
    at Object.next (/usr/lib/node_modules/homebridge/node_modules/tslib/tslib.js:124:57)
    at fulfilled (/usr/lib/node_modules/homebridge/node_modules/tslib/tslib.js:114:62)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
```

My restart loop has nothing to do with your code, so no worries there.

grzegorz914, 2021-09-28:
Please report negative Power Walde to the enphase Support, this is knowing Bug in envoy software, need to be updated to version D5.0.55

commit 549cfe8a (2021-09-27): fix #43
commit e2cde1d8 (2021-09-28): fix #43
closed 2021-09-28 by grzegorz914