grzegorz914/homebridge-enphase-envoy #46: Token Test - Not Working

thk-socal, 2022-02-24:
After adding in my token, the following error repeats and the devices are not discovered. I also removed the Envoy completely and recreated in the plugin with just the token and get the same error. I can confirm that the token item was checked and the token used is a valid operating token.

Error in Homebridge Logs: Device: 192.168.7.200 Envoy, probably uses new firmware 7.x.x, please enable authorization token support in plugin settings.

grzegorz914, 2022-02-26:
Thanks for test, can you try 5.6.3-beta.4, it will be very helpful because I have only envoy with firmware 5.x.x and can not test it locally.
Please also post the log if error occurs.

thk-socal, 2022-02-27:
Just pulled the latest beta and you need to change to support self signed certificates: 

Device: 192.168.7.200 Envoy, requesting infoData error: Error: self signed certificate
    at TLSSocket.onConnectSecure (node:_tls_wrap:1530:34)
    at TLSSocket.emit (node:events:520:28)
    at TLSSocket._finishInit (node:_tls_wrap:944:8)
    at TLSWrap.ssl.onhandshakedone (node:_tls_wrap:725:12) {
  code: 'DEPTH_ZERO_SELF_SIGNED_CERT',
  config: [Object],
  request: [Writable],
  response: undefined,
  isAxiosError: true,
  toJSON: [Function: toJSON]
}, trying again

grzegorz914, 2022-02-27:
next beta ready to test

thk-socal, 2022-02-28:
Not yet, and what I have learned is that not all of the previous URLs are available for query in firmware 7: 

Device: 192.168.7.200 Envoy, inventoryData error: Error: Request failed with status code 401
    at createError (/usr/lib/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/createError.js:16:15)
    at settle (/usr/lib/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/settle.js:17:12)
    at IncomingMessage.handleStreamEnd (/usr/lib/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/adapters/http.js:322:11)
    at IncomingMessage.emit (node:events:532:35)
    at endReadableNT (node:internal/streams/readable:1346:12)
    at processTicksAndRejections (node:internal/process/task_queues:83:21) {
  config: [Object],
  request: [ClientRequest],
  response: [Object],
  isAxiosError: true,
  toJSON: [Function: toJSON]
}, trying again.

thk-socal, 2022-02-28:
The following work with delayed data:

https://envoy.local/api/v1/consumption
https://envoy.local/api/v1/production

The inventory URL is not currently available via a bearer token and API request, but you can use the bearer token to login to the web UI and then go to the inventory URL.

grzegorz914, 2022-02-28:
Next beta ready, please report back with log

thk-socal, 2022-02-28:
Back to Self Signed Cert error: 

[2/28/2022, 9:16:14 AM] [homebridge-enphase-envoy] Device: 192.168.7.200 Envoy, inventoryData error: Error: self signed certificate
    at TLSSocket.onConnectSecure (node:_tls_wrap:1530:34)
    at TLSSocket.emit (node:events:520:28)
    at TLSSocket._finishInit (node:_tls_wrap:944:8)
    at TLSWrap.ssl.onhandshakedone (node:_tls_wrap:725:12) {
  code: 'DEPTH_ZERO_SELF_SIGNED_CERT',
  config: [Object],
  request: [Writable],
  response: undefined,
  isAxiosError: true,
  toJSON: [Function: toJSON]
}, trying again.

grzegorz914, 2022-05-25:
Support and test for new authentication method will not be added at this time.

closed 2022-05-25 by grzegorz914