grzegorz914/homebridge-enphase-envoy #48: 401 after update to latest plugin - using envoy-s firmware D5.0.62 (5e57a9)

MchlLyn, 2022-03-29:
Plugin is configured with the installer password (which stills works when browsing to the web interface of the envoy-s). Everything worked up untill the latest plugin update today (29-03-2022), which somehow broke the installer login/authentication.

Logging:

`Device: 10.0.0.222 Envoy-S, microinvertersData error: Error: Request failed with status code 401
    at createError (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/createError.js:16:15)
    at settle (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/settle.js:17:12)
    at IncomingMessage.handleStreamEnd (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/adapters/http.js:322:11)
    at IncomingMessage.emit (node:events:532:35)
    at endReadableNT (node:internal/streams/readable:1346:12)
    at processTicksAndRejections (node:internal/process/task_queues:83:21) {
  config: [Object],
  request: [ClientRequest],
  response: [Object],
  isAxiosError: true,
  toJSON: [Function: toJSON]`

MchlLyn, 2022-03-29:
Ps. via the enlighten toolkit app the firmware D5.0.62 (5e57a9) is up-to-date...there are seemingly no newer firmwares available for my device.

grzegorz914, 2022-03-29:
Hmm, same firmware on my Envoy all working without any error.
Can you post the config.json?

MchlLyn, 2022-03-29:
Yes, it's strange...everything worked, I only updated the plugin and noticed the error. I've also tried by removing the installer password but I'm still getting the 401 so i've put it back.

This is my config (installer pwd removed):

`        {
            "devices": [
                {
                    "name": "Envoy-S",
                    "host": "10.0.0.222",
                    "disableLogInfo": true,
                    "enableDebugMode": false,
                    "installerPasswd": "<redacted>",
                    "entrezAuth": false,
                    "powerProductionMaxDetected": 4500,
                    "energyProductionLifetimeOffset": 0,
                    "powerConsumptionTotalMaxDetected": 0,
                    "energyConsumptionTotalLifetimeOffset": 0,
                    "powerConsumptionNetMaxDetected": 0,
                    "energyConsumptionNetLifetimeOffset": 0,
                    "enableMqtt": false,
                    "mqttAuth": false,
                    "mqttDebug": false
                }
            ],
            "platform": "enphaseEnvoy"
        },`

grzegorz914, 2022-03-29:
Microinverters data need envoyPasswd not installerPasswd, if you don't change it please leave the field in config empty in other case need to be set. Also try to restart plugin or HB again.

MchlLyn, 2022-03-29:
I just reverted to version 5.6.2 and everything works, with or without the installer password.
Then I installed 5.6.3 and it does not work, with or without the installer password.

This is the debug output and it is receiving data, but stills end with the 401 error (sorry for the long output...i do not see an option to attach logfiles):

[3/29/2022, 12:56:32 PM] [enphaseEnvoy] Device: 10.0.0.222 Envoy-S, debug homeData: {
  "software_build_epoch": 1622651920,
  "is_nonvoy": false,
  "db_size": "6 MB",
  "db_percent_full": "1",
  "timezone": "Europe/Brussels",
  "current_date": "03/29/2022",
  "current_time": "12:56",
  "network": {
    "web_comm": true,
    "ever_reported_to_enlighten": true,
    "last_enlighten_report_time": 1648550144,
    "primary_interface": "eth0",
    "interfaces": [
      {
        "type": "ethernet",
        "interface": "eth0",
        "mac": "00:1D:C0:6E:1E:C3",
        "dhcp": true,
        "ip": "10.0.0.222",
        "signal_strength": 1,
        "signal_strength_max": 1,
        "carrier": true
      },
      {
        "signal_strength": 0,
        "signal_strength_max": 0,
        "type": "wifi",
        "interface": "wlan0",
        "mac": "28:EC:9A:A8:47:28",
        "dhcp": true,
        "ip": null,
        "carrier": false,
        "supported": true,
        "present": true,
        "configured": false,
        "status": "connecting"
      }
    ]
  },
  "tariff": "none",
  "comm": {
    "num": 14,
    "level": 5,
    "pcu": {
      "num": 13,
      "level": 5
    },
    "acb": {
      "num": 0,
      "level": 0
    },
    "nsrb": {
      "num": 1,
      "level": 5
    }
  },
  "alerts": [],
  "update_status": "satisfied"
}
[3/29/2022, 12:56:32 PM] [enphaseEnvoy] Device: 10.0.0.222 Envoy-S, debug inventoryData: [
  {
    "type": "PCU",
    "devices": [
      {
[... 376 lines cut ...]
    }
  ],
  "storage": [
    {
      "type": "acb",
      "activeCount": 0,
      "readingTime": 0,
      "wNow": 0,
      "whNow": 0,
      "state": "idle"
    }
  ]
}
[3/29/2022, 12:56:33 PM] [enphaseEnvoy] Device: 10.0.0.222 Envoy-S, microinvertersData error: Error: Request failed with status code 401
    at createError (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/createError.js:16:15)
    at settle (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/settle.js:17:12)
    at IncomingMessage.handleStreamEnd (/homebridge/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/adapters/http.js:322:11)
    at IncomingMessage.emit (node:events:532:35)
    at endReadableNT (node:internal/streams/readable:1346:12)
    at processTicksAndRejections (node:internal/process/task_queues:83:21) {
  config: [Object],
  request: [ClientRequest],
  response: [Object],
  isAxiosError: true,
  toJSON: [Function: toJSON]

grzegorz914, 2022-03-29:
Please try 3.6.4-beta.0 and let me know

MchlLyn, 2022-03-29:
Unfortunately no success, still getting the 401.

grzegorz914, 2022-03-29:
arę you sure You don't changed envoyPasswd?

MchlLyn, 2022-03-29:
Not that I'm aware. My envoy-s was broken and replaced in December. I installed it myself and most definitely did not change any password on the new one.

If it was changed, would the old plugin version 5.6.2 still work?

grzegorz914, 2022-03-29:
OK, probably fixed wit 3.6.4-beta.1

MchlLyn, 2022-03-29:
Yes! 3.6.4-beta.1 fixed it :)
Thanks!

MchlLyn, 2022-03-29:
ps. I've closed this, assuming I should once it's fixed :-)

closed 2022-03-29 by MchlLyn

CassiusPacheco, 2022-03-30:
I was having this issue yesterday and this update fixed, thanks!

On another note, is there a place where I can find more details about which piece of data I should use in order to trigger some automation? For example, "Do X when the production surplus is higher than Y"?

Thanks again for the plugin

commit a9e2dde8 (2022-03-29): fix #48