grzegorz914/homebridge-enphase-envoy #49: How to specific polling time?

CassiusPacheco, 2022-03-30:
Hello,

I noticed that the latest update removed the [refresh interval](https://github.com/grzegorz914/homebridge-enphase-envoy/commit/09176f7103da49d6b2989467e057013eb93cbe65#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5L92), but I was planning on using it in order to ensure my Automations don't end up turning ON and OFF all the time when the production is too volatile, by setting the poll interval to 5 minutes or something like that.

Here's screenshots of my Automation (done in the Controller app), that should only turn the pump ON if I have a surplus of production of 1.1Kw, and turn off when I'm actually importing 0.1Kw, since the pump itself costs around 1Kw, which gives me a margin of 0.2Kw.

Is there any other way I can achieve this behaviour now that the refresh interval is gone, besides making the margin mentioned above much bigger?

Thanks!

[image]

[image]

grzegorz914, 2022-03-30:
Hi @CassiusPacheco 

If you have installed Current Meters in Your installation is better to use data from Meter Production or Consumption, the refresh time here is set to 1,5sec.
Now I have fixed the lagging refresh time for production CT and set it to 3sec and for production Metes to 1,5sec
please try 5.6.5-beta.0 and let me know.

CassiusPacheco, 2022-03-30:
I've installed the beta, but couldnt tell any difference haha.

Are these two segments in the screenshots below the ones you're talking about? Which values do I use to find surplus production (that's usually being exported to the grid), so I can place in the automation?

PS: it's night time in Australia, so no production right now.

Thanks

[image]
[image]

grzegorz914, 2022-03-30:
You can use Consumption Meter >>> Active Power to Control Your Pump.
Foe example If Active Power <= -1.1 then Pump ON else OFF

CassiusPacheco, 2022-03-30:
Perfect, thank you! As a final question, currently there’s no way to set a bigger polling interval, right? Like to 5min to prevent the active power being too volatile and turning the pump on and off too often?

grzegorz914, 2022-03-30:
To prevent the Pump immediately ON/OFF just add timer OFF to Pump OFF automation.
Set polling interval to 5min is not god idea, the data need to be updates as quickly as possible (in real time).

CassiusPacheco, 2022-03-30:
Fair enough, thanks for your work. Do you have a "buy me a coffee" link or something? I'd love to buy you one ☕️

closed 2022-03-30 by CassiusPacheco

grzegorz914, 2022-03-30:
The link is in Plugins section in Homebridge