grzegorz914/homebridge-enphase-envoy #50: No MQTT data

xorguy, 2022-05-10:
Hi.
I have setup the plug-in and it is working but setting MQTT, even in debug mode, I can’t get any data via MQTT nor MQTT debug info at all.

Config part for MQTT is:
`                  "enableMqtt": true,
                    "mqttHost": "host_ip",
                    "mqttPort": 1883,
                    "mqttPrefix": "home/envoy",
                    "mqttAuth": true,
                    "mqttUser": "user",
                    "mqttPasswd": "pass",
                    "mqttDebug": true `

No idea why is not reporting anything via MQTT.

grzegorz914, 2022-05-11:
What is in the log?

xorguy, 2022-05-11:
Log restarting the homebridge:
`[enphaseEnvoy] Initializing enphaseEnvoy platform...
[5/11/2022, 6:16:36 AM] [enphaseEnvoy] Initializing child bridge 0E:DD:65:xx:xx:xx
[5/11/2022, 6:16:40 AM] [homebridge-enphase-envoy] Launched child bridge with PID 19644
[5/11/2022, 6:16:41 AM] Registering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[5/11/2022, 6:16:42 AM] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v5.6.11 child bridge successfully
[5/11/2022, 6:16:42 AM] Homebridge v1.4.1 (HAP v0.10.2) (homebridge-enphase-envoy) is running on port 45496.
[5/11/2022, 6:16:42 AM] [homebridge-enphase-envoy] Device: ip_address Envoy-S, MQTT Connected.
[5/11/2022, 6:16:49 AM] [homebridge-enphase-envoy] -------- Envoy-S --------
[5/11/2022, 6:16:49 AM] [homebridge-enphase-envoy] Manufacturer: Enphase
[5/11/2022, 6:16:49 AM] [homebridge-enphase-envoy] Model: ENV-S-WM-230
[5/11/2022, 6:16:49 AM] [homebridge-enphase-envoy] Firmware: Dx.0xx
[5/11/2022, 6:16:49 AM] [homebridge-enphase-envoy] SerialNr: xxxxxxxx
[5/11/2022, 6:16:49 AM] [homebridge-enphase-envoy] Time: 5/11/2022, 6:16:50 AM
[5/11/2022, 6:16:49 AM] [homebridge-enphase-envoy] ------------------------------
[5/11/2022, 6:16:49 AM] [homebridge-enphase-envoy] Q-Relays: 1
[5/11/2022, 6:16:49 AM] [homebridge-enphase-envoy] Inverters: x
[5/11/2022, 6:16:49 AM] [homebridge-enphase-envoy] Batteries: 0
[5/11/2022, 6:16:49 AM] [homebridge-enphase-envoy] --------------------------------
[5/11/2022, 6:16:49 AM] [homebridge-enphase-envoy] Meters: Yes
[5/11/2022, 6:16:49 AM] [homebridge-enphase-envoy] Production: Enabled
[5/11/2022, 6:16:49 AM] [homebridge-enphase-envoy] Consumption: Enabled
[5/11/2022, 6:16:49 AM] [homebridge-enphase-envoy] --------------------------------
[5/11/2022, 6:16:49 AM] [homebridge-enphase-envoy] Ensemble: Not installed
[5/11/2022, 6:16:49 AM] [homebridge-enphase-envoy] --------------------------------
`
I can see MQTT connected but then there is no data published to topic home/envoy, topic even does not exists.

commit f817ad13 (2022-05-11): fix #50
closed 2022-05-11 by grzegorz914