grzegorz914/homebridge-enphase-envoy #52: Information only updates when home bridge is rebooted

mopac01, 2022-07-31:
I'm getting this error from time to time in the homebridge log since I installed the home bridge-enphanse-envoy plugin yesterday:
[7/31/2022, 9:49:29 AM] [enphaseEnvoy] Device: 172.16.2.107 Envoy-S, microinvertersData error: AxiosError: connect ETIMEDOUT 172.16.2.107:80
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1187:16) {
  port: 80,
  address: '172.16.2.107',
  syscall: 'connect',
  code: 'ETIMEDOUT',
  errno: -60,
  config: [Object],
  request: [Writable]
}
In addition to the above "microinvertersData error" I also get basically the same error except it's a "productionCtData error", "homeData error", "requesting infoData error"

But I also see a multiple lines for each of the 24 microinverters with various pieces of info in the log and if I use Controller or Eve I can see this info for the inverters and the envoy-s.   So clearly the plugin is communicating with my envoy so I've got the IP address set correctly.  That was the only thing I adjusted on the default config.  And it is reporting accurate info on the Envoy as well in the log:
 -------- Envoy-S --------
Manufacturer: Enphase
Model: Envoy
Firmware: R5.0.55
SerialNr: 12162002xxxx
Time: 7/31/2022, 12:53:40 PM
------------------------------
Q-Relays: 0
Inverters: 24
Batteries: 0
--------------------------------
Meters: No
Ensemble: Not installed
--------------------------------

The 24 inverters are all M215s that were installed in 2011.  The original egg shaped gateway had the ethernet port fail in what I believe was connected to a lightning strike not far from the cabinet.   Then I installed the Envoy-S.   

I did try deleting the AccessoryInfo and AccessoryCashe files in the /.homebridge directory and reading the Envoy to HomeKit but that doesn't seem to have changed anything.  I'm not sure if the errors I'm seeing above are "operating as designed" or whether there's a problem with the config.

grzegorz914, 2022-07-31:
Looks like connection time out, check your network connection to the envoy. If you use WLAN check your signal strength.

mopac01, 2022-07-31:
> Looks like connection time out, check your network connection to the envoy. If you use WLAN check your signal strength.

I went out and rebooted the envoy but that didn't change the behavior.  It has a wired ethernet connection from the electrical panel back to inside the house to a switch on the router.  Pinging the device from my WiFi connected laptop I'm getting between 1ms and 10ms per ping.  That shouldn't be causing a timeout should it?

grzegorz914, 2022-07-31:
10ms is OK, ping the envoy from server where you have installed homebridge. Also run this plugin as child bridge . You can also enable debug mode in plugin config and post the complete log after HB restart.

mopac01, 2022-08-01:
1-4ms pings to the envoy from the server where I have installed home bridge.  This is only the 2nd day since I set up home bridge and everything has been pretty much just follow the directions.  I figured out moving enphase plugin to child bridge but now when I add it to Home app it shows up as a bridge.  But once that bridge was added I can't figure out how to add the envoy and the inverters.  Nothing else but the bridge is showing in Eve or Controller apps.  I did see the error in the main Homebridge log that I started this thread with once so far, which is considerably less than I had been seeing it.  But then again I'm not able to refresh any inverter or envoy accessories now.

grzegorz914, 2022-08-01:
Can You enable Beduinów mode in plugin settings, restart plugin and get the complete log?

mopac01, 2022-08-02:
Night before last I tried restarting the entire homebridge (with enphase plugin still a childbridge) and did get the envoy and the 24 inverters to show up in Eve - so both the enphase bridge shows as an accessory and the envoy and inverters show as accessories as well.  At the point I did this the sun was pretty much down but even after it came back up yesterday morning each inverter only ever showed 0W being produced every time I check for the entire day.  I knew this wasn't accurate given it was a sunny day and normal production was being reported on Enphase's Enlighten website.  Last night, again after the sun was pretty much down I restarted the homebridge again with debug mode enabled.  I've included the log below.  Interestingly enough, yesterday's production then showed for each inverter in Eve.  And now almost noon the next day I'm seeing in Eve the current power production showing up for each inverter as "Power".  However "Power Max" seems to be showing the max produced from yesterday later in the afternoon.  So it appears "Power Max" in your plugin is reporting the most recent max power seen for that inverter rather than the max power since the most recent sunrise.   There are 2 other devices in Eve besides the 24 inverters - the "Envoy serial #" and another entitled "Power and Energy Undefined".  That Power and Energy Undefined has a last reported time of about 20 minutes ago yet shows Energy Today of 0.0 kWh and Power of 0.0 kW.  So the inverters seem to be getting updated but not the total system power and energy.  And currently on the Enphase Enlighten website my system is showing today's power as 23W and energy as 5.67 kWh for today.  So I think I'm really close, but don't know why the total system numbers aren't updating like the inverters are.

[0;37m[8/1/2022, 8:52:22 PM] [0m[0;36m[Homebridge UI][0m Homebridge restart request received
[0;37m[8/1/2022, 8:52:23 PM] [0m[0;36m[Homebridge UI][0m UI / Bridge settings have not changed; only restarting Homebridge process
[0;37m[8/1/2022, 8:52:23 PM] [0m[0;36m[Homebridge UI][0m Sending SIGTERM to Homebridge
[37m[8/1/2022, 8:52:23 PM] [39mGot SIGTERM, shutting down Homebridge...
[37m[8/1/2022, 8:52:23 PM] [39mGot SIGTERM, shutting down child bridge process...
[37m[8/1/2022, 8:52:28 PM][0m [36m[HB Supervisor][0m Homebridge Process Ended. Code: 143, Signal: null
[37m[8/1/2022, 8:52:33 PM][0m [36m[HB Supervisor][0m Restarting Homebridge...
[37m[8/1/2022, 8:52:33 PM][0m [36m[HB Supervisor][0m Starting Homebridge with extra flags: -I
[37m[8/1/2022, 8:52:33 PM][0m [36m[HB Supervisor][0m Started Homebridge v1.5.0 with PID: 84699
[37m[8/1/2022, 8:52:48 PM] [39mLoaded config.json with 0 accessories and 3 platforms.
[37m[8/1/2022, 8:52:49 PM] [39mLoaded 0 cached accessories from cachedAccessories.
[37m[8/1/2022, 8:52:49 PM] [39m---
npm WARN config global `--global`, `--local` are deprecated. Use `--location=global` instead.
[37m[8/1/2022, 8:52:55 PM] [39mLoaded plugin: homebridge-config-ui-x@4.50.0
[37m[8/1/2022, 8:52:55 PM] [39mRegistering platform 'homebridge-config-ui-x.config'
[37m[8/1/2022, 8:52:55 PM] [39m---
[37m[8/1/2022, 8:52:55 PM] [39mLoaded plugin: homebridge-dummy@0.6.0
[37m[8/1/2022, 8:52:55 PM] [39mRegistering accessory 'homebridge-dummy.DummySwitch'
[37m[8/1/2022, 8:52:55 PM] [39m---
[37m[8/1/2022, 8:52:57 PM] [39mLoaded plugin: homebridge-enphase-envoy@5.6.13
[37m[8/1/2022, 8:52:57 PM] [39mRegistering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[37m[8/1/2022, 8:52:57 PM] [39m---
[37m[8/1/2022, 8:53:05 PM] [39mLoaded plugin: homebridge-unifi-protect@5.5.4
[37m[8/1/2022, 8:53:05 PM] [39mRegistering platform 'homebridge-unifi-protect.UniFi Protect'
[37m[8/1/2022, 8:53:05 PM] [39m---
[37m[8/1/2022, 8:53:05 PM] [39mLoading 3 platforms...
[37m[8/1/2022, 8:53:05 PM] [39m[36m[UniFi Protect][39m Initializing UniFi Protect platform...
[37m[8/1/2022, 8:53:05 PM] [39m[36m[UniFi Protect][39m No UniFi Protect controllers have been configured.
[37m[8/1/2022, 8:53:05 PM] [39m[36m[enphaseEnvoy][39m Initializing enphaseEnvoy platform...
[37m[8/1/2022, 8:53:05 PM] [39m[36m[enphaseEnvoy][39m Initializing child bridge 0E:67:6D:88:03:4B
Setup Payload:
X-HM://00241K7KL7XSK
Enter this code with your HomeKit app on your iOS device to pair with Homebridge:
[30m[47m                       [49m[39m
[30m[47m    ┌────────────┐     [49m[39m
[30m[47m    │ 346-49-173 │     [49m[39m
[30m[47m    └────────────┘     [49m[39m
[30m[47m                       [49m[39m
[37m[8/1/2022, 8:53:05 PM] [39mHomebridge v1.5.0 (HAP v0.10.2) (Homebridge 866B) is running on port 51959.
[37m[8/1/2022, 8:53:06 PM] [39m[36m[homebridge-enphase-envoy][39m Launched child bridge with PID 84722
[37m[8/1/2022, 8:53:06 PM] [39mRegistering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[37m[8/1/2022, 8:53:06 PM] [39m[36m[homebridge-enphase-envoy][39m Loaded homebridge-enphase-envoy v5.6.13 child bridge successfully
[37m[8/1/2022, 8:53:06 PM] [39mLoaded 0 cached accessories from cachedAccessories.0E676D88034B.
[37m[8/1/2022, 8:53:06 PM] [39mHomebridge v1.5.0 (HAP v0.10.2) (homebridge-enphase-envoy) is running on port 41648.
[37m[8/1/2022, 8:53:07 PM] [39m[36m[homebridge-enphase-envoy][39m Device: 172.16.2.206 Envoy-S, debug infoData: "<?xml version='1.0' encoding='UTF-8'?>\n<envoy_info>\n  <time>1659405185</time>\n  <device>\n    <sn>121620024514</sn>\n    <pn>800-00551-r02 [... cut ...]
[37m[8/1/2022, 8:53:40 PM] [39m[36m[homebridge-enphase-envoy][39m Device: 172.16.2.206 Envoy-S, microinverter: 121143124618, comm. level: 0
[37m[8/1/2022, 8:53:40 PM] [39m[36m[homebridge-enphase-envoy][39m Device: 172.16.2.206 Envoy-S, microinverter: 121143124618, status: DC Power Too Low
[37m[8/1/2022, 8:53:40 PM] [39m[36m[homebridge-enphase-envoy][39m Device: 172.16.2.206 Envoy-S, microinverter: 121143124618, firmware: 520-00040-r01-v03.42.04
[37m[8/1/2022, 8:53:40 PM] [39m[36m[homebridge-enphase-envoy][39m Device: 172.16.2.206 Envoy-S, microinverter: 121143124618 last report: 8/1/2022, 8:30:10 PM

grzegorz914, 2022-08-02:
Hi,
in the log all is correct, all devices are correct identiefed.
Please update to 5.6.15 and let me know

mopac01, 2022-08-03:
I updated to 5.6.15 but system numbers are not correct.  The individual inverters do seem to be.  I've attached an image out of the Eve app and you can see the inverter numbers are similar to those on Enlighten which I've attached a screenshot of the website.  There's timing differences because Enlighten has lag and is only updated at best 15 minutes behind.  But those inverter numbers look right.  However, if you look on the Eve screenshot you'll see it's reporting 0.0 kW of power being generated at present (which is what it's been showing all day and the system has been generating and is still generating at 3pm in the afternoon.  Also the 32.4kWh is what was generated in total YESTERDAY but is showing as "Energy Today".  You can see on the Enlighten screenshot 4.9kW is not the peak power from today either but is showing as "Power max".   ("Power max" on the individual inverters also seems to be a peak on a previous day.)

grzegorz914, 2022-08-03:
This data is reported from your envoy, please update to 5.6.20 enable debug mode and post the log again.

grzegorz914, 2022-08-03:
OK fixed in 5.6.21, let me know after test

closed 2022-08-03 by grzegorz914

mopac01, 2022-08-05:
I went through 2 days to see how things worked from one day to the next.  The system Energy Today and Power (under "Power and Energy Microinverters" accessory) are both updating correctly through the day and resetting correctly overnight.

However, it doesn't appear that "Power Max" under each individual micro inverter accessory or the "Power Max" for the entire system under the "Power and Energy Microinverters" accessory are ever reset.  Is that supposed to be that way so they show the max power the plugin has ever seen?  Wouldn't the max power for the day be more likely informative?  My panels are capable of a bit more than what my M215s can do so on a perfectly sunny day the M215s will all max out.  And once that happens then the max power number would be the same for every inverter and the system as a whole perpetually.    All in all that's not a huge deal.   Thank you so much for tweaking this so it'd work with my system without a battery attached.

One question of curiosity - each inverter accessory has "Communicating" "Operating" "Producing" and "Provisioned" accessory.  And Communicating, Producing and Provisioned go from No overnight to Yes during the day.  Operating is perpetually No for every inverter.   Is the Operating field meant just for a system with a battery attached?

mopac01, 2022-08-06:
5.6.22 did change Power Max to Power Peak but otherwise Power Peak for both the Envoy/System In Total as well for the individual inverters never resets from one day to the next.  I'm also still wondering what "Operating" is supposed to signify?  Communicating, Producing and Provisioned all go from No in overnight hours to Yes during daylight hours. But Operating is always No 24x7.

grzegorz914, 2022-08-06:
1. The Power Peak for Production/Consumption is stored to the file in ./var/lib/Homebridge/enphase/ folder and not resets, may be will be better to reset it every day. I will look at this.
2. The Microinverters Power Peak is stored in the Microinverters itself, so I can not clear it from plugin side.
3. The Microinverters Operating mode are changed on my side between No and Yes, I use IQ7+ Microinverters, may be on yours it’s not supported and only provide response NO.
4. Communicating get YES if micro is communicating with envoy.
5. Producing go to YES if micro producing power.
6. Provisioned go to YES if micro is registered on enphase server.
7. Operating go to YES if micro is ready to work.

grzegorz914, 2022-08-06:
added in 5.7.x

mopac01, 2022-08-06:
Installed the update and got this message:   This plugin threw an error from the characteristic 'Power peak reset': Unhandled error thrown inside read handler for characteristic: value is not defined. See https://homebridge.io/w/JtMGR for more info.

grzegorz914, 2022-08-06:
fixed in 5.7.2

commit 9d860508 (2022-08-03): fix #52