grzegorz914/homebridge-enphase-envoy #57: Self signed certificate error on startup

maj, 2022-10-01:
Hi, 

I'm seeing the following error on startup. Can you advise how I debug further?

Thanks,
Michael

    [10/1/2022, 8:08:25 AM] [homebridge-enphase-envoy] Device: 192.168.6.107 EnphaseIQ, requesting envoyBackboneAppData
     or save envoy id error: Error: self signed certificate, reconnect in 15s.

grzegorz914, 2022-10-01:
What a software is installed in Your Envoy?

maj, 2022-10-01:
Looks like 7.3.75

grzegorz914, 2022-10-01:
So, will not work, firmware 7.x.x and above  uses new Authentication method which is none implemented in my plugin at this time.
More here: https://github.com/grzegorz914/homebridge-enphase-envoy/wiki#know-issues

hugo-leij, 2022-10-13:
Hey @grzegorz914 

As expected, I also run into the same problem. A new installation has just been put down with Firmware: D7.0.88. 

Is it by any chance on your roadmap to develop this?

Thnx, Greetz Hugo.

eleventhaus, 2022-11-02:
How do you manually downgrade firmware and what are the risks?

savabg, 2022-11-08:
@grzegorz914 I would love to offer some assistance, but I am not well versed in node - I have been half-able to get the module to work (have been able to get successful responses for a number of the requests) with some very crude tweaks (one being directly injecting the Bearer token). 

In terms of steps that I have been able to identify to date 

1. Switch this.URL to https instead of http
2. Enable https  module (import https)
3. Allow for self signed certificates in axios.create add
httpsAgent: new https.Agent({  
        rejectUnauthorized: false
      }),

5. Inject Bearer token in the header in the same section
 headers: {
      'Authorization': 'Bearer <token>',
      }
6. Enable cookies and store sessionId cookie as it is needed for the .json requests (Crudely injecting Cookie in after Authorization)

How to obtain bearer token:
GET https://enlighten.enphaseenergy.com/entrez-auth-token?serial_num=<ENVOY_SERIAL_NUM>
How to validate validity of token:
GET https://<ENVOY_HOST>/auth/check_jwt
-upon successful validation sessionId cookie is generated

grzegorz914, 2022-11-08:
I will look in to this, hoverer I don't have device with firmware 7.x.x

grzegorz914, 2022-12-10:
@savabg, @del13r  

I have published v6.0.0-beta.2 to test.
In Authorization section You can enable support for FW.7xx.x and paste your generated token.
This test version may not working correct at this time, it is only to test correct login to envoy.
Please also enable debug mode and  always post the log.

closed 2022-12-10 by grzegorz914