grzegorz914/homebridge-enphase-envoy #63: Beta 6 - test for Envoy with firmware 7.x.x

grzegorz914, 2022-12-10:
I have published v6.0.0-beta to test.
In Authorization section You can enable support for FW.7xx.x and paste your generated token.
This test version may not working correct at this time, it is only to test correct login to envoy.
Please also enable debug mode and always post the log.

This will close #57 #61 #60 #

NoSnow3, 2022-12-10:
I am on V7.0.107 Firmware, Happy to help beta test.
First Project I've attempted to help with so go easy.

Attempted to use Un-Commisioned Token, Feedback:

Device: 192.168.0.41 Envoy-S, inventory error: AxiosError: Request failed with status code 401, reconnect in 15s.

Used token fed via "https://enlighten.enphaseenergy.com/entrez-auth-token?serial_num={}"
(Long Term Tokens)
Feedback:

Device: 192.168.0.41 Envoy-S, inventory error: AxiosError: Request failed with status code 401, reconnect in 15s.

Same errors.

Like I said I'm new not much involved with any sorting coding, Unsure how to correctly find logs.
If anyone can brief me i will certainly find them

From a user sadly missing V5.x.xxx Firmware

grzegorz914, 2022-12-11:
@NoSnow3 
Thanks for test but please enable debug in plugin settings advanced section and post the complete log.  As i see your error is from other section, looks like token login is working.

NoSnow3, 2022-12-11:
[homebridge.log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10201744/homebridge.log.txt)

Hoping this is what you are looking for.

grzegorz914, 2022-12-11:
@NoSnow3 
pushed beta.7 to test, also please test again with the log.

It will be very helpful for me if someone can get me the remote access to Envoy with FW 7.x.x.

NoSnow3, 2022-12-11:
[HomeBridge Beta7 Log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10201787/HomeBridge.Beta7.Log.txt)

New Log...

"Device: 192.168.0.41 Envoy-S, inventory error: Error: self signed certificate, reconnect in 15s."

Regards Remote access. To the actual Envoy? unsure on how we can do that envoy itself

grzegorz914, 2022-12-11:
beta.8 pushed, try again

NoSnow3, 2022-12-11:
[homebridge Beta8 Log.log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10201796/homebridge.Beta8.Log.log.txt)

"inventory error: TypeError: this.digestAuthInstaller is not a function, reconnect in 15s."

grzegorz914, 2022-12-11:
again beta.10

NoSnow3, 2022-12-11:
[homebridge Beta10 Log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10201802/homebridge.Beta10.Log.txt)

"Envoy-S, inventory error: Error: self signed certificate, reconnect in 15s."

grzegorz914, 2022-12-11:
OK, try beta.11

NoSnow3, 2022-12-11:
[homebridge Beta11 Log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10201825/homebridge.Beta11.Log.txt)

Same again,

"inventory error: Error: self signed certificate, reconnect in 15s."

grzegorz914, 2022-12-11:
beta.14 ready

NoSnow3, 2022-12-11:
[homebridge Beta14 Log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10201866/homebridge.Beta14.Log.txt)

"Envoy-S, inventory error: AxiosError: Request failed with status code 401, reconnect in 15s."

Seems to take longer to error this time

grzegorz914, 2022-12-11:
try beta.16

NoSnow3, 2022-12-11:
[homebridge Beta15 Log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10201965/homebridge.Beta15.Log.txt)

"Envoy-S, meters error: AxiosError: Request failed with status code 401, reconnect in 15s."

Beta 15 was all that available

NoSnow3, 2022-12-11:
[homebridge Beta16 Log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10202011/homebridge.Beta16.Log.txt)

"meters error: AxiosError: Request failed with status code 401, reconnect in 15s."

NoSnow3, 2022-12-11:
I Changed token from the shorter timed one that was on the "https://entrez.enphaseenergy.com/entrez_tokens" Link

To my longer Validated Token on "https://enlighten.enphaseenergy.com/entrez-auth-token?serial_num={}"

This time the Log is showing readouts from the envoy but still showing error.

" power mode error: AxiosError: Request failed with status code 401, reconnect in 15s."

[homebridge Beta16.2 Log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10202028/homebridge.Beta16.2.Log.txt)

grzegorz914, 2022-12-11:
ok looks better now, try beta.22

NoSnow3, 2022-12-11:
[homebridge Beta22 Log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10202088/homebridge.Beta22.Log.txt)

Looking better, Thought i had seen an error but its refreshing to fast to catch it

NoSnow3, 2022-12-11:
Ah found it,

"[11/12/2022, 11:57:04] [homebridge-enphase-envoy] Device: 192.168.0.41 Envoy-S, home error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'), reconnect in 15s."

grzegorz914, 2022-12-11:
try beta 23

NoSnow3, 2022-12-11:
Scrap the above as regards to errors

[homebridge Beta23.2 Log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10202230/homebridge.Beta23.2.Log.txt)

"[11/12/2022, 13:01:52] [homebridge-enphase-envoy] Device: 192.168.0.41 Envoy-S, home error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'), reconnect in 15s."

NoSnow3, 2022-12-11:
No errors spotted. 

Still Can not see Devices in Eve or Controller at this moment.

grzegorz914, 2022-12-11:
Arę you added it to the HomeKit app

grzegorz914, 2022-12-11:
you need add this accessory ti the home.app first

NoSnow3, 2022-12-11:
Can confirm i have done this.
[image]

grzegorz914, 2022-12-11:
this is bridge not the accessory itself, look here https://github.com/grzegorz914/homebridge-enphase-envoy/wiki#adding-to-homekit

grzegorz914, 2022-12-11:
beta.25 pushed to test

NoSnow3, 2022-12-11:
Updateded but still can not find the accessory when scanning once the child bridge is added

grzegorz914, 2022-12-11:
Can you post the log after update to b.25

Also try this if can not find the accessory https://github.com/grzegorz914/homebridge-enphase-envoy/wiki#troubleshooting

NoSnow3, 2022-12-11:
Followed Troubleshooting, No results as of yet.
[homebridge Beta 25.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10202505/homebridge.Beta.25.txt)

Log attached.

NoSnow3, 2022-12-11:
[homebridge Beta 26 Log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10202585/homebridge.Beta.26.Log.txt)

Log Attached

grzegorz914, 2022-12-11:
Nice, probably final beta 27 ready

NoSnow3, 2022-12-11:
Downloading, Still yet to see in Eve or controller, Scratching my head on that one

NoSnow3, 2022-12-11:
"Device: 192.168.0.41 Envoy-S, ensemble inventory error: TypeError: Cannot read properties of undefined (reading 'devices'), reconnect in 15s"

[homebridge Beta 27.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10202597/homebridge.Beta.27.txt)

grzegorz914, 2022-12-11:
28 ready

NoSnow3, 2022-12-11:
Might be an ios16 issue hold on

grzegorz914, 2022-12-11:
No,  try this https://github.com/grzegorz914/homebridge-enphase-envoy/wiki#troubleshooting and after this restart whole homebridge then try add accessory again, no need add bridge to home.app

grzegorz914, 2022-12-11:
any log from beta.28

NoSnow3, 2022-12-11:
[homebridge Beta 28 Log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10202724/homebridge.Beta.28.Log.txt)

Same Error again

grzegorz914, 2022-12-11:
ok beta.29 ready

NoSnow3, 2022-12-11:
[homebridge Beta 29 Log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10202779/homebridge.Beta.29.Log.txt)

No immediate errors

grzegorz914, 2022-12-11:
Nice, ready to publish release.
Many thanks for testing and patience.

NoSnow3, 2022-12-11:
No problem, still struggling with displaying in eve/controller

grzegorz914, 2022-12-11:
This must be added to Home.app first

NoSnow3, 2022-12-11:
I can not add just the accessories only the bridge.

grzegorz914, 2022-12-11:
May be try to change the name in plugin Config

grzegorz914, 2022-12-11:
Also try to clear Homebridge cashe

NoSnow3, 2022-12-11:
Cleared cache and added using iOS 15.5. Shows on Eve and Controller now

grzegorz914, 2022-12-11:
Nice.

grzegorz914, 2022-12-11:
v6.0.0 released

closed 2022-12-11 by grzegorz914