grzegorz914/homebridge-enphase-envoy #65: Firmware 7

BenouGui, 2022-12-11:
I have a problem witch firmware 7 
[11/12/2022, 19:31:34] [homebridge-enphase-envoy] Device: 192.168.1.6 Envoy-S, inventory error: AxiosError: Request failed with status code 401, reconnect in 15s.

BenouGui, 2022-12-11:
[homebridge.log-4.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10202973/homebridge.log-4.txt)

grzegorz914, 2022-12-11:
Please generate new Token here https://enlighten.enphaseenergy.com/entrez-auth-token?serial_num={} and ad it to the plugin Config and let me know.

BenouGui, 2022-12-11:
Same problem

grzegorz914, 2022-12-11:
Are you sure you have paste only the data after Token: without apostrophe

BenouGui, 2022-12-11:
[image]
[image]
 
 yes sure

grzegorz914, 2022-12-11:
Here https://github.com/grzegorz914/homebridge-enphase-envoy/issues/63#issuecomment-1345527583  by test was same problem but was fixed after generate new token.

BenouGui, 2022-12-11:
Dont work for me :/

grzegorz914, 2022-12-12:
can you try 6.0.1-beta.0 and let me know

BenouGui, 2022-12-12:
[homebridge.log-8.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10209691/homebridge.log-8.txt)
I retested to hand over my token via both:
https://enlighten.enphaseenergy.com/entrez-auth-token?serial_num=12224xxxxx
and 
https://entrez.enphaseenergy.com

But still the same error

BenouGui, 2022-12-13:
The on/off of the gateway did not change anything, on the internet I see a lot of people using session cookies.
When I try to curl
curl -f -k -H 'Authorization: Bearer TOKENTOKENTOEKN' -X GET https://192.168.1.6/api/v1/production/inverters
curl: (22) The requested URL returned error: 401 Unauthorized
But 
curl -f -k -H 'Authorization: Bearer TOKENTOKENTOEKN' -X GET https://192.168.1.6/api/v1/production/
-- > {
  "wattHoursToday": 0,
  "wattHoursSevenDays": 21534,
  "wattHoursLifetime": 21529,
  "wattsNow": 8
}
But in the web Gui after login witch token. 
[image]

grzegorz914, 2022-12-13:
can you try with CURL this: curl -f -k -H 'Authorization: Bearer TOKENTOKENTOEKN' -X GET https://inventory.json

and let me know the output

grzegorz914, 2022-12-13:
beta.2 enabled cookies, try again please

BenouGui, 2022-12-13:
pi@homebridge:~ $ curl -f -k -H 'Authorization: Bearer TOKENTOKEN' -X GET https://192.168.1.6/inventory.json
curl: (22) The requested URL returned error: 401 Unauthorized.

BenouGui, 2022-12-13:
After login witch token in the web Gui 
[image]

For the beta 2 the same error

BenouGui, 2022-12-13:
look here https://github.com/edasque/enphaseLocalToInfluxDB#authorization-flow

grzegorz914, 2022-12-13:
looks like some bug in envoy firmware, I pushed beta.3 to test, here skipped the inventory data, try it and post the log please

grzegorz914, 2022-12-13:
ups try beta.4

BenouGui, 2022-12-13:
[homebridge.log-10.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10215850/homebridge.log-10.txt)

grzegorz914, 2022-12-13:
arę you try beta.4

BenouGui, 2022-12-13:
yes at the bottom of the log it's beta 4

BenouGui, 2022-12-13:
[homebridge.log-11.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10215906/homebridge.log-11.txt)

[13/12/2022, 09:32:27] [homebridge-enphase-envoy] Device: 192.168.1.6 Envoy-Ss, debug production: {
  "wattHoursToday": 149,
  "wattHoursSevenDays": 21534,
  "wattHoursLifetime": 21675,
  "wattsNow": 520
}
[13/12/2022, 09:32:27] [homebridge-enphase-envoy] Device: 192.168.1.6 Envoy-Ss, production ct error: AxiosError: Request failed with status code 401, reconnect in 15s.

BenouGui, 2022-12-13:
[Uploading homebridge.log-12.txt…]()

BenouGui, 2022-12-13:
[Uploading homebridge.log-13.txt…]()

grzegorz914, 2022-12-13:
ok, try beta.8 and get me the log, I have added some debug to the response

grzegorz914, 2022-12-13:
beta.9 ready

BenouGui, 2022-12-13:
[homebridge.log-15.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10216699/homebridge.log-15.txt)

grzegorz914, 2022-12-13:
try beta.10

BenouGui, 2022-12-13:
[homebridge.log-16.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10217270/homebridge.log-16.txt)

grzegorz914, 2022-12-13:
beta.11 ready

BenouGui, 2022-12-13:
[homebridge.log-17.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10217321/homebridge.log-17.txt)

grzegorz914, 2022-12-13:
try beta.12

BenouGui, 2022-12-13:
[homebridge.log-18.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10217340/homebridge.log-18.txt)

BenouGui, 2022-12-13:
[homebridge.log-19.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10219630/homebridge.log-19.txt)
beta 20

grzegorz914, 2022-12-13:
beta.21

BenouGui, 2022-12-13:
[homebridge.log-20.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10219788/homebridge.log-20.txt)
Beta 21

grzegorz914, 2022-12-13:
try 22

grzegorz914, 2022-12-13:
try 23

BenouGui, 2022-12-13:
[homebridge.log-22.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10219951/homebridge.log-22.txt)
beta 23

grzegorz914, 2022-12-13:
24

BenouGui, 2022-12-13:
[homebridge.log-23.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10219984/homebridge.log-23.txt)
beta 24

grzegorz914, 2022-12-13:
25

BenouGui, 2022-12-13:
[homebridge.log-24.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10220151/homebridge.log-24.txt)

BenouGui, 2022-12-13:
[homebridge.log-25.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10220223/homebridge.log-25.txt)

https://github.com/H4nsie/EnphaseEnvoy

grzegorz914, 2022-12-13:
it working 28

BenouGui, 2022-12-13:
[homebridge.log-26.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10220277/homebridge.log-26.txt)

yes it seems to work, what was the problem?

grzegorz914, 2022-12-13:
try 29 too, need to pas cookies after check jet and set instance

BenouGui, 2022-12-13:
[homebridge.log-27.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10220305/homebridge.log-27.txt)

grzegorz914, 2022-12-13:
nice, thanks for test

NoSnow3, 2022-12-13:
Sorry been AFK.

We all good now?

NoSnow3, 2022-12-13:
[Homebridge 6.0.1 Beta 30.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10221406/Homebridge.6.0.1.Beta.30.txt)

As Requested

grzegorz914, 2022-12-13:
please again 31

NoSnow3, 2022-12-13:
[homebridge 6.0.1 Beta 31.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10221466/homebridge.6.0.1.Beta.31.txt)

"Envoy-S, power mode error: AxiosError: Request failed with status code 401, reconnect in 15s."

grzegorz914, 2022-12-13:
again 32

NoSnow3, 2022-12-13:
[homebridge 6.0.1 Beta 32.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10221505/homebridge.6.0.1.Beta.32.txt)

Nothing spotted

grzegorz914, 2022-12-13:
OK ready for release

Thanks

grzegorz914, 2022-12-13:
v6.0.1 released

closed 2022-12-13 by grzegorz914