grzegorz914/homebridge-enphase-envoy #66: Is there anyway you can implement the batteries to show their info?

HunterEnphase, 2022-12-25:
Here is the address below for the Encharge battery and Enpower status. Thanks.
https://envoy/ivp/ensemble/inventory

grzegorz914, 2022-12-25:
This is already integrated but not tested because I don’t have en Charles and empower, if you want to help please enable debug mode in plugin Config advanced section and get the log after HB restart.

HunterEnphase, 2022-12-28:
I have 3 10T encharge batteries, it should show 9 3T batteries. Hope this helps!

[27/12/2022, 19:42:38] [homebridge-enphase-envoy] Restarting Process...
[27/12/2022, 19:42:39] [homebridge-enphase-envoy] Launched child bridge with PID 31406
[27/12/2022, 19:42:40] Registering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[27/12/2022, 19:42:40] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v6.0.4 child bridge successfully
[27/12/2022, 19:42:40] Loaded 0 cached accessories from cachedAccessories.0E51B87F59B9.
[27/12/2022, 19:42:40] Homebridge v1.6.0 (HAP v0.11.0) (homebridge-enphase-envoy) is running on port 38124.
[27/12/2022, 19:42:40] [homebridge-enphase-envoy] Device: envoy Envoy, debug validate jwt token: <!DOCTYPE html><h2>Valid token.</h2>
, server: openresty/1.17.8.1
date: Wed, 28 Dec 2022 00:42:39 GMT
content-type: application/octet-stream
content-length: 37
connection: keep-alive
set-cookie: sessionId=LiF61mWaIJanvMHXLV4iEvbdx7zTAIca; Secure; HttpOnly; path=/
strict-transport-security: max-age=63072000; includeSubdomains, max-age=63072000; includeSubdomains
x-frame-options: DENY, DENY
x-content-type-options: nosniff, nosniff
[27/12/2022, 19:42:41] [homebridge-enphase-envoy] Device: envoy Envoy, debug info: "<?xml version='1.0' encoding='UTF-8'?>\n<envoy_info>\n  <time>1672188159</time>\n  <device>\n    <sn></sn>\n    <pn>800-00647-r09</pn>\n    <software>D7.3.120</software>\n    <euaid>4c8675</euaid>\n    <seqnum>0</seqnum>\n    <apiver>1</apiver>\n    <imeter>true</imeter>\n  </device>\n  <web-tokens>true</web-tokens>\n  <package name='rootfs'>\n    <pn>500-00001-r01</pn>\n    <version>02.00.00</version>\n    <build>1210</build>\n  </package>\n  <package name='kernel'>\n    <pn>500-00011-r02</pn>\n    <version>04.04.225</version>\n    <build>3eb4d3</build>\n  </package>\n  <package name='boot'>\n    <pn>590-00019-r01</pn>\n    <version>02.00.01</version>\n    <build>1f421b</build>\n  </package>\n  <package name='app'>\n    <pn>500-00002-r01</pn>\n    <version>07.03.120</version>\n    <build>ac5b3c</build>\n  </package>\n  <package name='devimg'>\n    <pn>500-00004-r01</pn>\n    <version>01.02.413</version>\n    <build>9c64cd</build>\n  </package>\n  <package name='geo'>\n    <pn>500-00008-r01</pn>\n    <version>02.01.24</version>\n    <build>a74d96</build>\n  </package>\n  <package name='backbone'>\n    <pn>500-00010-r01</pn>\n    <version>07.00.20</version>\n    <build>176d57</build>\n  </package>\n  <package name='meter'>\n    <pn>500-00013-r01</pn>\n    <version>03.02.08</version>\n    <build>f9ead3</build>\n  </package>\n  <package name='agf'>\n    <pn>500-00012-r01</pn>\n    <version>02.02.00</version>\n    <build>887e04</build>\n  </package>\n  <package name='essimg'>\n    <pn>500-00020-r01</pn>\n    <version>22.10.09</version>\n    <build>983d9b</build>\n  </package>\n  <package name='security'>\n    <pn>500-00016-r01</pn>\n    <version>02.00.00</version>\n    <build>54a6dc</build>\n  </package>\n  <package name='pkgsec'>\n    <pn>500-00021-r01</pn>\n    <version>01.00.00</version>\n    <build>19ae14</build>\n  </package>\n  <build_info>\n    <build_id>ec2-user-envoy_uber-pkg_master:pkg-Oct-13-22-17:54:08</build_id>\n    <build_time_gmt>1665683795</build_time_gmt>\n    <release_ver>02.00.3244</release_ver>\n    <release_stage>700-GA</release_stage>\n  </build_info>\n</envoy_info>\n"
[27/12/2022, 19:42:41] [homebridge-enphase-envoy] Device: envoy Envoy, debug parse info: {
  "envoy_info": {
    "time": [
      "1672188159"
    ],
    "device": [
      {
        "sn": [
          ""
        ],
        "pn": [
          "800-00647-r09"
        ],
        "software": [
          "D7.3.120"
        ],
        "euaid": [
          "4c8675"
        ],
        "seqnum": [
          "0"
        ],
        "apiver": [
          "1"
        ],
        "imeter": [
          "true"
        ]
      }
    ],
    "web-tokens": [
      "true"
    ],
    "package": [
      {
        "$": {
          "name": "rootfs"
        },
        "pn": [
          "500-00001-r01"
        ],
        "version": [
          "02.00.00"
        ],
        "build": [
          "1210"
        ]
      },
      {
        "$": {
          "name": "kernel"
[... 5254 lines cut ...]
        "pwrFactor": 0.017,
        "voltage": 125.01,
        "current": 17.145,
        "freq": 60
      },
      {
        "eid": 1778385427,
        "timestamp": 1672188193,
        "actEnergyDlvd": 0,
        "actEnergyRcvd": 0,
        "apparentEnergy": 0,
        "reactEnergyLagg": 0,
        "reactEnergyLead": 0,
        "instantaneousDemand": 0,
        "activePower": 0,
        "apparentPower": 0,
        "reactivePower": 0,
        "pwrFactor": 0,
        "voltage": 0,
        "current": 0,
        "freq": 60
      }
    ]
  }
]

grzegorz914, 2022-12-28:
Thanks for the log this help me to fully integrate ensemble devices, im pushed 6.0.5-beta.0 to test.
Please check this version and let me know, the log can be attached as file.

[image]

HunterEnphase, 2022-12-28:
Getting this error
Device: envoy Envoy, ensemble inventory error: TypeError: Cannot read properties of undefined (reading 'part_num'), reconnect in 15s.
[homebridge.log (1).txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10315621/homebridge.log.1.txt)

grzegorz914, 2022-12-28:
fixed in beta.4, try again

HunterEnphase, 2022-12-28:
Getting this error now
Device: envoy Envoy, ensemble inventory error: TypeError: Cannot read properties of undefined (reading 'grid_profile_name'), reconnect in 15s.
[homebridge.log.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10315859/homebridge.log.txt)

grzegorz914, 2022-12-28:
fixed in beta.5

HunterEnphase, 2022-12-28:
New error lol
Device: envoy Envoy, ensemble status error: AxiosError: Request failed with status code 401, reconnect in 15s.

grzegorz914, 2022-12-28:
please try beta.6

HunterEnphase, 2022-12-28:
Showing this now.
Device: envoy Envoy, ensemble status error: Error: self-signed certificate, reconnect in 15s.

grzegorz914, 2022-12-28:
try again beta.7

HunterEnphase, 2022-12-28:
Back to this
Device: envoy Envoy, ensemble status error: AxiosError: Request failed with status code 401, reconnect in 15s.

grzegorz914, 2022-12-28:
Strange, beta.8 ready

HunterEnphase, 2022-12-28:
Device: envoy Envoy, home error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'), reconnect in 15s.

grzegorz914, 2022-12-28:
OK, beta.9 fixed

HunterEnphase, 2022-12-28:
Hmm it stops here.
[28/12/2022, 14:31:26] [homebridge-enphase-envoy] Launched child bridge with PID 13465
[28/12/2022, 14:31:26] Registering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[28/12/2022, 14:31:26] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v6.0.5-beta.9 child bridge successfully
[28/12/2022, 14:31:26] Loaded 0 cached accessories from cachedAccessories.0EFB8DB18C8D.
[28/12/2022, 14:31:26] Homebridge v1.6.0 (HAP v0.11.0) (homebridge-enphase-envoy) is running on port 52202.
[28/12/2022, 14:31:27] [homebridge-enphase-envoy] Device: envoy Envoy, debug validate jwt token: <!DOCTYPE html><h2>Valid token.</h2>
, server: openresty/1.17.8.1
date: Wed, 28 Dec 2022 19:31:24 GMT
content-type: application/octet-stream
content-length: 37
connection: keep-alive
set-cookie: sessionId=LiF61mWaIJanvMHXLV4iEvbdx7zTAIca; Secure; HttpOnly; path=/
strict-transport-security: max-age=63072000; includeSubdomains, max-age=63072000; includeSubdomains
x-frame-options: DENY, DENY
x-content-type-options: nosniff, nosniff

grzegorz914, 2022-12-28:
Try beta.10

HunterEnphase, 2022-12-28:
Same also new one.
[28/12/2022, 15:35:19] [homebridge-enphase-envoy] Device: envoy Envoy, home error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'), reconnect in 15s.
[28/12/2022, 15:35:26] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Real power': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.

grzegorz914, 2022-12-28:
Beta.11 ready

HunterEnphase, 2022-12-28:
[28/12/2022, 16:30:56] [homebridge-enphase-envoy] Device: envoy Envoy, requesting info error: AxiosError: Request failed with status code 504, reconnect in 15s.

HunterEnphase, 2022-12-28:
Looks like it's reporting fine according to the logs, but I removed the bridge and added again but the accessories for it is not showing up, does it take awhile before it shows up?

grzegorz914, 2022-12-29:
Can you post the log too?
Are you added it to the Home.app? If yes to show all data need to use EVE or Controller app because Home.app does not support custom characteristics.

HunterEnphase, 2022-12-29:
Yes I've added to the Home.app and it only shows that I've added the bridge, no accessories show up. Yes I use the Eve app to see the stats. Here is the log. Thanks!
[homebridge.log (1).txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10317609/homebridge.log.1.txt)

grzegorz914, 2022-12-29:
You need add the accessory itself to the Home.app, the bridge is irrelevant. If you don’t see the accessory to add in Home.app try this: https://github.com/grzegorz914/homebridge-enphase-envoy/wiki#troubleshooting

In other case can you check this link:
https://envoy.local/ivp/livedata/status 
and get me the output?

HunterEnphase, 2022-12-29:
Here you go, will try the troubleshooting.
[livestatus.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10317622/livestatus.txt)

HunterEnphase, 2022-12-29:
Got it! Had to go into Homebridge settings, Unpair Bridges / Cameras / TVs / External Accessories and removed the envoy bridges and accessories as the troubleshooting didn't work. The Eve app reports now! Thanks!

grzegorz914, 2022-12-29:
Im pushed beta.12 added wirelecconnectionskit to test, please try it with log

HunterEnphase, 2022-12-29:
Error showed up
[29/12/2022, 00:26:14] [homebridge-enphase-envoy] Device: envoy Envoy, home error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'), reconnect in 15s.

grzegorz914, 2022-12-29:
ok, try beta.13

HunterEnphase, 2022-12-29:
Same error but still reports
[29/12/2022, 00:41:57] [homebridge-enphase-envoy] Device: envoy Envoy, home error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'), reconnect in 15s.

grzegorz914, 2022-12-29:
14 ready for test

HunterEnphase, 2022-12-29:
Same
[29/12/2022, 01:04:50] [homebridge-enphase-envoy] Device: envoy Envoy, home error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'), reconnect in 15s.

grzegorz914, 2022-12-29:
try 15

HunterEnphase, 2022-12-29:
Yep that was 15.

grzegorz914, 2022-12-29:
then check 16

HunterEnphase, 2022-12-29:
Same:
 "alerts": [],
  "update_status": "not-satisfied",
  "wireless_connection": [
    {
      "signal_strength": 0,
      "signal_strength_max": 0,
      "type": "zigbee",
      "connected": true
    },
    {
      "signal_strength": 0,
      "signal_strength_max": 0,
      "type": "subghz",
      "connected": true
    }
  ],
  "enpower": {
    "connected": true,
    "grid_status": "closed"
  }
}
[29/12/2022, 01:19:24] [homebridge-enphase-envoy] Device: envoy Envoy, home error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'), reconnect in 15s.

grzegorz914, 2022-12-29:
17 ready

HunterEnphase, 2022-12-29:
It's good no errors.

grzegorz914, 2022-12-29:
Log please

HunterEnphase, 2022-12-29:
Heading to bed, thanks!
[homebridge.log (1).txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10317996/homebridge.log.1.txt)

grzegorz914, 2022-12-29:
beta.22 should display enpower, try and get the log please.

grzegorz914, 2022-12-29:
Please check also output from:
https://envoy.local/ivp/ensemble/status

HunterEnphase, 2022-12-29:
It stops at enpower, doesn't report either to homekit either. Can't access (https://envoy.local/ivp/ensemble/status)
Says 401 and returns to home, password and token inputted but still returns to home.
[homebridge.log (2).txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10320823/homebridge.log.2.txt)

grzegorz914, 2022-12-29:
Fixed in beta.25

HunterEnphase, 2022-12-29:
Error below but reports now
This plugin threw an error from the characteristic 'Status': Unhandled error thrown inside read handler for characteristic: Cannot read properties of undefined (reading '0'). See https://homebridge.io/w/JtMGR for more info.

grzegorz914, 2022-12-29:
Log will help to fix this warning

grzegorz914, 2022-12-29:
Fixed in beta.26

HunterEnphase, 2022-12-29:
[homebridge.log (1).txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10320907/homebridge.log.1.txt)

HunterEnphase, 2022-12-29:
Ok trying 26 now

HunterEnphase, 2022-12-29:
Looks like it's timing out, have to wait for a bit

[29/12/2022, 13:20:49] [homebridge-enphase-envoy] Launched child bridge with PID 18111
[29/12/2022, 13:20:49] Registering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[29/12/2022, 13:20:50] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v6.0.5-beta.26 child bridge successfully
[29/12/2022, 13:20:50] Loaded 0 cached accessories from cachedAccessories.0EB539B8CD32.
[29/12/2022, 13:20:50] Homebridge v1.6.0 (HAP v0.11.0) (homebridge-enphase-envoy) is running on port 48054.
[29/12/2022, 13:20:50] [homebridge-enphase-envoy] Device: envoy Envoy, debug validate jwt token: <!DOCTYPE html><h2>Valid token.</h2>
, server: openresty/1.17.8.1
date: Thu, 29 Dec 2022 18:20:49 GMT
content-type: application/octet-stream
content-length: 37
connection: keep-alive
set-cookie: sessionId=bPT7efXapnQ2hD6fu2ifVuHSSX3ZTGXB; Secure; HttpOnly; path=/
strict-transport-security: max-age=63072000; includeSubdomains, max-age=63072000; includeSubdomains
x-frame-options: DENY, DENY
x-content-type-options: nosniff, nosniff

grzegorz914, 2022-12-29:
Reboot whole Homebridge please.

HunterEnphase, 2022-12-29:
Ok no errors

grzegorz914, 2022-12-29:
OK, I think can be pushed as release.
Thanks for Your time and testing 👍

HunterEnphase, 2022-12-29:
No problem, if you need anything else, just let me know!

closed 2022-12-29 by grzegorz914

grzegorz914, 2022-12-30:
I have fixed wireless connection kit characteristic, can You Check v6.0.6 and get the log?

HunterEnphase, 2022-12-31:
This?

[30/12/2022, 23:29:53] [homebridge-enphase-envoy] Device: 192.168.1.11 Envoy, Wireless connection: ZigBee
[30/12/2022, 23:29:53] [homebridge-enphase-envoy] Device: 192.168.1.11 Envoy, Wireless connection: ZigBee, state: Connected
[30/12/2022, 23:29:53] [homebridge-enphase-envoy] Device: 192.168.1.11 Envoy, Wireless connection: ZigBee, signal strength: 0 %
[30/12/2022, 23:29:53] [homebridge-enphase-envoy] Device: 192.168.1.11 Envoy, Wireless connection: ZigBee, signal strength mex: 0 %
[30/12/2022, 23:29:53] [homebridge-enphase-envoy] Device: 192.168.1.11 Envoy, Wireless connection: Sub GHz
[30/12/2022, 23:29:53] [homebridge-enphase-envoy] Device: 192.168.1.11 Envoy, Wireless connection: Sub GHz, state: Connected
[30/12/2022, 23:29:53] [homebridge-enphase-envoy] Device: 192.168.1.11 Envoy, Wireless connection: Sub GHz, signal strength: 0 %
[30/12/2022, 23:29:53] [homebridge-enphase-envoy] Device: 192.168.1.11 Envoy, Wireless connection: Sub GHz, signal strength mex: 0 %

HunterEnphase, 2023-01-07:
This plugin threw an error from the characteristic 'PLC level': Unhandled error thrown inside read handler for characteristic: Cannot read properties of undefined (reading '3'). See https://homebridge.io/w/JtMGR for more info.

HunterEnphase, 2023-01-07:
It's actually doing that to all the inverters and encharge batteries

grzegorz914, 2023-01-07:
Can you get the complete debug log

HunterEnphase, 2023-01-13:
Sorry I am currently on vacation will be back next week and will get it to you.

This shows up when ensemble status is checked.
Device: 192.168.1.11 Envoy, Update ensemble status data error: AxiosError: Request failed with status code 401, reconnect in 15s.

This shows up when PLC level is checked.
Device: 192.168.1.11 Envoy, Update plc level data error: AxiosError: Request failed with status code 401, reconnect in 15s.

If these aren't checked, then it functions normally.

[6 more replies]

commit 8c7a498d (2022-12-29): fix #66