grzegorz914/homebridge-enphase-envoy #67: plugin crashes enphase

RobBuijs, 2022-12-25:
Hi,

My enphase has problems when I'm using this plugin. After restarting the enphase, within a couple of minutes of running the plugin it crashes with these errors.

`[25/12/2022, 11:28:48] [homebridge-enphase-envoy] Device: 192.168.3.122 Envoy, validate jwt token error: Error: read ECONNRESET, reconnect in 15s.
[25/12/2022, 11:28:48] [homebridge-enphase-envoy] Device: 192.168.3.122 Envoy, validate jwt token error: Error: read ECONNRESET, reconnect in 15s.
[25/12/2022, 11:28:48] [homebridge-enphase-envoy] Device: 192.168.3.122 Envoy, validate jwt token error: AxiosError: Request failed with status code 502, reconnect in 15s.
[25/12/2022, 11:28:48] [homebridge-enphase-envoy] Device: 192.168.3.122 Envoy, validate jwt token error: AxiosError: Request failed with status code 502, reconnect in 15s.
`

I feel it's querying the endpoint to fast untill it just crashes. Not 100% sure about this though, as I haven't been able to change timeouts. 

You probably need more information about this issue, please let me know what's needed!

grzegorz914, 2022-12-25:
Looks like Envoy is not available on this iP address. Can you login to Envoy using web browser? 
For detailed log please enable debug mode in plugin Config advanced section and restart HB. After this get the log please.

RobBuijs, 2022-12-25:
The envoy is there initially, but disappears after a couple of minutes, just reporting the 502 error. I’ll collect logs

grzegorz914, 2023-01-09:
still the issue exist?

rb2k, 2023-01-15:
I just had that as well.

```
[1/15/2023, 1:52:17 PM] [enphaseEnvoy] Device: 192.168.1.70 Envoy, validate jwt token error: Error: connect ETIMEDOUT 192.168.1.70:443, reconnect in 15s.
[1/15/2023, 1:52:17 PM] [enphaseEnvoy] Device: 192.168.1.70 Envoy, validate jwt token error: Error: connect ETIMEDOUT 192.168.1.70:443, reconnect in 15s.
[1/15/2023, 1:52:17 PM] [enphaseEnvoy] Device: 192.168.1.70 Envoy, validate jwt token error: Error: read ECONNRESET, reconnect in 15s.
[1/15/2023, 1:52:17 PM] [enphaseEnvoy] Device: 192.168.1.70 Envoy, validate jwt token error: Error: read ECONNRESET, reconnect in 15s.
[1/15/2023, 1:52:17 PM] [enphaseEnvoy] Device: 192.168.1.70 Envoy, validate jwt token error: Error: read ECONNRESET, reconnect in 15s.
[1/15/2023, 1:52:17 PM] [enphaseEnvoy] Device: 192.168.1.70 Envoy, validate jwt token error: Error: read ECONNRESET, reconnect in 15s.
[1/15/2023, 1:52:17 PM] [enphaseEnvoy] Device: 192.168.1.70 Envoy, validate jwt token error: Error: read ECONNRESET, reconnect in 15s.
[1/15/2023, 1:52:17 PM] [enphaseEnvoy] Device: 192.168.1.70 Envoy, validate jwt token error: Error: read ECONNRESET, reconnect in 15s.
[1/15/2023, 1:52:17 PM] [enphaseEnvoy] Device: 192.168.1.70 Envoy, validate jwt token error: Error: read ECONNRESET, reconnect in 15s.
```

Restart solved it for me, so it didn't seem to be a problem with the IP/endpoint.

I didn't debug it further, but I think an actionable item here could be a backoff so it doesn't do a tight loop and prints millions of log lines :)

grzegorz914, 2023-01-15:
It’s not a plugin problem, you need investigate on your environment, your device just lose connection from network.

rb2k, 2023-01-15:
It did instantly recover the second I restarted homebridge. I wonder if the code path for trying to reconnect has some unexpected outcomes.

grzegorz914, 2023-01-15:
The log you provided took only the envoy is not available on network. Just is not available for homebridge.

grzegorz914, 2023-01-15:
What a envoy model you have and installed firmware?

rb2k, 2023-01-16:
Software Version D7.3.120 (ac5b3c)
Part Number: 800-00695-r01

rb2k, 2023-01-16:
I wonder if there's just a memory leak somewhere (maybe even based on system setup) and the frequent requests are overflowing things.
Is there a way to reduce the polling frequency to e.g. once a minute?

rb2k, 2023-01-16:
So the 'refreshInterval' over here: https://github.com/grzegorz914/homebridge-enphase-envoy/blob/main/README.md?plain=1#L90 doesn't do anything, right?

grzegorz914, 2023-01-16:
Ah its old properties not used long time, just removed from readme. For best practice always use GUI Config.

grzegorz914, 2023-01-16:
On which device  you have installed Homebridge?

rb2k, 2023-01-16:
> On which device you have installed Homebridge?

Running in docker on an Unraid based NAS (tons of CPU/RAM)

grzegorz914, 2023-01-16:
Is your Envoy connected with LAN or WIFi

rb2k, 2023-01-16:
Wifi

grzegorz914, 2023-01-16:
Here is the meters refresh data:
 https://github.com/grzegorz914/homebridge-enphase-envoy/blob/5851cc21155eec7489ab373425acd85855c65c9c/index.js#L2568 
but this poll every 2 seconds other poll for live data is 3 seconds but this can be completely disable in plugin settings. I don’t think increase pool time solve your problem. I use this 2 years long and no any Probleme like you have, others users with big installations with encharges and enpower also not reported any Probleme with envoy freeze.

rb2k, 2023-01-16:
I don't think it depends on the size of the install or anything like that.
Probably just some dumb thing where they lose a few bytes everytime they read something. Usually it wouldn't matter because they probably restart the process every 12 hours, but if you ask twice a second, it adds up :)

In general, I'd appreciate a way to make this user configurable just because I don't want home assistant to keep all that data around. Once a minute is plenty for me, I just want to get a feel for how much it's producing / consuming

grzegorz914, 2023-01-17:
in 6.5.0 added possibility to set refresh time for live data, meters data and production data, available in advanced section, may be this help you

alexyu650, 2023-02-03:
i started to get this error "Device: 192.168.0.204 Envoy-S, Update ensemble status data error: AxiosError: Request failed with status code 401, reconnect in 15s." since the third update this past month. was working fine before the update, the ip address allows me to connect to envoy still but just doesnt work for homebridge

grzegorz914, 2023-02-03:
Try to recreate token, if not help please disable Ensemble Status in plugin config advanced section.

alexyu650, 2023-02-03:
thanks for the the quick response! i have tried multiple times with creating a new token and still got this issue. i just tried again creating a new token as well as disabling the ensmble status and still getting the same error.

grzegorz914, 2023-02-03:
If you disabled ensemble status is not possible you get same error, try again please.

alexyu650, 2023-02-03:
this is copy paste of the json file 

"       "powerConsumptionNetMaxAutoReset": 0,
            "powerConsumptionNetMaxDetected": 10000,
            "energyConsumptionNetLifetimeOffset": 0,
            "supportEnsembleStatus": false,
            "supportLiveData": true,
            "liveDataRefreshTime": 1000,
            "metersDataRefreshTime": 2500,"

copy paste of homebridge logs
[03/02/2023, 00:53:12] [Homebridge UI] Terminal session ended.
[03/02/2023, 00:53:26] [homebridge-enphase-envoy] Device: 192.168.0.204 Envoy-S, Update power mode error: AxiosError: Request failed with status code 401, reconnect in 15s.
[03/02/2023, 00:53:55] [homebridge-enphase-envoy] Device: 192.168.0.204 Envoy-S, Update power mode error: AxiosError: Request failed with status code 401, reconnect in 15s.
[03/02/2023, 00:54:18] [homebridge-enphase-envoy] Device: 192.168.0.204 Envoy-S, Update power mode error: AxiosError: Request failed with status code 401, reconnect in 15s.
[03/02/2023, 00:54:18] [Homebridge UI] Starting terminal session
[03/02/2023, 00:54:40] [homebridge-enphase-envoy] Device: 192.168.0.204 Envoy-S, Update power mode error: AxiosError: Request failed with status code 401, reconnect in 15s.
[03/02/2023, 00:55:05] [homebridge-enphase-envoy] Device: 192.168.0.204 Envoy-S, Updat

grzegorz914, 2023-02-03:
here is about power mode data, please disable support to check/control power mode

alexyu650, 2023-02-03:
hi made the changes,
"devices": [
        {
            "name": "Envoy-S",
            "host": "removed",
            "envoyFirmware7xx": true,
            "envoyFirmware7xxToken": removed
            "disableLogInfo": false,
            "disableLogDeviceInfo": false,
            "enableDebugMode": false,
            "powerProductionMax": false,
            "powerProductionMaxAutoReset": 0,
            "powerProductionMaxDetected": 5400,
            "energyProductionLifetimeOffset": 0,
            "powerConsumptionTotalMax": false,
            "powerConsumptionTotalMaxAutoReset": 0,
            "powerConsumptionTotalMaxDetected": 10000,
            "energyConsumptionTotalLifetimeOffset": 0,
            "powerConsumptionNetMax": false,
            "powerConsumptionNetMaxAutoReset": 0,
            "powerConsumptionNetMaxDetected": 10000,
            "energyConsumptionNetLifetimeOffset": 0,
            "supportProductionPowerMode": false,
            "supportPlcLevel": true,
            "supportEnsembleStatus": false,
            "supportLiveData": true,
            "liveDataRefreshTime": 1000,
            "metersDataRefreshTime": 2500,
            "productionDataRefreshTime": 5000,
            "enableMqtt": false,
            "mqttAuth": false,
            "mqttDebug": false
        }
    ],
    "_bridge": {
        "username": "0E:AD:C4:16:DC:D0",
        "port": 40860
    },
    "platform": "enphaseEnvoy"
}
still same error 

[03/02/2023, 09:26:01] [homebridge-enphase-envoy] Device: 192.168.0.204 Envoy-S, Validate jwt token data error: Error: connect EHOSTUNREACH 192.168.0.204:443, reconnect in 15s.
[03/02/2023, 09:26:19] [homebridge-enphase-envoy] Device: 192.168.0.204 Envoy-S, Validate jwt token data error: Error: connect EHOSTUNREACH 192.168.0.204:443, reconnect in 15s.
[03/02/2023, 09:26:37] [homebridge-enphase-envoy] Device: 192.168.0.204 Envoy-S, Validate jwt token data error: Error: connect EHOSTUNREACH 192.168.0.204:443, reconnect in 15s.
[03/02/2023, 09:26:44] [Homebridge UI] Terminal session ended.
[03/02/2023, 09:26:55] [homebridge-enphase-envoy] Device: 192.168.0.204 Envoy-S, Validate jwt token data error: Error: connect EHOSTUNREACH 192.168.0.204:443, reconnect in 15s.
[03/02/2023, 09:27:14] [homebridge-enphase-envoy] Device: 192.168.0.204 Envoy-S, Validate jwt token data error: Error: connect EHOSTUNREACH 192.168.0.204:443, reconnect in 15s.
[03/02/2023, 09:27:32] [homebridge-enphase-envoy] Device: 192.168.

grzegorz914, 2023-02-03:
It’s not same error, now your envoy is not available in Netz. Check it please again or restart envoy .

alexyu650, 2023-02-03:
have restarted 

[03/02/2023, 09:45:45] [homebridge-enphase-envoy] Restarting child bridge...
[03/02/2023, 09:45:45] Got SIGTERM, shutting down child bridge process...
[03/02/2023, 09:45:50] [homebridge-enphase-envoy] Child bridge process ended
[03/02/2023, 09:45:50] [homebridge-enphase-envoy] Process Ended. Code: 143, Signal: null
[03/02/2023, 09:45:57] [homebridge-enphase-envoy] Restarting Process...
[03/02/2023, 09:45:58] [homebridge-enphase-envoy] Launched child bridge with PID 233021
[03/02/2023, 09:45:58] Registering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[03/02/2023, 09:45:58] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v6.5.4 child bridge successfully
[03/02/2023, 09:45:58] Loaded 0 cached accessories from cachedAccessories.0EADC416DCD0.
[03/02/2023, 09:45:58] Homebridge v1.6.0 (HAP v0.11.0) (homebridge-enphase-envoy) is running on port 40860.
[03/02/2023, 09:46:01] [homebridge-enphase-envoy] Device: 192.168.0.204 Envoy-S, Validate jwt token data error: Error: connect EHOSTUNREACH 192.168.0.204:443, reconnect in 15s.

grzegorz914, 2023-02-03:
Envoy restart not the plugin, or ping your envoy

alexyu650, 2023-02-03:
getting this now

[03/02/2023, 09:48:56] [homebridge-enphase-envoy] Device: 192.168.0.204 Envoy-S, Update home data error: TypeError: Cannot convert undefined or null to object, reconnect in 15s.

alexyu650, 2023-02-03:
also this 

[03/02/2023, 09:50:02] [homebridge-enphase-envoy] Device: 192.168.0.204 Envoy-S, Update plc level data error: AxiosError: Request failed with status code 401, reconnect in 15s.

grzegorz914, 2023-02-03:
Can you enable debug mode in plugin config and post the complete log?

grzegorz914, 2023-02-03:
Disable plc level check

alexyu650, 2023-02-03:
here you go 
[03/02/2023, 09:58:06] [homebridge-enphase-envoy] Device: 192.168.0.204 Envoy-S, debug live data: {
  "connection": {
    "mqtt_state": "connected",
    "prov_state": "configured",
    "auth_state": "ok",
    "sc_stream": "disabled",
    "sc_debug": "disabled"
  },
  "meters": {
    "last_update": 1675275639,
    "soc": 100,
    "main_relay_state": 1,
    "gen_relay_state": 5,
    "backup_bat_mode": 0,
    "backup_soc": 100,
    "is_split_phase": 1,
    "phase_count": 0,
    "enc_agg_soc": 100,
    "enc_agg_energy": 10080,
    "acb_agg_soc": 0,
    "acb_agg_energy": 0,
    "pv": {
      "agg_p_mw": 586234,
      "agg_s_mva": 827987,
      "agg_p_ph_a_mw": 586234,
      "agg_p_ph_b_mw": 0,
      "agg_p_ph_c_mw": 0,
      "agg_s_ph_a_mva": 827987,
      "agg_s_ph_b_mva": 0,
      "agg_s_ph_c_mva": 0
    },
    "storage": {
      "agg_p_mw": -4000,
      "agg_s_mva": -4000,
      "agg_p_ph_a_mw": -4000,
      "agg_p_ph_b_mw": 0,
      "agg_p_ph_c_mw": 0,
      "agg_s_ph_a_mva": -4000,
      "agg_s_ph_b_mva": 0,
      "agg_s_ph_c_mva": 0
    },
    "grid": {
      "agg_p_mw": 959438,
      "agg_s_mva": 1572615,
      "agg_p_ph_a_mw": 959438,
      "agg_p_ph_b_mw": 0,
      "agg_p_ph_c_mw": 0,
      "agg_s_ph_a_mva": 1572615,
      "agg_s_ph_b_mva": 0,
      "agg_s_ph_c_mva": 0
    },
    "load": {
      "agg_p_mw": 1541672,
      "agg_s_mva": 2396602,
      "agg_p_ph_a_mw": 1541672,
      "agg_p_ph_b_mw": 0,
      "agg_p_ph_c_mw": 0,
      "agg_s_ph_a_mva": 2396602,
      "agg_s_ph_b_mva": 0,
      "agg_s_ph_c_mva": 0
    },
    "generator": {
      "agg_p_mw": 0,
      "agg_s_mva": 0,
      "agg_p_ph_a_mw": 0,
      "agg_p_ph_b_mw": 0,
      "agg_p_ph_c_mw": 0,
      "agg_s_ph_a_mva": 0,
      "agg_s_ph_b_mva": 0,
      "agg_s_ph_c_mva": 0
    }
  },
  "tasks": {
    "task_id": 1353142665,
    "timestamp": 1675439734
  },
  "counters": {
    "main_CfgLoad": 2,
    "main_CfgChanged": 2,
    "main_CfgNotFound": 29,
    "main_taskUpdate": 42,
    "main_sigHUP": 13,
    "MqttClient_publish": 6401,
    "MqttClient_respond": 108,
    "MqttClient_msgarrvd": 54,
    "MqttClient_reconnect": 13,
    "MqttClient_create": 47,
    "MqttClient_setCallbacks": 47,
    "MqttClient_connect": 47,
    "MqttClient_connect_err": 7,
    "MqttClient_connect_Err": 7,
    "MqttClient_subscribe": 40,
    "SSL_Keys_Create": 47,
    "sc_hdlDataPub": 6363,
    "sc_SendStreamCtrl": 19,
    "sc_SendDemandRspCtrl": 1,
    "rest_Status": 8569
  }

grzegorz914, 2023-02-03:
All is correct here in this log

alexyu650, 2023-02-03:
looks like disabling plc resolved the errors, thanks for all your help!

alexyu650, 2023-02-04:
hi again

getting a new error today, no changes have been made

[04/02/2023, 09:45:02] [homebridge-enphase-envoy] Device: 192.168.0.204 Envoy-S, Update home data error: TypeError: Cannot convert undefined or null to object, reconnect in 15s.

grzegorz914, 2023-02-04:
Debug log pleas

alexyu650, 2023-02-04:
here it is

        ],
        "version": [
          "07.03.120"
        ],
        "build": [
          "ac5b3c"
        ]
      },
      {
        "$": {
          "name": "devimg"
        },
        "pn": [
          "500-00004-r01"
        ],
        "version": [
          "01.02.413"
        ],
        "build": [
          "9c64cd"
        ]
      },
      {
        "$": {
          "name": "geo"
        },
        "pn": [
          "500-00008-r01"
        ],
        "version": [
          "02.01.24"
        ],
        "build": [
          "a74d96"
        ]
      },
      {
        "$": {
          "name": "backbone"
        },
        "pn": [
          "500-00010-r01"
        ],
        "version": [
          "07.00.20"
        ],
        "build": [
          "176d57"
        ]
      },
      {
        "$": {
          "name": "meter"
        },
        "pn": [
          "500-00013-r01"
        ],
        "version": [
          "03.02.08"
        ],
        "build": [
          "f9ead3"
        ]
      },
      {
        "$": {
          "name": "agf"
        },
[... 62 lines cut ...]
          "02.00.3244"
        ],
        "release_stage": [
          "700-GA"
        ]
      }
    ]
  }
}
[04/02/2023, 09:58:45] [homebridge-myq] myQ API: Successfully connected to the myQ API.
[04/02/2023, 09:58:47] [CameraUI] Driveway: Setting up camera, please be patient...
[04/02/2023, 09:58:47] [CameraUI] Backyard: Setting up camera, please be patient...
[04/02/2023, 09:58:47] [CameraUI] Front Door: Setting up camera, please be patient...
[04/02/2023, 09:58:47] [CameraUI] Back Door: Setting up camera, please be patient...
[04/02/2023, 09:58:47] [CameraUI] Right Gate: Setting up camera, please be patient...
[04/02/2023, 09:58:47] [CameraUI] Left Gate: Setting up camera, please be patient...
[04/02/2023, 09:58:46] [homebridge-myq] myQ API: Discovered device family gateway: Garage Door Double  [Chamberlain WiFi GDO DC 1.25 HP] (serial number: GW1200049987).
[04/02/2023, 09:58:46] [homebridge-myq] myQ API: Discovered device family gateway: Garage Door Single [Chamberlain WiFi GDO DC 1.25 HP] (serial number: GW1200049977).
[04/02/2023, 09:58:46] [homebridge-myq] myQ API: Discovered device family gateway: MyQ Bridge [Chamberlain myQ Home Bridge] (serial number: GW200000461D).
[04/02/2023, 09:58:46] [homebridge-myq] myQ API: Discovered device family garagedoor: Round Rock Garage Door Double [Liftmaster Commercial WiFi Wall Mount Operator] (serial number: CG2863447AF9, gateway: GW1200049977).
[04/02/2023, 09:58:46] [homebridge-myq] myQ API: Discovered device family garagedoor: Round Rock Garage Door Single [Liftmaster Commercial WiFi Wall Mount Operator] (serial number: CG2863447B09, gateway: GW1200049987).
[04/02/2023, 09:58:46] [homebridge-myq] myQ API: Discovered device family garagedoor: San Bruno Garage Door (serial number: CG0846030723, gateway: GW200000461D).
[04/02/2023, 09:58:49] [CameraUI] camera.ui v5.0.27 is listening on port 8081 (http)
[04/02/2023, 09:58:50] [homebridge-enphase-envoy] Device: 192.168.0.204 Envoy-S, debug home: "      <!DOCTYPE html PUBLIC \"-//W3C//DTD HTML 4.01//EN\">\n      <html lang=\"en-US\">\n      \n      <head>\n        <title>error</title>\n        <meta http-equiv=\"Content-Type\" content=\"text/html; charset=utf-8\">\n        <link rel=\"icon\"\n              type=\"image/png\" href=\"/images/favicon.ico?v=6\">\n        \n        <link rel=\"styleSheet\" type=\"text/css\" href=\"/include/css/style.css\" >\n        <style type=\"text/css\">.nav { width: 190px; font-size: 10px; padding: 10px; margin: 10px;float: left; background: #D8DBD4; color: #213d30; border: 2px solid #333; }\n\t.navholder { width: auto; height: auto; float: left; margin: 10px 0px 0 0 }\n\t.tbl_cap { font-size:8pt; background-color:#DD4F05; color:white; border-style:solid; border-width:1px; border-color:white; }\n\t.tbl_bod { font-family:arial; font-size:8pt; color:#213d30; background-color:#D8DBD4; border-style:solid; border-color:white; border-width:1px; }\n\t.tbl_bod_right { font-family:arial; font-size:8pt; color:#213d30; background-color:#D8DBD4; border-style:solid; border-color:white; border-width:1px; text-align:right }\n\t.tbl_hdr { font-size:8pt; background-color:#CCCCCC; border-style:solid; border-width:1px; border-color:white; color:#213d30; }\n\t.clearer { clear: both; font-size: 1px; }\n\t.outer { width: auto; height: auto; padding: 0px; font-size: 10px }\n\t.table1 { width: 100%; margin-left: auto; margin-right: auto;border: 1px solid #CCCCCC }\n\t.table2 { width: 100%;text-align: center;margin-left: auto; margin-right: auto;border: 1px solid #CCCCCC }\n\t.table3 { width: 720px;margin-left: auto; margin-right: auto;border: 1px solid #CCCCCC }\n\t.div_params { height: 190px; width: 380px; overflow: auto;margin-left: auto; margin-right: auto; }\n\t.div_edit_params { height: 230px; width: 470px; overflow: auto;margin-left: auto; margin-right: auto; }\n\t.div_outer_box { width: 720px;margin-left: auto; margin-right: auto;border: 1px solid #FFFFFF }\n\t.txt_expl_center { font-size: 12px; text-align:center;height: auto; border: 1px solid #CCCCCC;margin-top: 5px; margin-left: 5px; margin-right: auto; }\n\t.label_text_right { text-align: right; border: 1px solid #CCCCCC;font-size: 10px; }\n\t.label_text_left { text-align: left; border: 1px solid #CCCCCC;font-size: 10px; }\n\t.label_text_left_alert { text-align: left; color: #7d3020; border: 1px solid #CCCCCC;font-size: 12px; }\n\t.link_cell_r [... cut ...] in-right:auto; width: 100%; text-align: center; \">\n      &copy; 2007-2022, translation missing: en-US.envoy.global.copyright_notice_01 |\n      <a href=\"http://www.enphaseenergy.com/licenses\">translation missing: en-US.lwui.nav.footer.licenses</a>\n      \n      <br><br>\n      </div>\n      </body>\n      </html>\n"
[04/02/2023, 09:58:50] [homebridge-enphase-envoy] Device: 192.168.0.204 Envoy-S, Update home data error: TypeError: Cannot convert undefined or null to object, reconnect in 15s.

grzegorz914, 2023-02-04:
Looks like Your envoy dosent respond correct, its not a plugin issue, Try to restart Your envoy and let me know.

grzegorz914, 2023-02-26:
I close this now because no activity from user.

closed 2023-02-26 by grzegorz914