grzegorz914/homebridge-enphase-envoy #69: Error with firmware 7

hugo-leij, 2023-01-08:
Hi,

I installed the plugin and get the token but i get the flowwing error: This plugin threw an error from the characteristic 'PLC level': Unhandled error thrown inside read handler for characteristic: Cannot read properties of undefined (reading '13'). See https://homebridge.io/w/JtMGR for more info.

Anyone have an idea how I can solve this?

**Config:**
```
 "devices": [
                {
                    "name": "Enphase",
                    "host": "10.0.1.153",
                    "envoyFirmware7xx": true,
                    "envoyFirmware7xxToken": "*******",
                    "disableLogInfo": false,
                    "disableLogDeviceInfo": false,
                    "enableDebugMode": false,
                    "powerProductionMaxAutoReset": 0,
                    "powerProductionMaxDetected": 0,
                    "energyProductionLifetimeOffset": 0,
                    "powerConsumptionTotalMaxAutoReset": 0,
                    "powerConsumptionTotalMaxDetected": 0,
                    "energyConsumptionTotalLifetimeOffset": 0,
                    "powerConsumptionNetMaxAutoReset": 0,
                    "powerConsumptionNetMaxDetected": 0,
                    "energyConsumptionNetLifetimeOffset": 0,
                    "enableMqtt": false,
                    "mqttAuth": false,
                    "mqttDebug": false
                }

```
**Error**
```

1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Envoy: 122233017226, alerts: No alerts
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Envoy: 122233017226, network interface: WiFi
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Envoy: 122233017226, web communication: Yes
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Envoy: 122233017226, report to enlighten: Yes
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Envoy: 122233017226, communication devices and level: 15 / 100
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Envoy: 122233017226, communication qRelays and level: 1 / 100
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Envoy: 122233017226, communication Microinverters and level: 14 / 100
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Envoy: 122233017226, data base size: 3284992 /   0.87%
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Envoy: 122233017226, tariff: None
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Envoy: 122233017226, update status: Satisfied
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Envoy: 122233017226, firmware: D7.0.88
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Envoy: 122233017226, time zone: Europe/Amsterdam
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Envoy: 122233017226, current date and time: 1/8/2023, 1 12:31
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Envoy: 122233017226, last report to enlighten: 1/8/2023, 12:26:45 PM
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Envoy: 122233017226, checking comm level: No
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Q-Relay: 122236087561, relay state: Closed
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Q-Relay: 122236087561, lines: 3
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Q-Relay: 122236087561, line 1: Closed
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Q-Relay: 122236087561, line 2: Closed
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Q-Relay: 122236087561, line 3: Closed
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Q-Relay: 122236087561, producing: No
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Q-Relay: 122236087561, communicating: Yes
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Q-Relay: 122236087561, provisioned: Yes
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Q-Relay: 122236087561, operating: Yes
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Q-Relay: 122236087561, status: Normal
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Q-Relay: 122236087561, firmware: 520-00086-r01-v02.12.11
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Q-Relay: 122236087561, last report: 1/8/2023, 12:17:10 PM
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Production power: 0.814 kW
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Production power peak: 0.814 kW
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Production power peak detected: Yes
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Production energy today: 1.489 kWh
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Production energy last seven days: 14.732 kWh
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Production energy lifetime: 403.311 kWh
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Production last report: 1/8/2023, 12:32:17 PM
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Production power peak reset: Off
[... 117 lines cut ...]
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Microinverter: 122213033506, last power: 35 W
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Microinverter: 122213033506, max power: 35 W
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Microinverter: 122213033506, producing: Yes
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Microinverter: 122213033506, communicating: Yes
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Microinverter: 122213033506, provisioned: Yes
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Microinverter: 122213033506, operating: Yes
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Microinverter: 122213033506, status: Normal
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Microinverter: 122213033506, firmware: 520-00082-r01-v04.27.04
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] Device: 10.0.1.153 Enphase, Microinverter: 122213033506, last report: 1/8/2023, 12:17:05 PM
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] This plugin threw an error from the characteristic 'PLC level': Unhandled error thrown inside read handler for characteristic: Cannot read properties of undefined (reading '0'). See https://homebridge.io/w/JtMGR for more info.
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] This plugin threw an error from the characteristic 'PLC level': Unhandled error thrown inside read handler for characteristic: Cannot read properties of undefined (reading '0'). See https://homebridge.io/w/JtMGR for more info.
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] This plugin threw an error from the characteristic 'PLC level': Unhandled error thrown inside read handler for characteristic: Cannot read properties of undefined (reading '1'). See https://homebridge.io/w/JtMGR for more info.
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] This plugin threw an error from the characteristic 'PLC level': Unhandled error thrown inside read handler for characteristic: Cannot read properties of undefined (reading '2'). See https://homebridge.io/w/JtMGR for more info.
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] This plugin threw an error from the characteristic 'PLC level': Unhandled error thrown inside read handler for characteristic: Cannot read properties of undefined (reading '3'). See https://homebridge.io/w/JtMGR for more info.
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] This plugin threw an error from the characteristic 'PLC level': Unhandled error thrown inside read handler for characteristic: Cannot read properties of u [... cut ...]
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] This plugin threw an error from the characteristic 'PLC level': Unhandled error thrown inside read handler for characteristic: Cannot read properties of undefined (reading '12'). See https://homebridge.io/w/JtMGR for more info.
[1/8/2023, 12:32:32 PM] [homebridge-enphase-envoy] This plugin threw an error from the characteristic 'PLC level': Unhandled error thrown inside read handler for characteristic: Cannot read properties of undefined (reading '13'). See https://homebridge.io/w/JtMGR for more info.
```

GevaudanBeast, 2023-01-09:
Hello, same error here. homebridge version is 1.4.1

hugo-leij, 2023-01-09:
He @grzegorz914 

With the debug mode i dont get te error:

log:

```

[37m[1/9/2023, 12:52:35 PM] [39m[36m[homebridge-enphase-envoy][39m Device: 10.0.1.153 Enphase, debug read production power peak:0.814 kW
[37m[1/9/2023, 12:52:36 PM] [39m[36m[homebridge-enphase-envoy][39m Device: 10.0.1.153 Enphase, debug live data: {
  "connection": {
    "mqtt_state": "connected",
    "prov_state": "configured",
    "auth_state": "ok",
    "sc_stream": "disabled",
    "sc_debug": "disabled"
  },
  "meters": {
    "last_update": 0,
    "soc": 0,
    "main_relay_state": 13,
    "gen_relay_state": 0,
    "backup_bat_mode": 0,
    "backup_soc": 0,
    "pv": {
      "agg_p_mw": 0,
      "agg_s_mva": 0
    },
    "storage": {
      "agg_p_mw": 0,
      "agg_s_mva": 0
    },
    "grid": {
      "agg_p_mw": 0,
      "agg_s_mva": 0
    },
    "load": {
      "agg_p_mw": 0,
      "agg_s_mva": 0
    },
    "generator": {
      "agg_p_mw": 0,
      "agg_s_mva": 0
    }
  },
  "tasks": {
    "task_id": 1277979412,
    "timestamp": 1673170202
  },
  "counters": {
    "main_CfgLoad": 1,
    "main_CfgChanged": 1,
    "main_taskUpdate": 1,
    "MqttClient_respond": 1,
    "MqttClient_msgarrvd": 1,
    "MqttClient_create": 7059,
    "MqttClient_setCallbacks": 7059,
    "MqttClient_connect": 7059,
    "MqttClient_connect_err": 7046,
    "MqttClient_connect_Err": 7046,
    "MqttClient_subscribe": 13,
    "SSL_Keys_Create": 7059,
    "rest_Status": 37922
  }
}
[37m[1/9/2023, 12:52:39 PM] [39m[36m[homebridge-enphase-envoy][39m Device: 10.0.1.153 Enphase, debug live data: {
  "connection": {
    "mqtt_state": "connected",
    "prov_state": "configured",
    "auth_state": "ok",
[... 2335 lines cut ...]
      "agg_p_mw": 0,
      "agg_s_mva": 0
    }
  },
  "tasks": {
    "task_id": 1277979412,
    "timestamp": 1673170202
  },
  "counters": {
    "main_CfgLoad": 1,
    "main_CfgChanged": 1,
    "main_taskUpdate": 1,
    "MqttClient_respond": 1,
    "MqttClient_msgarrvd": 1,
    "MqttClient_create": 7059,
    "MqttClient_setCallbacks": 7059,
    "MqttClient_connect": 7059,
    "MqttClient_connect_err": 7046,
    "MqttClient_connect_Err": 7046,
    "MqttClient_subscribe": 13,
    "SSL_Keys_Create": 7059,
    "rest_Status": 37950
  }
}
```

grzegorz914, 2023-01-09:
The log is not complete, you need get the complete log direct after plugin restart, your log missing first plugin run.

hugo-leij, 2023-01-09:
```
[37m[1/9/2023, 1:36:56 PM][0m [36m[HB Supervisor][0m Homebridge Process Ended. Code: 143, Signal: null
[37m[1/9/2023, 1:37:01 PM][0m [36m[HB Supervisor][0m Restarting Homebridge...
[37m[1/9/2023, 1:37:01 PM][0m [36m[HB Supervisor][0m Starting Homebridge with extra flags: -I -P /var/lib/homebridge/node_modules --strict-plugin-resolution
[37m[1/9/2023, 1:37:01 PM][0m [36m[HB Supervisor][0m Started Homebridge v1.6.0 with PID: 27961
[37m[1/9/2023, 1:37:02 PM] [39mLoaded config.json with 1 accessories and 8 platforms.
[37m[1/9/2023, 1:37:02 PM] [39mLoaded 18 cached accessories from cachedAccessories.
[37m[1/9/2023, 1:37:02 PM] [39m---
[37m[1/9/2023, 1:37:02 PM] [39mLoaded plugin: homebridge-dummy@0.8.0
[37m[1/9/2023, 1:37:02 PM] [39mRegistering accessory 'homebridge-dummy.DummySwitch'
[37m[1/9/2023, 1:37:02 PM] [39m---
[37m[1/9/2023, 1:37:02 PM] [39mLoaded plugin: homebridge-edomoticz@2.1.46
[37m[1/9/2023, 1:37:02 PM] [39mRegistering platform 'homebridge-edomoticz.eDomoticz'
[37m[1/9/2023, 1:37:02 PM] [39m---
[37m[1/9/2023, 1:37:02 PM] [39mLoaded plugin: homebridge-enphase-envoy@6.0.8
[37m[1/9/2023, 1:37:03 PM] [39mRegistering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[37m[1/9/2023, 1:37:03 PM] [39m---
[37m[1/9/2023, 1:37:03 PM] [39m[33mDisabled plugin: homebridge-harmony@1.6.2[39m
[37m[1/9/2023, 1:37:03 PM] [39m---
[37m[1/9/2023, 1:37:03 PM] [39m[33mDisabled plugin: homebridge-lg-thinq@1.3.0[39m
[37m[1/9/2023, 1:37:03 PM] [39m---
[37m[1/9/2023, 1:37:04 PM] [39m[33mDisabled plugin: homebridge-slide-shutter@1.0.6[39m
[37m[1/9/2023, 1:37:04 PM] [39m---
[37m[1/9/2023, 1:37:04 PM] [39m[33mDisabled plugin: homebridge-sonos-starter-track@1.0.19[39m
[37m[1/9/2023, 1:37:04 PM] [39m---
[37m[1/9/2023, 1:37:04 PM] [39mLoaded plugin: homebridge-tado-platform@6.0.14
[37m[1/9/2023, 1:37:04 PM] [39mRegistering platform 'homebridge-tado-platform.TadoPlatform'
[37m[1/9/2023, 1:37:04 PM] [39m---
[37m[1/9/2023, 1:37:05 PM] [39mLoaded plugin: homebridge-tesla@4.2.0
[37m[1/9/2023, 1:37:05 PM] [39mRegistering accessory 'homebridge-tesla.Tesla'
[37m[1/9/2023, 1:37:05 PM] [39m---
[37m[1/9/2023, 1:37:05 PM] [39mLoaded plugin: homebridge-unifi-protect@5.5.4
[37m[1/9/2023, 1:37:05 PM] [39mRegistering platform 'homebridge-unifi-protect.UniFi Protect'
[37m[1/9/2023, 1:37:05 PM] [39m---
[37m[1/9/2023, 1:37:05 PM] [39mLoading 8 platforms...
[37m[1/9/2023, 1:37:05 PM] [39m[33mIgnoring config for the platform "LGThinQ" in your config.json as the plugin "homebridge-lg-thinq" has been disabled.[39m
[37m[1/9/2023, 1:37:05 PM] [39m[33mIgnoring config for the platform "slide-shutter" in your config.json as the plugin "homebridge-slide-shutter" has been disabled.[39m
[37m[1/9/2023, 1:37:05 PM] [39m[33mIgnoring config for the platform "HarmonyHubWebSocket" in your config.json as the plugin "homebridge-harmony" has been disabled.[39m
[37m[1/9/2023, 1:37:05 PM] [39m[36m[TadoPlatform][39m Initializing TadoPlatform platform...
[37m[1/9/2023, 1:37:07 PM] [39m[36m[Domoticz][39m Initializing eDomoticz platform...
[37m[1/9/2023, 1:37:07 PM] [39m[36m[UniFi Protect][39m Initializing UniFi Protect platform...
[37m[1/9/2023, 1:37:07 PM] [39m[36m[UniFi Protect][39m Initializing child bridge 0E:DE:96:9B:BC:AB
[37m[1/9/2023, 1:37:07 PM] [39m[36m[enphaseEnvoy][39m Initializing enphaseEnvoy platform...
[37m[1/9/2023, 1:37:07 PM] [39m[36m[enphaseEnvoy][39m Initializing child bridge 0E:65:FD:E1:9B:69
[37m[1/9/2023, 1:37:07 PM] [39mLoading 1 accessories...
[37m[1/9/2023, 1:37:07 PM] [39m[36m[Model S][39m Initializing Tesla accessory...
[37m[1/9/2023, 1:37:07 PM] [39m[36m[TadoPlatform][39m Allerheiligenweg Huiskamer Heater: Configuring accessory...
[37m[1/9/2023, 1:37:07 PM] [39m[36m[Domoticz][39m synchronizeAccessories in progress...
Setup Payload:
X-HM://0023I7JB6KXU8
Enter this code with your HomeKit app on your iOS device to pair with Homebridge:
[30m[47m                       [49m[39m
[30m[47m    ┌────────────┐     [49m[39m
[30m[47m    │ *******│     [49m[39m
[30m[47m    └────────────┘     [49m[39m
[30m[47m                       [49m[39m
[37m[1/9/2023, 1:37:08 PM] [39m[36m[Model S][39m Exchanging refresh token for an access token…
[37m[1/9/2023, 1:37:08 PM] [39mHomebridge v1.6.0 (HAP v0.11.0) (Homebridge) is running on port 51825.
[37m[1/9/2023, 1:37:08 PM] [39m[36m[Domoticz][39m There was an error while getting the MQTT Hardware Device from Domoticz.
Please verify that you have added the MQTT Hardware Device and that the hardware device is enabled.
[37m[1/9/2023, 1:37:08 PM] [39m[36m[Model S][39m Got an access token.
[37m[1/9/2023, 1:37:08 PM] [39m[36m[homebridge-enphase-envoy][39m Launched child bridge with PID 27978
[37m[1/9/2023, 1:37:08 PM] [39m[36m[homebridge-unifi-protect][39m Launched child bridge with PID 27977
[37m[1/9/2023, 1:37:09 PM] [39m[36m[Model S][39m Tesla reports vehicle is awake.
[37m[1/9/2023, 1:37:09 PM] [39m[36m[Model S][39m Getting latest vehicle data from Tesla…
[37m[1/9/2023, 1:37:09 PM] [39mRegistering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[37m[1/9/2023, 1:37:09 PM] [39m[36m[homebridge-enphase-envoy][39m Loaded homebridge-enphase-envoy v6.0.8 child bridge successfully
[37m[1/9/2023, 1:37:09 PM] [39mLoaded 0 cached accessories from cachedAccessories.0E65FDE19B69.
[... cut ...] ersion>02.00.00</version>\n    <build>54a6dc</build>\n  </package>\n  <package name='pkgsec'>\n    <pn>500-00021-r01</pn>\n    <version>01.00.00</version>\n    <build>19ae14</build>\n  </package>\n  <build_info>\n    <build_time_gmt>1645731441</build_time_gmt>\n    <build_id>ec2-user-envoy_uber-pkg_master:pkg-Feb-24-22-19:35:10</build_id>\n  </build_info>\n</envoy_info>\n"
[37m[1/9/2023, 1:37:10 PM] [39m[36m[homebridge-enphase-envoy][39m Device: 10.0.1.153 Enphase, debug parse info: {
``````

grzegorz914, 2023-01-09:
This is not all.

hugo-leij, 2023-01-09:
```

[0;37m[1/9/2023, 2:11:34 PM] [0m[0;36m[Homebridge UI][0m Homebridge restart request received
[0;37m[1/9/2023, 2:11:34 PM] [0m[0;36m[Homebridge UI][0m UI / Bridge settings have not changed; only restarting Homebridge process
[0;37m[1/9/2023, 2:11:34 PM] [0m[0;36m[Homebridge UI][0m Sending SIGTERM to Homebridge
[37m[1/9/2023, 2:11:34 PM] [39mGot SIGTERM, shutting down Homebridge...
[37m[1/9/2023, 2:11:34 PM] [39mGot SIGTERM, shutting down child bridge process...
[37m[1/9/2023, 2:11:34 PM] [39mGot SIGTERM, shutting down child bridge process...
[37m[1/9/2023, 2:11:35 PM] [39m[36m[homebridge-enphase-envoy][39m Device: 10.0.1.153 Enphase, debug live data: {
  "connection": {
    "mqtt_state": "connected",
    "prov_state": "configured",
    "auth_state": "ok",
    "sc_stream": "disabled",
    "sc_debug": "disabled"
  },
  "meters": {
    "last_update": 0,
    "soc": 0,
    "main_relay_state": 13,
    "gen_relay_state": 0,
    "backup_bat_mode": 0,
    "backup_soc": 0,
    "pv": {
      "agg_p_mw": 0,
      "agg_s_mva": 0
    },
    "storage": {
      "agg_p_mw": 0,
      "agg_s_mva": 0
    },
    "grid": {
      "agg_p_mw": 0,
      "agg_s_mva": 0
    },
    "load": {
      "agg_p_mw": 0,
      "agg_s_mva": 0
    },
    "generator": {
      "agg_p_mw": 0,
      "agg_s_mva": 0
    }
  },
  "tasks": {
    "task_id": 1277979412,
    "timestamp": 1673170202
  },
  "counters": {
    "main_CfgLoad": 1,
    "main_CfgChanged": 1,
    "main_taskUpdate": 1,
    "MqttClient_respond": 1,
    "MqttClient_msgarrvd": 1,
    "MqttClient_create": 7059,
    "MqttClient_setCallbacks": 7059,
    "MqttClient_connect": 7059,
    "MqttClient_connect_err": 7046,
    "MqttClient_connect_Err": 7046,
    "MqttClient_subscribe": 13,
    "SSL_Keys_Create": 7059,
    "rest_Status": 39830
  }
}
[37m[1/9/2023, 2:11:36 PM] [39m[36m[homebridge-enphase-envoy][39m Device: 10.0.1.153 Enphase, debug production ct: {
  "production": [
    {
      "type": "inverters",
      "activeCount": 14,
      "readingTime": 1673269620,
[... 4958 lines cut ...]
      "agg_p_mw": 0,
      "agg_s_mva": 0
    }
  },
  "tasks": {
    "task_id": 1277979412,
    "timestamp": 1673170202
  },
  "counters": {
    "main_CfgLoad": 1,
    "main_CfgChanged": 1,
    "main_taskUpdate": 1,
    "MqttClient_respond": 1,
    "MqttClient_msgarrvd": 1,
    "MqttClient_create": 7059,
    "MqttClient_setCallbacks": 7059,
    "MqttClient_connect": 7059,
    "MqttClient_connect_err": 7046,
    "MqttClient_connect_Err": 7046,
    "MqttClient_subscribe": 13,
    "SSL_Keys_Create": 7059,
    "rest_Status": 39875
  }
}
```

grzegorz914, 2023-01-09:
please try 6.0.9-beta.2 and let me know

hugo-leij, 2023-01-09:
With 6.0.9-beta.4 no errors!

grzegorz914, 2023-01-09:
can you post also complete log?

closed 2023-01-09 by grzegorz914

hugo-leij, 2023-01-09:
Update to: 6.0.9

Log:
```
[0;37m[1/9/2023, 4:50:26 PM] [0m[0;36m[Homebridge UI][0m Changes to config.json saved.
[0;37m[1/9/2023, 4:50:28 PM] [0m[0;36m[Homebridge UI][0m Homebridge restart request received
[0;37m[1/9/2023, 4:50:28 PM] [0m[0;36m[Homebridge UI][0m UI / Bridge settings have not changed; only restarting Homebridge process
[0;37m[1/9/2023, 4:50:28 PM] [0m[0;36m[Homebridge UI][0m Sending SIGTERM to Homebridge
[37m[1/9/2023, 4:50:28 PM] [39mGot SIGTERM, shutting down Homebridge...
[37m[1/9/2023, 4:50:28 PM] [39mGot SIGTERM, shutting down child bridge process...
[37m[1/9/2023, 4:50:28 PM] [39mGot SIGTERM, shutting down child bridge process...
[37m[1/9/2023, 4:50:33 PM][0m [36m[HB Supervisor][0m Homebridge Process Ended. Code: 143, Signal: null
[37m[1/9/2023, 4:50:38 PM][0m [36m[HB Supervisor][0m Restarting Homebridge...
[37m[1/9/2023, 4:50:38 PM][0m [36m[HB Supervisor][0m Starting Homebridge with extra flags: -I -P /var/lib/homebridge/node_modules --strict-plugin-resolution
[37m[1/9/2023, 4:50:38 PM][0m [36m[HB Supervisor][0m Started Homebridge v1.6.0 with PID: 29299
[37m[1/9/2023, 4:50:39 PM] [39mLoaded config.json with 1 accessories and 8 platforms.
[37m[1/9/2023, 4:50:39 PM] [39mLoaded 18 cached accessories from cachedAccessories.
[37m[1/9/2023, 4:50:39 PM] [39m---
[37m[1/9/2023, 4:50:39 PM] [39mLoaded plugin: homebridge-dummy@0.8.0
[37m[1/9/2023, 4:50:39 PM] [39mRegistering accessory 'homebridge-dummy.DummySwitch'
[37m[1/9/2023, 4:50:39 PM] [39m---
[37m[1/9/2023, 4:50:40 PM] [39mLoaded plugin: homebridge-edomoticz@2.1.46
[37m[1/9/2023, 4:50:40 PM] [39mRegistering platform 'homebridge-edomoticz.eDomoticz'
[37m[1/9/2023, 4:50:40 PM] [39m---
[37m[1/9/2023, 4:50:40 PM] [39mLoaded plugin: homebridge-enphase-envoy@6.0.9-beta.4
[37m[1/9/2023, 4:50:40 PM] [39mRegistering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[37m[1/9/2023, 4:50:40 PM] [39m---
[37m[1/9/2023, 4:50:40 PM] [39m[33mDisabled plugin: homebridge-harmony@1.6.2[39m
[37m[1/9/2023, 4:50:40 PM] [39m---
[37m[1/9/2023, 4:50:40 PM] [39m[33mDisabled plugin: homebridge-lg-thinq@1.3.0[39m
[37m[1/9/2023, 4:50:40 PM] [39m---
[37m[1/9/2023, 4:50:41 PM] [39m[33mDisabled plugin: homebridge-slide-shutter@1.0.6[39m
[37m[1/9/2023, 4:50:41 PM] [39m---
[37m[1/9/2023, 4:50:41 PM] [39m[33mDisabled plugin: homebridge-sonos-starter-track@1.0.19[39m
[37m[1/9/2023, 4:50:41 PM] [39m---
[37m[1/9/2023, 4:50:41 PM] [39mLoaded plugin: homebridge-tado-platform@6.0.14
[37m[1/9/2023, 4:50:41 PM] [39mRegistering platform 'homebridge-tado-platform.TadoPlatform'
[37m[1/9/2023, 4:50:41 PM] [39m---
[37m[1/9/2023, 4:50:42 PM] [39mLoaded plugin: homebridge-tesla@4.2.0
[37m[1/9/2023, 4:50:42 PM] [39mRegistering accessory 'homebridge-tesla.Tesla'
[37m[1/9/2023, 4:50:42 PM] [39m---
[37m[1/9/2023, 4:50:42 PM] [39mLoaded plugin: homebridge-unifi-protect@5.5.4
[37m[1/9/2023, 4:50:42 PM] [39mRegistering platform 'homebridge-unifi-protect.UniFi Protect'
[37m[1/9/2023, 4:50:42 PM] [39m---
[37m[1/9/2023, 4:50:42 PM] [39mLoading 8 platforms...
[37m[1/9/2023, 4:50:42 PM] [39m[33mIgnoring config for the platform "LGThinQ" in your config.json as the plugin "homebridge-lg-thinq" has been disabled.[39m
[37m[1/9/2023, 4:50:42 PM] [39m[33mIgnoring config for the platform "slide-shutter" in your config.json as the plugin "homebridge-slide-shutter" has been disabled.[39m
[37m[1/9/2023, 4:50:42 PM] [39m[33mIgnoring config for the platform "HarmonyHubWebSocket" in your config.json as the plugin "homebridge-harmony" has been disabled.[39m
[37m[1/9/2023, 4:50:42 PM] [39m[36m[TadoPlatform][39m Initializing TadoPlatform platform...
[37m[1/9/2023, 4:50:44 PM] [39m[36m[Domoticz][39m Initializing eDomoticz platform...
[37m[1/9/2023, 4:50:44 PM] [39m[36m[UniFi Protect][39m Initializing UniFi Protect platform...
[37m[1/9/2023, 4:50:44 PM] [39m[36m[UniFi Protect][39m Initializing child bridge 0E:DE:96:9B:BC:AB
[37m[1/9/2023, 4:50:44 PM] [39m[36m[enphaseEnvoy][39m Initializing enphaseEnvoy platform...
[37m[1/9/2023, 4:50:44 PM] [39m[36m[enphaseEnvoy][39m Initializing child bridge 0E:65:FD:E1:9B:69
[37m[1/9/2023, 4:50:44 PM] [39mLoading 1 accessories...
[37m[1/9/2023, 4:50:44 PM] [39m[36m[Model S][39m Initializing Tesla accessory...
[37m[1/9/2023, 4:50:45 PM] [39m[36m[TadoPlatform][39m Allerheiligenweg Huiskamer Heater: Configuring accessory...
[37m[1/9/2023, 4:50:45 PM] [39m[36m[Domoticz][39m synchronizeAccessories in progress...
Setup Payload:
X-HM://0023I7JB6KXU8
Enter this code with your HomeKit app on your iOS device to pair with Homebridge:
[30m[47m                       [49m[39m
[30m[47m    ┌────────────┐     [49m[39m
[30m[47m    │ ***** │     [49m[39m
[30m[47m    └────────────┘     [49m[39m
[30m[47m                       [49m[39m
[37m[1/9/2023, 4:50:45 PM] [39m[36m[Model S][39m Exchanging refresh token for an access token…
[37m[1/9/2023, 4:50:45 PM] [39mHomebridge v1.6.0 (HAP v0.11.0) (Homebridge) is running on port 51825.
[37m[1/9/2023, 4:50:45 PM] [39m[36m[Domoticz][39m There was an error while getting the MQTT Hardware Device from Domoticz.
Please verify that you have added the MQTT Hardware Device and that the hardware device is enabled.
[... 4503 lines cut ...]
      "agg_s_mva": 0
    }
  },
  "tasks": {
    "task_id": 1277979412,
    "timestamp": 1673170202
  },
  "counters": {
    "main_CfgLoad": 1,
    "main_CfgChanged": 1,
    "main_taskUpdate": 1,
    "MqttClient_respond": 1,
    "MqttClient_msgarrvd": 1,
    "MqttClient_create": 7059,
    "MqttClient_setCallbacks": 7059,
    "MqttClient_connect": 7059,
    "MqttClient_connect_err": 7046,
    "MqttClient_connect_Err": 7046,
    "MqttClient_subscribe": 13,
    "SSL_Keys_Create": 7059,
    "rest_Status": 43914
  }
}

```

grzegorz914, 2023-01-09:
Thanks, 6.0.9 released

grzegorz914, 2023-01-09:
@GevaudanBeast 

debug log please

GevaudanBeast, 2023-01-09:
HB version 1.4.1
Plugin version 6.0.9
`Loading logs using "systemd" method...
CMD: sudo -n journalctl -o cat -n 500 -f -u homebridge

The log tail command "-n journalctl -o cat -n 500 -f -u homebridge" exited with code 0.
Please check the command in your config.json is correct.

See https://github.com/oznu/homebridge-config-ui-x#log-viewer-configuration for instructions.
        "last_rpt_date": "1673281325",
        "admin_state": 2,
        "dev_type": 1,
        "created_date": "1656005853",
        "img_load_date": "1656502831",
        "img_pnum_running": "520-00082-r01-v04.28.07",
        "ptpn": "540-00169-r01-v04.28.93",
        "chaneid": 1627390993,
        "device_control": [
          {
            "gficlearset": false
          }
        ],
        "producing": false,
        "communicating": true,
        "provisioned": true,
        "operating": false
      },
      {
        "part_num": "800-01127-r02",
        "installed": "1662981639",
        "serial_num": "482229010903",
        "device_status": [
          "envoy.cond_flags.pcu_ctrl.dc-pwr-low"
        ],
        "last_rpt_date": "1673281355",
        "admin_state": 2,
        "dev_type": 1,
        "created_date": "1662981639",
        "img_load_date": "1613405094",
        "img_pnum_running": "520-00082-r01-v04.27.04",
        "ptpn": "540-00169-r01-v04.27.09",
        "chaneid": 1627391249,
        "device_control": [
          {
            "gficlearset": false
          }
        ],
        "producing": true,
        "communicating": true,
        "provisioned": true,
        "operating": true
      },
      {
        "part_num": "800-01127-r02",
        "installed": "1662981640",
        "serial_num": "482229010806",
        "device_status": [
          "envoy.cond_flags.pcu_ctrl.dc-pwr-low"
        ],
        "last_rpt_date": "1673281384",
        "admin_state": 2,
        "dev_type": 1,
        "created_date": "1662981640",
        "img_load_date": "1613405094",
        "img_pnum_running": "520-00082-r01-v04.27.04",
        "ptpn": "540-00169-r01-v04.27.09",
        "chaneid": 1627391505,
        "device_control": [
          {
            "gficlearset": false
[... 418 lines cut ...]
/usr/local/lib/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/settle.js:19
    reject(new AxiosError(
           ^
AxiosError: Request failed with status code 401
    at settle (/usr/local/lib/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/settle.js:19:12)
    at IncomingMessage.handleStreamEnd (/usr/local/lib/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/adapters/http.js:512:11)
    at IncomingMessage.emit (node:events:539:35)
    at endReadableNT (node:internal/streams/readable:1345:12)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
[09/01/2023, 16:35:03] [homebridge-enphase-envoy] Child bridge process ended
[09/01/2023, 16:35:03] [homebridge-enphase-envoy] Process Ended. Code: 1, Signal: null
homebridge : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/kill -9 10868
pam_unix(sudo:session): session opened for user root by (uid=0)
pam_unix(sudo:session): session closed for user root
homebridge : TTY=pts/0 ; PWD=/var/lib/homebridge ; USER=root ; COMMAND=/bin/journalctl -o cat -n 500 -f -u homebridge
pam_unix(sudo:session): session opened for user root by (uid=0)
homebridge : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/kill -9 10938
homebridge : TTY=pts/1 ; PWD=/var/lib/homebridge ; USER=root ; COMMAND=/bin/journalctl -o cat -n 500 -f -u homebridge
pam_unix(sudo:session): session opened for user root by (uid=0)
pam_unix(sudo:session): session opened for user root by (uid=0)
pam_unix(sudo:session): session closed for user root
[09/01/2023, 16:35:10] [homebridge-enphase-envoy] Restarting Process...
[09/01/2023, 16:35:10] [Baie 1.8m] Current position (retrieved via http): 0%
[09/01/2023, 16:35:10] [Fenêtre cuisine] Current position (retrieved via http): 0%
[09/01/2023, 16:35:10] [Chambre fenêtre] Current position (retrieved via http): 10`

commit 1e7afcd3 (2023-01-09): fix #69