grzegorz914/homebridge-enphase-envoy #71: Incorrect inventory with firmware 7

cbrandlehner, 2023-01-13:
My enphase installation features 34 microinverters and 3 batteries.

[image]

Here a screenshot showing battery details:
[image]

However, the plugin states "Batteries: 0"

```
[1/13/2023, 8:54:10 PM] [homebridge-enphase-envoy] Restarting Process...
[1/13/2023, 8:54:10 PM] [homebridge-enphase-envoy] Launched child bridge with PID 3168
[1/13/2023, 8:54:10 PM] Registering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[1/13/2023, 8:54:10 PM] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v6.3.1 child bridge successfully
[1/13/2023, 8:54:10 PM] Loaded 0 cached accessories from cachedAccessories.0E9D3754CD5E.
[1/13/2023, 8:54:10 PM] Homebridge v1.6.0 (HAP v0.11.0) (homebridge-enphase-envoy) is running on port 56673.
[1/13/2023, 8:54:10 PM] [homebridge-enphase-envoy] Device: 192.168.71.147 Envoy-S, MQTT Connected.
[1/13/2023, 8:54:23 PM] [homebridge-enphase-envoy] -------- Envoy-S --------
[1/13/2023, 8:54:23 PM] [homebridge-enphase-envoy] Manufacturer: Enphase
[1/13/2023, 8:54:23 PM] [homebridge-enphase-envoy] Model: ENV-S-WM-230
[1/13/2023, 8:54:23 PM] [homebridge-enphase-envoy] Firmware: D7.6.111
[1/13/2023, 8:54:23 PM] [homebridge-enphase-envoy] SerialNr: 122134029222
[1/13/2023, 8:54:23 PM] [homebridge-enphase-envoy] Time: 1/13/2023, 8:55:08 PM
[1/13/2023, 8:54:23 PM] [homebridge-enphase-envoy] ------------------------------
[1/13/2023, 8:54:23 PM] [homebridge-enphase-envoy] Q-Relays: 5
[1/13/2023, 8:54:23 PM] [homebridge-enphase-envoy] Inverters: 34
[1/13/2023, 8:54:23 PM] [homebridge-enphase-envoy] Batteries: 0
[1/13/2023, 8:54:23 PM] [homebridge-enphase-envoy] --------------------------------
[1/13/2023, 8:54:23 PM] [homebridge-enphase-envoy] Meters: Yes
[1/13/2023, 8:54:23 PM] [homebridge-enphase-envoy] Production: Enabled
[1/13/2023, 8:54:23 PM] [homebridge-enphase-envoy] Consumption: Enabled
[1/13/2023, 8:54:23 PM] [homebridge-enphase-envoy] --------------------------------
[1/13/2023, 8:54:23 PM] [homebridge-enphase-envoy] Ensemble: No
[1/13/2023, 8:54:23 PM] [homebridge-enphase-envoy] --------------------------------
```

grzegorz914, 2023-01-13:
Please post the debug log, enable in plugin settings >> advanced section

cbrandlehner, 2023-01-13:
Debug log was too large to share in a comment. Here is a link to github gist.

[gist](https://gist.github.com/cbrandlehner/fcd777e8677bc0191ea20f049f676f5a)

grzegorz914, 2023-01-14:
Thanks, please try 6.3.2-beta.1 and let me know, also log please.

cbrandlehner, 2023-01-14:
New debug log on [gist](https://gist.github.com/cbrandlehner/68193f9980aee740465a5a43322a2f2f).

The relevant part might be this one:

[1/14/2023, 9:13:40 AM] [homebridge-enphase-envoy] Device: 192.168.71.147 Envoy-S, debug ensemble inventory: [
  {
    "type": "ENCHARGE",
    "devices": [
      {
        "part_num": "830-01760-r09",
        "installed": 1672585152,
        "serial_num": "122118038611",
        "device_status": [
          "envoy.global.ok",
          "prop.done"
        ],
        "last_rpt_date": 1673683867,
        "admin_state": 6,
        "admin_state_str": "ENCHG_STATE_READY",
        "created_date": 1672585152,
        "img_load_date": 1672585152,
        "img_pnum_running": "2.6.5549_rel/22.11",
        "zigbee_dongle_fw_version": "1009",
        "bmu_fw_version": "2.1.16",
        "operating": true,
        "communicating": true,
        "sleep_enabled": false,
        "percentFull": 5,
        "temperature": 10,
        "maxCellTemp": 11,
        "comm_level_sub_ghz": 3,
        "comm_level_2_4_ghz": 3,
        "led_status": 12,
        "dc_switch_off": false,
        "encharge_rev": 2,
        "encharge_capacity": 3500
      },
      {
        "part_num": "830-01760-r09",
        "installed": 1672585147,
        "serial_num": "122119003446",
        "device_status": [
          "envoy.global.ok",
          "prop.done"
        ],
        "last_rpt_date": 1673683932,
        "admin_state": 6,
        "admin_state_str": "ENCHG_STATE_READY",
        "created_date": 1672585147,
        "img_load_date": 1672585147,
        "img_pnum_running": "2.6.5549_rel/22.11",
        "zigbee_dongle_fw_version": "1009",
        "bmu_fw_version": "2.1.16",
        "operating": true,
        "communicating": true,
        "sleep_enabled": false,
        "percentFull": 5,
        "temperature": 10,
        "maxCellTemp": 10,
        "comm_level_sub_ghz": 3,
        "comm_level_2_4_ghz": 3,
        "led_status": 12,
        "dc_switch_off": false,
        "encharge_rev": 2,
        "encharge_capacity": 3500
      },
      {
        "part_num": "830-01760-r32",
        "installed": 1672585246,
        "serial_num": "122223055385",
        "device_status": [
          "envoy.global.ok",
          "prop.done"
        ],
        "last_rpt_date": 1673683892,
        "admin_state": 6,
        "admin_state_str": "ENCHG_STATE_READY",
        "created_date": 1672585246,
        "img_load_date": 1672585246,
        "img_pnum_running": "2.6.5549_rel/22.11",
        "zigbee_dongle_fw_version": "100B",
        "bmu_fw_version": "2.1.16",
        "operating": true,
        "communicating": true,
        "sleep_enabled": false,
        "percentFull": 5,
        "temperature": 10,
        "maxCellTemp": 11,
        "comm_level_sub_ghz": 3,
        "comm_level_2_4_ghz": 3,
        "led_status": 12,
        "dc_switch_off": false,
        "encharge_rev": 2,
        "encharge_capacity": 3500
      }
    ]
  }
]`
[1/14/2023, 9:13:40 AM] [homebridge-enphase-envoy] Device: 192.168.71.147 Envoy-S, Update ensemble inventory data error: TypeError: Cannot read properties of undefined (reading 'devices'), reconnect in 15s.`

grzegorz914, 2023-01-14:
try beta.2, if no error please post complete debug log

cbrandlehner, 2023-01-14:
New file on [gist](https://gist.github.com/cbrandlehner/31a7ba306a4f26db1d726c40ed15c95c).

Looking good, although I see some warnings after successfully reading the inventory:

`[1/14/2023, 11:56:13 AM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Energy lifetime': characteristic was supplied illegal value: number -2779.829 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.`
This is a known issue #72 

```
[1/14/2023, 11:56:13 AM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Grid profile': characteristic value expected string and received undefined. See https://homebridge.io/w/JtMGR for more info.
[1/14/2023, 11:56:13 AM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Grid profile': characteristic value expected string and received undefined. See https://homebridge.io/w/JtMGR for more info.
[1/14/2023, 11:56:13 AM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Grid profile': characteristic value expected string and received undefined. See https://homebridge.io/w/JtMGR for more info.
[1/14/2023, 11:56:13 AM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'AGG backup energy': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.
[1/14/2023, 11:56:13 AM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'AGG available energy': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.
```
This seems to be new warnings.

grzegorz914, 2023-01-14:
Nice try beta.3

grzegorz914, 2023-01-14:
beta.4 is UP, fixed AGG warnings

cbrandlehner, 2023-01-14:
New gile on [gist](https://gist.github.com/cbrandlehner/654055053d8fe72f6841b1e80271d710)

"Grid profile" is now looking good.
"AGG backup energy" still showing a warning.

Let me add something in this context:
I was having a discussion with Enphase, asking why I am not able to choose from different battery profiles. They explained that this is due to regulations in Austria, so they blocked this feature in the app on purpose.

[image]

cbrandlehner, 2023-01-14:
Beta-5 debug on [gist](https://gist.github.com/cbrandlehner/91c36a78c65f55b0b3fb7e11b5e915d9)

grzegorz914, 2023-01-14:
workaround energy lifetime added in beta.6, ready to test

cbrandlehner, 2023-01-14:
Beta 6 homebridge.log on [gist](https://gist.github.com/cbrandlehner/6150d291af78450635cd4c77213bcf9e)

Thanks, excellent work - all warnings are gone.

grzegorz914, 2023-01-14:
Regardingg lock of load different grid profile,  the workaround is to use old Installer App. It working in my case but tested only with Micro and qRelays, not tested with Encharge(not installed in my enviroment)

closed 2023-01-14 by grzegorz914

cbrandlehner, 2023-01-14:
Thanks.

I will now delete all debug logs on gist for privacy reasons.

grzegorz914, 2023-01-14:
Thanks for test.

commit c6ee2312 (2023-01-14): fix #71 #72