grzegorz914/homebridge-enphase-envoy #73: 6.3.2 don't work with firmware 7

BenouGui, 2023-01-15:
Hello, 
6.3.2 don't work with firmware 7 

Thank

grzegorz914, 2023-01-15:
6.3.1 works?

BenouGui, 2023-01-15:
yes

grzegorz914, 2023-01-15:
can you post debug

BenouGui, 2023-01-15:
[
[homebridge.log-28.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10420017/homebridge.log-28.txt)
]

grzegorz914, 2023-01-15:
try 6.4.0-beta.8 please

BenouGui, 2023-01-15:
[homebridge.log-29.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10420154/homebridge.log-29.txt)

nok :/

grzegorz914, 2023-01-15:
try .11

BenouGui, 2023-01-15:
[homebridge.log-30.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10420168/homebridge.log-30.txt)
 it ok I think

grzegorz914, 2023-01-15:
OK, can you get the whole log after plugin restart, I want to compare with other 7.x.x firmware, looks like Enphase publish different local api in different 7.x.x firmwares

BenouGui, 2023-01-15:
[homebridge.log-31.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10420186/homebridge.log-31.txt)

grzegorz914, 2023-01-15:
6.4.0 released

closed 2023-01-15 by grzegorz914