grzegorz914/homebridge-enphase-envoy #74: AC voltage/frequency out of range, grid instability

GrossyGrosmann, 2023-01-25:
Just a general question; I’ve been using this plugin without an issue for the past 12months. Lately my solar runs for an hour or so then cuts out (production goes to 0) and I get the message “AC voltage and frequency out of range, grid instability”
I’ve had both local energy company and installers out and can’t find issues (they didn’t look to hard)… is there anything in this plugin I could I have potentially played around with (a setting) to cause this issue? I don’t have access to the installer toolkit or anything within enphase, just wondering if anyone has faced this issue before?
Any help would be appreciated! It may not have anything to do with the plugin, but would love clarity before closing this issue.

grzegorz914, 2023-01-26:
This happen if frequency or voltage go outside the range, it’s normal Microinverters behavior to protect Netz and devices to be damaged. The plugin have nothing to do only display this info.  Power mode will automatically on after Netz return to its normal state.

closed 2023-01-26 by grzegorz914

GrossyGrosmann, 2023-01-26:
Thank you @grzegorz914 for the reassurance, and thank you (and the others devs) for developing an awesome plugin!