grzegorz914/homebridge-enphase-envoy #75: Parse Error: Content-Length can't be present with Transfer-Encoding

dstupakov, 2023-02-02:
Hi,

I get the following error when trying to run your plugin. Is this a known issue, or something I can fix in the config? 

[2/2/2023, 3:15:35 PM] [homebridge-enphase-envoy] Device: envoy.local Envoy, Update home data error: Error: Parse Error: Content-Length can't be present with Transfer-Encoding, reconnect in 15s.

Thank you for your contribution.

grzegorz914, 2023-02-03:
Post Your config Please, also  Please enable debug in plugin config advanced section and get the complete log again.

HGordonH, 2023-02-10:
Hi - Ive started to get the same error

grzegorz914, 2023-02-11:
I need complete debug log, Please go to plugin section advanced settings and enable debug, then save and restart plugin. Next post the complete log.

HGordonH, 2023-02-11:
looked at the logs again and seems to be OK now, no errors. When I posted the issue there was no solar generation, there is now and error no longer shows. I’ll check again tonight and if the error comes back I’ll post the log.

commit 80c79f5b (2023-02-14): fix #75
closed 2023-02-14 by grzegorz914