grzegorz914/homebridge-enphase-envoy #76: Cannot get it to work

reinhochstenbach, 2023-02-16:
Hi. I installed homebridge, added the bridge to my home app.
Installed this enphase envoy plugin
But am confused on how to proceed now. 
I somehow need to tell the plugin where to find the envoy I assume and how to use it?
So I add new device in the plugin settings, and enter the ip mentioned by my router as the ip voor envoy.
But I cannot see an accessory to add in home app.
In log for homebridge I see this :
[2/16/2023, 12:51:55 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, start.
[2/16/2023, 12:51:55 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, requesting info.

grzegorz914, 2023-02-16:
Look at this https://github.com/grzegorz914/homebridge-enphase-envoy/wiki#adding-to-homekit

reinhochstenbach, 2023-02-16:
Each accessory needs to be manually paired.
Open the Home  app on your device.
Tap the Home tab, then tap .
Tap Add Accessory, and select I Don't Have a Code, Cannot Scan or More options.
**Select Your accessory and press add anyway.**
Enter the PIN or scan the QR code, this can be found in Homebridge UI or Homebridge logs.
Complete the accessory setup.

There is no accessory to add...

grzegorz914, 2023-02-16:
Please enable debug mode in plugin config advanced section restart HB and get the complete log

reinhochstenbach, 2023-02-16:
Loading logs using "custom" method...
CMD: pm2 logs --raw --out --lines 80 homebridge

[TAILING] Tailing last 80 lines for [homebridge] process (change the value with --lines option)
/root/.pm2/logs/homebridge-out.log last 80 lines:
[2/16/2023, 4:56:59 PM] [Ring] Initializing Ring platform...
[2/16/2023, 4:56:59 PM] [enphaseEnvoy] Initializing enphaseEnvoy platform...
[2/16/2023, 4:56:59 PM] [Ring] Configuring cached accessory d90xxxxxxxxxx Voordeur
[2/16/2023, 4:56:59 PM] [Ring] [Circular]
[2/16/2023, 4:56:59 PM] [Ring] Configuring cached accessory 889fdfdxxxxxx Beneden
[2/16/2023, 4:56:59 PM] [Ring] [Circular]
[2/16/2023, 4:56:59 PM] [Ring] Configuring cached accessory 889fdfdxxxxxx 'S-Hertogenbosch Mode
[2/16/2023, 4:56:59 PM] [Ring] [Circular]
[2/16/2023, 4:56:59 PM] [Ring] didFinishLaunching
[2/16/2023, 4:56:59 PM] [enphaseEnvoy] didFinishLaunching
[2/16/2023, 4:56:59 PM] [enphaseEnvoy] Device: xxx.xxx.x.xxx  Envoy, start.
[2/16/2023, 4:56:59 PM] [enphaseEnvoy] Device: xxx.xxx.x.xxx Envoy, requesting info.
[2/16/2023, 4:56:59 PM] Publishing bridge accessory (name: Homebridge, publishInfo: {
  username: 'XX',
  port: 51826,
  pincode: '***-**-***',
  category: 2,
  bind: undefined,
  mdns: undefined,
  addIdentifyingMaterial: true,
  advertiser: undefined
}).
Setup Payload:
X-HM://xx
Scan this code with your HomeKit app on your iOS device to pair with Homebridge:
                   
Or enter this code with your HomeKit app on your iOS device to pair with Homebridge:
                       
    ┌────────────┐     
    │ xxx-xx-xxx  
    └────────────┘     
                       
[2/16/2023, 4:56:59 PM] Homebridge v1.6.0 (HAP v0.11.0) (Homebridge) is running on port 51826.
[2/16/2023, 4:57:01 PM] [Ring] Found the following locations:
[2/16/2023, 4:57:01 PM] [Ring]   locationId: xxxxxxxx - 'S-Hertogenbosch
[2/16/2023, 4:57:01 PM] [Ring] Configuring 1 cameras and 3 devices for location "'S-Hertogenbosch" - locationId: xxxxx
[2/16/2023, 4:57:15 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, start.
[2/16/2023, 4:57:15 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, requesting info.
[2/16/2023, 4:57:30 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, start.
[2/16/2023, 4:57:30 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, requesting info.
[2/16/2023, 4:57:45 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, start.
[2/16/2023, 4:57:45 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, requesting info.
[2/16/2023, 4:58:00 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, start.
[2/16/2023, 4:58:00 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, requesting info.
[2/16/2023, 4:58:15 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, start.
[2/16/2023, 4:58:15 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, requesting info.
[2/16/2023, 4:58:31 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, start.
[2/16/2023, 4:58:31 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, requesting info.
[2/16/2023, 4:58:46 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, start.
[2/16/2023, 4:58:46 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, requesting info.
[2/16/2023, 4:59:01 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, start.
[2/16/2023, 4:59:01 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, requesting info.
[2/16/2023, 4:59:16 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, start.
[2/16/2023, 4:59:16 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, requesting info.

[2/16/2023, 4:59:31 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, start.
[2/16/2023, 4:59:31 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, requesting info.
[2/16/2023, 4:59:47 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, start.
[2/16/2023, 4:59:47 PM] [enphaseEnvoy] Device: 192.168.1.253 Envoy, requesting info.

grzegorz914, 2023-02-16:
It’s not a debug from my plugin, you need carefully read what I wrote in previous post.

reinhochstenbach, 2023-02-16:
Ok. I try :-)
_Please enable debug mode in plugin config advanced section restart HB and get the complete log:_
I selected the settings for the plugin, enabled debug, restarted homebridge and copied what I saw in the log.
Apparently there is another log where the debug for the plugin is?

So I entered shell, to find the logs for the plugin:

root@Homebridge:~/.homebridge # ls
.ring.json              auth.json               enphaseEnvoy
.uix-dashboard.json     backups                 persist
.uix-secrets            config.json
accessories             config.json-e
root@Homebridge:~/.homebridge # cd enphaseEnvoy/
root@Homebridge:~/.homebridge/enphaseEnvoy # ls
consumptionNetPowerPeak_1921681253      envoyId_1921681253
consumptionTotalPowerPeak_1921681253    productionPowerPeak_1921681253
root@Homebridge:~/.homebridge/enphaseEnvoy #

No clue as where to find the debug for the plugin?

Sorry if I am wasting your time, I am fairly new to this.
Regards

grzegorz914, 2023-02-16:
You need go to this points:
[image]
[image]

The log will be displayed in Config UI X log tab

reinhochstenbach, 2023-02-16:
Thanks, i did those steps, but I do not have a homebridge config ui x log tab.:
[image]

reinhochstenbach, 2023-02-16:
wow, I think i triggered something, by selecting the firmware 7.xx option and copying the enphase auth token key from the suggested site. There is tons of logs created now, but I cannot read it, it goes super fast.. :-(

grzegorz914, 2023-02-26:
Ok, I close this as resolved.

closed 2023-02-26 by grzegorz914