grzegorz914/homebridge-enphase-envoy #77: Am I supposed to see data from the microinverters within the Controller App?

kvpoma, 2023-02-19:
I can see on my homebridge that my microinverters are reporting, but when I go into the Controller app, all I can see is the Envoy as an accessory.  I’m not seeing any other data.  Is there something that I need to edit within the settings to get the data into the Controller app?

grzegorz914, 2023-02-19:
Are you added this accessory to Home.app?

kvpoma, 2023-02-19:
Have I? Yes.  Using my Homebridge HomeKit code, I added the envoy to the Home app, although I don't see it in any of my rooms

grzegorz914, 2023-02-19:
Accessory itself will not be displayed in Apple Home.app because do not support custom characteristics, only sensors will be displayed.. 
In Controller or EVE app need to be displayed all services as Envoy, Microinverters, Meters, etc.

[image]

kvpoma, 2023-02-19:
I understand that the data can’t be showed in HomeKit. My problem is the data isn’t showing up in the controller app like yours. All I see is the envoy. What settings do I need to configure in the homebridge plugin to have the data show in controller app?

grzegorz914, 2023-02-19:
Can you post your Config here. What is displayed in HB log console?

kvpoma, 2023-02-19:
[image]

kvpoma, 2023-02-22:
Any other info needed?

grzegorz914, 2023-02-26:
Enable debug mode in plugin Config and get the complete log please.

kvpoma, 2023-02-26:
[Enphaselog.txt](https://github.com/grzegorz914/homebridge-enphase-envoy/files/10834368/Enphaselog.txt)
See attached

grzegorz914, 2023-02-26:
All is correct in log and accessory Is published, are you sure you have added it correct to the Home.app, can you record video or screenshots.

kvpoma, 2023-02-26:
https://user-images.githubusercontent.com/68412035/221431314-0731260e-e16f-4424-b1f3-e2e652899c22.MOV

grzegorz914, 2023-02-26:
You need add the accessory itself not the bridge, in your case is not visible use this:
https://github.com/grzegorz914/homebridge-enphase-envoy/wiki#troubleshooting 
procedure to fix this and try again. If not help clear HB cache and old accessory.

kvpoma, 2023-02-26:
Thanks for troubleshooting this for me. Could you give me the command lines to get to there?

grzegorz914, 2023-02-26:
What so you use

kvpoma, 2023-02-26:
RPi running homebridge

grzegorz914, 2023-02-26:
The files are in /var/lib/Homebridge/persist/ folder, but first Try to clear casche in HB and restart complete HB

kvpoma, 2023-02-26:
That was the ticket. Removed cached bridges fixed the issue. Consumption is showing up in Controller now. Thank you!

closed 2023-02-26 by grzegorz914