grzegorz914/homebridge-enphase-envoy #78: Question: Energy consumption of batteries

cbrandlehner, 2023-02-23:
I am trying to build some logic to charge an electric vehicle. I can change from 1- to 3-phase and I can change the amps.

The data I use is not HomeKit but the data broadcasted by this plugin to MQTT.

As a basis for my logic I am looking at PV production
  enphase/Envoy-S/Production CT - payload.consumption[0].wNow
and the Watts to or from the grid
  enphase/Envoy-S/Production CT - payload.consumption[1].wNow

Based on the last figure I can decide to charge the car (if available amps > 6).

However, If I want to give priority to charging the car over charging the encharge batteries, I miss additional information:
Either: how much consumption is there without the encharge batteries
OR how many Watts do the encharge batteries consume to charge.

grzegorz914, 2023-02-23:
This data is not available in any Api, you can check this itself how many watt is consumed to charge encharges and lather use it to calculate available power. Some data is available in Ensemble Status but I don’t think is that what you want.

cbrandlehner, 2023-02-25:
[image]

I would like to know the energy flow as displayed in the Enphase app. The app shows 0.6kW consumption and 1.8kW charging. The data I actually see is those two numbers combined - or I am completely wrong.

grzegorz914, 2023-02-25:
can you please enable debug in plugin config advanced section and get me the whole log. I can look into this but self have no encharges installed in my environment and is difficult to found it.

cbrandlehner, 2023-02-25:
@grzegorz914 thanks for looking into this.

Link to debug log: [Github gist](https://gist.github.com/cbrandlehner/95d22eb10a039338d519c65169bb270d)

grzegorz914, 2023-02-25:
@cbrandlehner 
im pushed 6.6.0-beta to test:

Added in Ensemble Summary to Mqtt  and HomeKit:
1. `encharge_rated_power` -  I think this is charging but note, this is charge rated power from 1 ENC device, in your case You have 3 ENC installed. So in the MQTT can be found 3 properties which need to be summarize.
2. `Max_energy `, `ENC_agg_soc`, `ENC_agg_avail_energy`
3. Consumption Power from your screenshot can me found in Mqtt Production CT `Consumption[0].wNow`, in Consumption[1].wNow is only consumption from Network, can be negative if production is greater as consumption.

I think this help you to realize your automation.
Let me know if this will be OK then pushed this update as release.

cbrandlehner, 2023-02-25:
"encharge_rated_power"  is a constant value of 1280. I think this is telling that in case of discharging, the battery can provide a maximum of 1280 Watts.
Max_energy: 10500 - 10.5 kWh battery capacity in total
ENC_agg_soc: 100 - all batteries are full at this time
ENC_agg_soh: 100 - State of Health (let's hope this one does not drop in the next years to come)
ENC_agg_backup_energy: 1050 - battery is configured to keep a backup of 10% (unfortunately)
ENC_agg_avail_energy: 10500 - same as Max_energy

I will have a look at "Production CT" tomorrow as the battery is currently full.

grzegorz914, 2023-02-25:
ENC_agg_avail_energy: 10500 - same as Max_energy this are two different values, ENC is for Encharges and Max energy get summary if you have already installed ACB old batteries

cbrandlehner, 2023-02-26:
[image]
[image]
The battery is currently charging so I checked the numbers again.
The two numbers in "Production CT" provide exactly what the name suggests.
"Total-consumption" is consumption of the house including the batteries.
"net-consumption" is almost zero, that number varies in a range from -50 to +50.

These two numbers plus production in a time series chart:
Batteries stopped backing the house at 07:40.
[image]

So I have no idea, how Enhase can visualize the energy flow with separate flow info for house and batteries. It seems these numbers are not exposed in the API.

grzegorz914, 2023-02-26:
Yes, it must be internal calculated but not exposed to the api. 
One tip, please also look on the Ensemble Summary `rest_Power', may be this the rest charging power.

cbrandlehner, 2023-02-26:
I just noticed that there is a new beta 3. That version causes errors in my setup:

```
[2/26/2023, 10:13:05 AM] [homebridge-enphase-envoy] Restarting Process...
[2/26/2023, 10:13:06 AM] [homebridge-enphase-envoy] Launched child bridge with PID 2787
[2/26/2023, 10:13:06 AM] Registering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[2/26/2023, 10:13:06 AM] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v6.6.0-beta.3 child bridge successfully
[2/26/2023, 10:13:06 AM] Loaded 0 cached accessories from cachedAccessories.0E9D3754CD5E.
[2/26/2023, 10:13:06 AM] Homebridge v1.6.0 (HAP v0.11.0) (homebridge-enphase-envoy) is running on port 56673.
[2/26/2023, 10:13:06 AM] [homebridge-enphase-envoy] Device: 192.168.71.147 Envoy-S, MQTT Connected.
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] -------- Envoy-S --------
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] Manufacturer: Enphase
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] Model: ENV-S-WM-230
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] Firmware: D7.6.125
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] SerialNr: 122134029222
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] Time: 2/26/2023, 10:13:19 AM
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] ------------------------------
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] Q-Relays: 5
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] Inverters: 34
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] Batteries: 0
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] --------------------------------
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] Meters: Yes
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] Production: Enabled
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] Consumption: Enabled
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] --------------------------------
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] Encharges: Yes 3
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] Wireless Kit: Yes 2
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] --------------------------------
[2/26/2023, 10:13:15 AM] [homebridge-enphase-envoy] Device: 192.168.71.147 Envoy-S, Error: Cannot add a Characteristic with the same UUID as another Characteristic in this Service: 00000192-000B-1000-8000-0026BB765291 Reconnect in 15s.
[2/26/2023, 10:13:36 AM] [homebridge-enphase-envoy] Device: 192.168.71.147 Envoy-S, Update ensemble status data error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Reconnect in 15s.
[2/26/2023, 10:13:57 AM] [homebridge-enphase-envoy] Device: 192.168.71.147 Envoy-S, Update ensemble status data error: TypeError: Cannot read properties of undefined (reading 'updateCharacteristic'). Reconnect in 15s.
```

grzegorz914, 2023-02-26:
Ups, fixed in 6.6.1

grzegorz914, 2023-02-26:
fixed too in 6.6.2

cbrandlehner, 2023-02-26:
Sorry, but again an error:

```
[2/26/2023, 10:29:27 AM] [homebridge-enphase-envoy] Restarting Process...
[2/26/2023, 10:29:28 AM] [homebridge-enphase-envoy] Launched child bridge with PID 3113
[2/26/2023, 10:29:28 AM] Registering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[2/26/2023, 10:29:28 AM] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v6.6.2 child bridge successfully
[2/26/2023, 10:29:28 AM] Loaded 0 cached accessories from cachedAccessories.0E9D3754CD5E.
[2/26/2023, 10:29:28 AM] Homebridge v1.6.0 (HAP v0.11.0) (homebridge-enphase-envoy) is running on port 56673.
[2/26/2023, 10:29:28 AM] [homebridge-enphase-envoy] Device: 192.168.71.147 Envoy-S, MQTT Connected.
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] -------- Envoy-S --------
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] Manufacturer: Enphase
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] Model: ENV-S-WM-230
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] Firmware: D7.6.125
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] SerialNr: 122134029222
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] Time: 2/26/2023, 10:29:40 AM
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] ------------------------------
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] Q-Relays: 5
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] Inverters: 34
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] Batteries: 0
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] --------------------------------
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] Meters: Yes
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] Production: Enabled
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] Consumption: Enabled
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] --------------------------------
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] Encharges: Yes 3
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] Wireless Kit: Yes 2
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] --------------------------------
[2/26/2023, 10:29:35 AM] [homebridge-enphase-envoy] Device: 192.168.71.147 Envoy-S, TypeError: Cannot read properties of undefined (reading 'onGet') Reconnect in 15s.
```

grzegorz914, 2023-02-26:
Sorry of mistake, was typo in characteristic, fixed in 6.6.3

cbrandlehner, 2023-02-26:
Confirmed, working now.

I think we can close this issue as I can not ask you to expose a characteristic not availabe with Enphases API.

Data currently flows from Homebridge-Plugin -> MQTT -> Node-Red -> InfluxDB -> Grafana. 

The result is this nice dashbaord:
[image]

grzegorz914, 2023-02-26:
Very nice, here my on iPad in HMI Editor:
[image]
[image]
[image]

grzegorz914, 2023-02-26:
Are you check `rest_Power` in Ensemble Status?

cbrandlehner, 2023-02-26:
I have no idea yet, what rest_Power is. I will need to observe that number.

In my setup, rest_Power does not change with load. While I observed it for some time I have seen a number of 1074 for some time and then it jumped to 1077. I have not seen any other numbers yet.

closed 2023-02-26 by grzegorz914