grzegorz914/homebridge-enphase-envoy #79: Issues on Firmware 7.X.X authentication

thdevita, 2023-02-26:
Hello.
I'm trying to setup the plugin with Envoy firmware 7.0.88
I have checked the firmware 7.X.X box in Authorisation and pasted the generated token.
When I reboot the bridge I get the following message:
[2/26/2023, 1:55:01 PM] [enphaseEnvoy] Device: 192.168.1.21 Envoy, Validate jwt token data error: AxiosError: Request failed with status code 401 Reconnect in 15s.

Any clue on why this isn't working?
Best Regards

grzegorz914, 2023-02-26:
Where you generated the token?

thdevita, 2023-02-26:
following the manual I used
https://enlighten.enphaseenergy.com/entrez-auth-token?serial_num=envoySerialNumber
using the SN of the bridge. I just pasted the token, not itself not the fully generate message (just the XXXXXX part)
{"generation_time":1677418054,"token":"XXXXXX","expires_at":1708954054}

grzegorz914, 2023-02-26:
Can you login to Envoy using web browser and this token?

thdevita, 2023-02-26:
Yes it works. I just tested it.

grzegorz914, 2023-02-26:
Try again, may be during copy/paste something happens, some time is space added to end or beginning of string.

thdevita, 2023-02-26:
It works, now, I didn't change anything but it seems that login for the first time on the Envoy using a browser did the trick.
Thanks!

grzegorz914, 2023-02-26:
Nice, then I close the issue.

closed 2023-02-26 by grzegorz914